v923z/micropython-ulab #1: ndarrays are not modifiable

andrewteta, 2019-10-25:
I've been playing around with ulab and it's awesome, but I've run into some issues because I can't modify the elements of an ndarray. 

For example: 

`>>> a = 100 * np.linspace(0,10,num=1024)`
`Traceback (most recent call last):`
`    File "<stdin>", line 1, in <module>  `
`TypeError: unsupported types for __mul__: 'int', 'ndarray'`

and

`>>> for i in range(len(b)):`
`...     b[i] = i`
`...`
`Traceback (most recent call last):`
`    File "<stdin>", line 2, in <module>`
`ValueError: index is out of range`

By the way, THANK YOU for making this module. I'm still getting my bearings with mpy and usermods, but your documentation has been very helpful. Hopefully soon I'll have some time to add some of these features in myself.

v923z, 2019-10-25:
```
for i in range(len(b)):
... b[i] = i
...
Traceback (most recent call last):
File "<stdin>", line 2, in <module>
ValueError: index is out of range
```
Indeed, this is a bug. Thanks for bringing it to the fore! A fix is in the pipeline (the subscription method required for this is undergoing major modifications), I should be able to push a solution in a couple of days.

> EDIT: I just tried `a = np.linspace(0,10,num=1024) * 100` and it worked, so that's a solid workaround!

This issue has been mentioned in the manual (https://micropython-ulab.readthedocs.io/en/latest/ulab.html#binary-operators), and has been discussed here: https://forum.micropython.org/viewtopic.php?f=3&t=7005&start=30#p40469 

In short, a hook is missing in micropython itself. (You can see from the traceback message that the exception was thrown by `int`, and not the `ndarray`. I will try to implement the fix, and issue a pull request at micropython. Till then, keep ndarrays on the left hand side!

v923z, 2019-10-29:
@andrewteta 
I have just pushed an update on the subscription methods. I believe that should fix the issue that you raised. Let me know, if it still doesn't work. I have also modified the manual, in particular, https://micropython-ulab.readthedocs.io/en/latest/ulab.html#slicing-and-indexing .

v923z, 2019-11-10:
Since no complaint has been filed in the last two weeks, I am closing the issue.

closed 2019-11-10 by v923z