v923z/micropython-ulab #2: ndarray from image ?

iabdalkader, 2019-10-29:
How can I create an ndarray of shape (w, h, c) from a C array of pixels (uint8_t) ? I can create 1D arrays by implementing getiter and unary_op, do I need to return an iter as the items of the first iter and so on ?

v923z, 2019-10-29:
Sorry, I don't understand the question: you don't have to implement anything, if the array is one-, or two-dimensional. Do you want to have a three-dimensional array? 
The title of your question indicates that you want to work with images. But those are just 2D objects. It is a different matter, whether there are facilities to read your particular format into a matrix, and there might not be. I am not sure I see what the goal is. Can you elaborate?

iabdalkader, 2019-10-29:
Hi, the image array is one dimensional but in numpy, opencv etc.. it's typically loaded as a three-dimensional array (height, width, depth), the third dimension is the depth/channels (RGB), for example:

```
import cv2
im = cv2.imread("test.png")
print(im.shape)
>>> (300, 300, 3)
```
Is there a way to achieve this with ulab ?

v923z, 2019-10-29:
The short answer is no. The slightly longer one is it is not trivial at all. Third, you also have to read the image itself. I mean the imread function. That is not trivial either, given that somehow you have to fit the png library onto the microcontroller. But if you implement that, you could easily return a 3-tuple with three two-dimensional ndarrays, each holding the data for one colour channel.

iabdalkader, 2019-10-29:
We do have some image loading functions, not png specifically but other formats, and we may have SDRAM on board at some point. Anyway thanks for the answers, and for this great library. I think I got my answer so I'll close this issue now.

v923z, 2019-12-06:
I am re-opening the issue, because I am working on an upgrade for `ulab` that would support images. In particular, the `ndim` branch has a skeleton implementation of arbitrary tensor ranks. The branch is not yet complete, but I think I should be done in a matter of weeks. I just wanted to indicate that this is really in the pipeline.

iabdalkader, 2019-12-07:
Awesome! Please let me know when it's ready and I'll help test it, thanks!

v923z, 2019-12-08:
> Awesome! Please let me know when it's ready and I'll help test it, thanks!

Help with testing would be fantastic. I assign you to this issue, then.

closed 2021-01-30 by v923z