v923z/micropython-ulab #6: Apparent C-syntax compiling errors

mdaeron, 2019-11-03:
I'm attempting to compile `ulab` for the PYBD_SF6 from a Mac running macOS 10.13.4.

- I cloned the 0.24 release of `micropython-ulab` and the current version of micropython.
- I ran `make` in `micropython/mpy-cross`
- I ran `make BOARD=PYBD_SF6 CROSS_COMPILE=/usr/local/bin/arm-none-eabi-` in `micropython/mpy-cross`, which successfully compiled the "normal" firmware for SF6.
- But when I instead run `make BOARD=PYBD_SF6 CROSS_COMPILE=/usr/local/bin/arm-none-eabi- USER_C_MODULES=../../../ulab all`, compilation fails with the error messages below.

Am I doing something wrong here? I'm not an experienced firmware compiler.

```
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c: In function 'ndarray_init_helper':
../../../ulab/code/ndarray.c:157:11: error: 'MP_QSTR_dtype' undeclared (first use in this function); did you mean 'MP_QSTR_type'?
         { MP_QSTR_dtype, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = NDARRAY_FLOAT } },
           ^~~~~~~~~~~~~
           MP_QSTR_type
../../../ulab/code/ndarray.c:157:11: note: each undeclared identifier is reported only once for each function it appears in
../../../ulab/code/ndarray.c: In function 'ndarray_flatten':
../../../ulab/code/ndarray.c:648:11: error: 'MP_QSTR_order' undeclared (first use in this function); did you mean 'MP_QSTR_ord'?
         { MP_QSTR_order, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_QSTR(MP_QSTR_C)} },
           ^~~~~~~~~~~~~
           MP_QSTR_ord
../../../ulab/code/ndarray.c:648:11: error: initialization of 'short unsigned int' from 'const mp_arg_t *' {aka 'const struct _mp_arg_t *'} makes integer from pointer without a cast [-Werror=int-conversion]
../../../ulab/code/ndarray.c:648:11: note: (near initialization for 'allowed_args[0].qst')
../../../ulab/code/ndarray.c:648:11: error: initializer element is not constant
../../../ulab/code/ndarray.c:648:11: note: (near initialization for 'allowed_args[0].qst')
In file included from ../../py/mpstate.h:35,
                 from ../../py/runtime.h:29,
                 from ../../../ulab/code/ndarray.c:15:
../../../ulab/code/ndarray.c:648:81: error: 'MP_QSTR_C' undeclared (first use in this function); did you mean 'MP_QSTR_r'?
         { MP_QSTR_order, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_QSTR(MP_QSTR_C)} },
                                                                                 ^~~~~~~~~
../../py/obj.h:92:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^~~
../../../ulab/code/ndarray.c:648:69: note: in expansion of macro 'MP_ROM_QSTR'
         { MP_QSTR_order, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_QSTR(MP_QSTR_C)} },
                                                                     ^~~~~~~~~~~
../../py/obj.h:92:30: error: initializer element is not constant
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                              ^
../../py/obj.h:248:24: note: in expansion of macro 'MP_OBJ_NEW_QSTR'
 #define MP_ROM_QSTR(q) MP_OBJ_NEW_QSTR(q)
                        ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:648:69: note: in expansion of macro 'MP_ROM_QSTR'
         { MP_QSTR_order, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_QSTR(MP_QSTR_C)} },
                                                                     ^~~~~~~~~~~
../../py/obj.h:92:30: note: (near initialization for 'allowed_args[0].defval.u_rom_obj')
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                              ^
../../py/obj.h:248:24: note: in expansion of macro 'MP_OBJ_NEW_QSTR'
 #define MP_ROM_QSTR(q) MP_OBJ_NEW_QSTR(q)
                        ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:648:69: note: in expansion of macro 'MP_ROM_QSTR'
         { MP_QSTR_order, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_QSTR(MP_QSTR_C)} },
                                                                     ^~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [build-PYBD_SF6/code/ndarray.o] Error 1
```

mdaeron, 2019-11-04:
Update: I can now compile successfully for the `unix`, `pybd_sf6` and `pybv11` ports. The issue was on my end (I finally understood that I need to do edit the `mpconfigport.h` inside each port directory. Sorry for the noise!

closed 2019-11-04 by mdaeron

v923z, 2019-11-04:
Thanks for posting the fix, too! Your original post indicates that the manual was not clear enough. I will extend the section on compilation.

mdaeron, 2019-11-04:
No problem. Constructive criticism: speaking as an inexperienced micropython compiler, it was a bit disorienting that the link to compilation instructions for `micropython-ulab` points to a section in the middle of the `micropython-usermod` documentation: it was not really clear how far back up I had to read the documentation if all I wanted to do was compile ulab.

In my opinion it would really help if the readme page for micropython-ulab just listed a simple series of steps to follow, e.g., (1) git clone micropython; (2) git clone micropython-ulab; (3) edit `mpconfigport.h` in the `micropython/ports/xxx` directory; (4) install cross-compiler; (5) run make command...

PS - IMHO `ulab` is the best improvement to micropython in years, and the quality of the `micropython-ulab` and `micropython-usermod` documentation makes a big difference for many users. Thanks again!

v923z, 2019-11-04:
> No problem. Constructive criticism: speaking as an inexperienced micropython compiler, it was a bit disorienting that the link to compilation instructions for `micropython-ulab` points to a section [...]

This is a fair point. I will update the README on the front page. By the way, I am more than happy to accept pull requests;)

mdaeron, 2019-11-04:
On Nov 4, 2019, at 19:59, Zoltán Vörös <notifications@github.com> wrote:
> 
> This is a fair point. I will update the README on the front page. By the way, I am more than happy to accept pull requests;)

I'm not truly familiar with the whole git pull/push/merge workflow. If I wanted to propose edits to the readme, I assume I need to git clone, make edits, git commit, and then send a push request to your `micropython-ulab` repo? I'll read up on that again and make an attempt.

v923z, 2019-11-04:
> On Nov 4, 2019, at 19:59, Zoltán Vörös ***@***.***> wrote: This is a fair point. I will update the README on the front page. By the way, I am more than happy to accept pull requests;)
> I'm not truly familiar with the whole git pull/push/merge workflow. If I wanted to propose edits to the readme, I assume I need to git clone, make edits, git commit, and then send a push request to your `micropython-ulab` repo? I'll read up on that again and make an attempt.

You would have to create a fork, which is then yours. Then you implement changes on that fork (you would probably pull that, modify the files on your computer, and then push it back to the server), and once you are satisfied with the modifications, you press the pull request button, which will then notify the original copy (i.e. v923z/micropython-ulab). I can then simply accept the changes, which then they become part of my branch. 

But i have already updated the README. Let me know if you would still like to extend it. But thanks for the comments, I really appreciate them!