v923z/micropython-ulab #8: Compilation errors for unix port on macOS

mdaeron, 2019-11-07:
I am able to compile micropython+ulab for the PYBD_SF6 but not for unix.

My batch script is the following:

```
mkdir build
cd build

git clone https://github.com/micropython/micropython.git
git clone https://github.com/v923z/micropython-ulab.git ulab

cd micropython
git submodule update --init

cd mpy-cross/
make

cd ../ports/stm32
echo '#define MODULE_ULAB_ENABLED (1)' | cat >> mpconfigport.h
make BOARD=PYBD_SF6 CROSS_COMPILE=/usr/local/bin/arm-none-eabi- USER_C_MODULES=../../../ulab all

cd ../unix
echo '#define MODULE_ULAB_ENABLED (1)' | cat >> mpconfigport.h
make USER_C_MODULES=../../../ulab all
```

When running the above code, I get a functional PUBD_SF6 firmware but the unix make results in 3 `-Wparentheses-equality` errors in `ulab/code/ndarray.c`:

```
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c:583:20: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
    if((ndarray->m == 1)) {
        ~~~~~~~~~~~^~~~
../../../ulab/code/ndarray.c:583:20: note: remove extraneous parentheses around the comparison to silence this warning
    if((ndarray->m == 1)) {
       ~           ^   ~
../../../ulab/code/ndarray.c:583:20: note: use '=' to turn this equality comparison into an assignment
    if((ndarray->m == 1)) {
                   ^~
                   =
../../../ulab/code/ndarray.c:891:39: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
            if((self->array->typecode == NDARRAY_INT8)) {
                ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:891:39: note: remove extraneous parentheses around the comparison to silence this warning
            if((self->array->typecode == NDARRAY_INT8)) {
               ~                      ^              ~
../../../ulab/code/ndarray.c:891:39: note: use '=' to turn this equality comparison into an assignment
            if((self->array->typecode == NDARRAY_INT8)) {
                                      ^~
                                      =
../../../ulab/code/ndarray.c:896:46: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
            } else if((self->array->typecode == NDARRAY_INT16)) {
                       ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:896:46: note: remove extraneous parentheses around the comparison to silence this warning
            } else if((self->array->typecode == NDARRAY_INT16)) {
                      ~                      ^               ~
../../../ulab/code/ndarray.c:896:46: note: use '=' to turn this equality comparison into an assignment
            } else if((self->array->typecode == NDARRAY_INT16)) {
                                             ^~
                                             =
3 errors generated.
make: *** [build/code/ndarray.o] Error 1
```

If I add `-Wno-parentheses-equality` to the `CFLAGS` in `micropython/ports/unix/Makefile`, I get different errors, in `ulab/code/linalg.c` this time:

```
CC ../../../ulab/code/ndarray.c
CC ../../../ulab/code/linalg.c
../../../ulab/code/linalg.c:127:12: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
        if(abs(data[m*(N+1)]) < epsilon) {
           ^
../../../ulab/code/linalg.c:127:12: note: use function 'fabs' instead
        if(abs(data[m*(N+1)]) < epsilon) {
           ^~~
           fabs
../../../ulab/code/linalg.c:310:12: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
        if(abs(tmp[m*(in->n+1)]) < epsilon) {
           ^
../../../ulab/code/linalg.c:310:12: note: use function 'fabs' instead
        if(abs(tmp[m*(in->n+1)]) < epsilon) {
           ^~~
           fabs
../../../ulab/code/linalg.c:349:26: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
            if(epsilon < abs(array[m*in->n + n] - array[n*in->n + m])) {
                         ^
../../../ulab/code/linalg.c:349:26: note: use function 'fabs' instead
            if(epsilon < abs(array[m*in->n + n] - array[n*in->n + m])) {
                         ^~~
                         fabs
3 errors generated.
make: *** [build/code/linalg.o] Error 1
```

I suspect this might be due to some quirk of the macOS C compiler, but that's not much more than a random guess.

Is there anything obvious that I'm doing wrong?

v923z, 2019-11-08:
For the unix port, I think you have to compile the axtls module, too: https://github.com/micropython/micropython/wiki/Getting-Started 
There is also a section on mac. I can't verify that, because I haven't a mac. In any case, if that works, please, let us know.

mdaeron, 2019-11-08:
`make axtls` yields: `make: Nothing to be done for 'axtls'`. Everything else is the same.

Also, if I leave out `ulab`, I can compile for unix with just `make` (no `make axtls` is necessary)

v923z, 2019-11-08:
Does https://github.com/micropython/micropython/wiki/Getting-Started help in any way? I also noticed that the mpconfigport.h contains a lot of flags for apple. Can you learn anything from there?

v923z, 2019-11-08:
Could you, please, check out the testing branch: https://github.com/v923z/micropython-ulab/tree/testing , and see, whether you can compile for the unix port? Also note the caveats in https://github.com/v923z/micropython-ulab/issues/9#issuecomment-551949087 .

mdaeron, 2019-11-08:
> On the other hand, by looking at the errors, I think I know what the problem is: the abs function is missing; you would need fabs for unix. I will push a fix over the weekend. Can you check, whether the error goes away, if you replace abs with MICROPY_FLOAT_C_FUN(abs)? I meant to replace all occurrances, but I missed this one.
I replaced `abs` by `MICROPY_FLOAT_C_FUN(fabs)` (note `fabs` not `abs`), and I get additional errors, this time in `poly.c`.

```
CC ../../../ulab/code/poly.c
../../../ulab/code/poly.c:111:15: error: variable 'deg' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
    } else if(n_args == 3) {
              ^~~~~~~~~~~
../../../ulab/code/poly.c:129:29: note: uninitialized use occurs here
    XT = m_new(mp_float_t, (deg+1)*leny); // XT is a matrix of shape (deg+1, len) (rows, columns)
                            ^~~
../../py/misc.h:60:60: note: expanded from macro 'm_new'
#define m_new(type, num) ((type*)(m_malloc(sizeof(type) * (num))))
                                                           ^~~
../../../ulab/code/poly.c:111:12: note: remove the 'if' if its condition is always true
    } else if(n_args == 3) {
           ^~~~~~~~~~~~~~~~
../../../ulab/code/poly.c:93:16: note: initialize the variable 'deg' to silence this warning
    uint8_t deg;
               ^
                = '\0'
../../../ulab/code/poly.c:111:15: error: variable 'leny' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
    } else if(n_args == 3) {
              ^~~~~~~~~~~
../../../ulab/code/poly.c:130:26: note: uninitialized use occurs here
    for(uint8_t i=0; i < leny; i++) { // column index
                         ^~~~
../../../ulab/code/poly.c:111:12: note: remove the 'if' if its condition is always true
    } else if(n_args == 3) {
           ^~~~~~~~~~~~~~~~
../../../ulab/code/poly.c:92:24: note: initialize the variable 'leny' to silence this warning
    uint16_t lenx, leny;
                       ^
                        = 0
2 errors generated.
```

I'll check out the testing branch now.

mdaeron, 2019-11-08:
> Could you, please, check out the testing branch: https://github.com/v923z/micropython-ulab/tree/testing , and see, whether you can compile for the unix port?

I now get the following error message:
```
make: *** No rule to make target `code/test.c', needed by `build/genhdr/qstr.i.last'.  Stop.
```

When I run `make all` without `USER_C_MODULES=../../../ulab`, micropython compiles just fine.

v923z, 2019-11-09:
> ```
> make: *** No rule to make target `code/test.c', needed by `build/genhdr/qstr.i.last'.  Stop.
> ```

I have removed the the test module.

mdaeron, 2019-11-09:
Just checking, here's how I switched to the `testing` branch:

```
git clone https://github.com/v923z/micropython-ulab.git ulab
cd ulab
git checkout testing

cd ..
git clone https://github.com/micropython/micropython.git
cd micropython
git submodule update --init

cd mpy-cross/
make

cd ../ports/unix
echo '#define MODULE_ULAB_ENABLED (1)' >> mpconfigport.h
make USER_C_MODULES=../../../ulab all
```

Is that the correct way? This results in the following error:

```
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c:925:39: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
            if((self->array->typecode == NDARRAY_INT8)) {
                ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:925:39: note: remove extraneous parentheses around the comparison to silence this warning
            if((self->array->typecode == NDARRAY_INT8)) {
               ~                      ^              ~
../../../ulab/code/ndarray.c:925:39: note: use '=' to turn this equality comparison into an assignment
            if((self->array->typecode == NDARRAY_INT8)) {
                                      ^~
                                      =
../../../ulab/code/ndarray.c:930:46: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
            } else if((self->array->typecode == NDARRAY_INT16)) {
                       ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../../../ulab/code/ndarray.c:930:46: note: remove extraneous parentheses around the comparison to silence this warning
            } else if((self->array->typecode == NDARRAY_INT16)) {
                      ~                      ^               ~
../../../ulab/code/ndarray.c:930:46: note: use '=' to turn this equality comparison into an assignment
            } else if((self->array->typecode == NDARRAY_INT16)) {
                                             ^~
                                             =
2 errors generated.
make: *** [build/code/ndarray.o] Error 1
```

I then removed the extra parentheses in `ndarray.c`, but this yields a new error in `poly.c`:
```
CC ../../../ulab/code/poly.c
../../../ulab/code/poly.c:111:15: error: variable 'deg' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
    } else if(n_args == 3) {
              ^~~~~~~~~~~
../../../ulab/code/poly.c:129:29: note: uninitialized use occurs here
    XT = m_new(mp_float_t, (deg+1)*leny); // XT is a matrix of shape (deg+1, len) (rows, columns)
                            ^~~
../../py/misc.h:60:60: note: expanded from macro 'm_new'
#define m_new(type, num) ((type*)(m_malloc(sizeof(type) * (num))))
                                                           ^~~
../../../ulab/code/poly.c:111:12: note: remove the 'if' if its condition is always true
    } else if(n_args == 3) {
           ^~~~~~~~~~~~~~~~
../../../ulab/code/poly.c:93:16: note: initialize the variable 'deg' to silence this warning
    uint8_t deg;
               ^
                = '\0'
../../../ulab/code/poly.c:111:15: error: variable 'leny' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
    } else if(n_args == 3) {
              ^~~~~~~~~~~
../../../ulab/code/poly.c:130:26: note: uninitialized use occurs here
    for(uint8_t i=0; i < leny; i++) { // column index
                         ^~~~
../../../ulab/code/poly.c:111:12: note: remove the 'if' if its condition is always true
    } else if(n_args == 3) {
           ^~~~~~~~~~~~~~~~
../../../ulab/code/poly.c:92:24: note: initialize the variable 'leny' to silence this warning
    uint16_t lenx, leny;
                       ^
                        = 0
2 errors generated.
```
This one is a little out of my current skill level, so I haven't tried to correct it myself.

Just wondering: is all of this essentially due to the fact that the Mac `clang` is stricter and/or different in what it considers to be errors than the Linux compiler?

v923z, 2019-11-09:
> Just checking, here's how I switched to the `testing` branch:
>
> ```
> git clone https://github.com/v923z/micropython-ulab.git ulab
> cd ulab
> git checkout testing
>
> cd ..
> git clone [...]

This looks all right.

> ```
> CC ../../../ulab/code/ndarray.c
> ../../../ulab/code/ndarray.c:925:39: error: equality comparison with extraneous parentheses [-Werror,-Wparentheses-equality]
>             if((self->array->typecode [...]

I have tried to weed out all occurrences. Let me know if you still get this warning. 

> I then removed the extra parentheses in `ndarray.c`, but this yields a new error in `poly.c`:
>
> ```
> CC ../../../ulab/code/poly.c
> ../../../ulab/code/poly.c:111:15: error: variable 'deg' is used [...]

This extremely strict, but I have removed the if statement, and that should shut up the compiler. 

> ../../../ulab/code/poly.c:129:29: note: uninitialized use occurs here
>     XT = m_new(mp_float_t, (deg+1)*leny); // XT is a matrix of shape (deg+1, len) (rows, columns) [...]

Is this an error, or a warning only? 

> ../../../ulab/code/poly.c:111:12: note: remove the 'if' if its condition is always true
>     } else if(n_args == 3) {
>            ^~~~~~~~~~~~~~~~
> ../../../ulab/code/poly.c:93:16: note: initialize the variable 'deg' to silence this warning
>     uint8_t deg;
>                ^

I think, this is actually an incorrect warning from the compiler, but I initialised `deg` with 0 (it will be overwritten later, and this is why I am a bit baffled by the complaint). 

> = '\0'
> ../../../ulab/code/poly.c:111:15: error: variable 'leny' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
>     } else if(n_args == 3) { [...]

`lenx = 0, leny = 0` now, so this error should no longer appear. 

> ```
> 
> This one is a little out of my current skill level, so I haven't tried to correct it myself.
>
> Just wondering: is all of this essentially due to the fact that the Mac `clang` is stricter and/or different in what it considers to be errors than the Linux compiler?

Probably. It is hard for me to comment on this, because I am compiling on linux. But I really appreciate your efforts, and I hope that we can find a solution. 

In any case, can you, please, check out testing, and try to compile the code again?

mdaeron, 2019-11-09:
> In any case, can you, please, check out testing, and try to compile the code again?

Congratulations, it works! Thanks for taking the time to troubleshoot this, and let me know if you ever need to test Mac compilation again on a test branch. On my end I can check new releases when they come out and let you know if anything looks wrong.

v923z, 2019-11-09:
> Thanks for taking the time to troubleshoot this, and let me know if you ever need to test Mac compilation again on a test branch. 

Fantastic, thanks for the update! I close the issue, then. 

> On my end I can check new releases when they come out and let you know if anything looks wrong.

If you don't mind getting a notification once or twice a day, you can press the watch button:) I think it would be really great to have some feedback from mac.

closed 2019-11-09 by v923z