v923z/micropython-ulab #11: Support for matrix multiplication operator

nickovs, 2019-11-10:
Micropython now supports the PEP 465 syntax for using `@` as an infix binary operator for matrix multiplication (following the merge of micropython/micropython#4947). Classes implemented in C will get their `binary_op` method called with the `MP_BINARY_OP_MAT_MULTIPLY` operator code when the user does this. It would be great if `ulab` supported this syntax in the same way that `numpy` does, since it makes code much more concise and readable, as well as making `numpy` code more directly portable.

v923z, 2019-11-10:
> ... It would be great if `ulab` supported this syntax in the same way that `numpy` does, since it makes code much more concise and readable, as well as making `numpy` code more directly portable.

I definitely agree.

v923z, 2019-11-12:
I have added the subject to https://github.com/v923z/micropython-ulab/issues/14, so I am closing this now.

closed 2019-11-12 by v923z