v923z/micropython-ulab #13: Missing .T transpose property

nickovs, 2019-11-11:
In `numpy` the transpose of an array `x` can be accessed using the property `x.T`. This is a widely used feature that should be supported.

It should be noted in `numpy` the `T` property points to the same underlying value set, with the array stride values and dimensions reversed. Unless `ulab` changes to track the array strides for all dimensions this may be difficult to replicate.

v923z, 2019-11-12:
> 
> 
> In `numpy` the transpose of an array `x` can be accessed using the property `x.T`. This is a widely used feature that should be supported.

I would love to, but I haven't yet been able to figure out what properties look like in C. This actually pertains to `.size`, `.shape`, and `.dtype`, too. 

On the other hand, I believe, a word of caution would be in place here: in `numpy`, many things can be achieved in more than one way. But implementing a one-to-one behavioural copy on a microcontroller is flash-wise expensive. So, I am actually wondering, wouldn't it be better to stick to bare minimum?

> It should be noted in `numpy` the `T` property points to the same underlying value set, with the array stride values and dimensions reversed. Unless `ulab` changes to track the array strides for all dimensions this may be difficult to replicate.

In `ulab`, data are laid out in C format, and the highest dimension is two, which means, it doesn't really matter, what you have: a 1-D array is basically identical to a 2-D array. For this reason, strides and the like are not really needed. The downside of that is that transposition can't be done in place, so in the function, the memory positions have to be shifted: https://github.com/v923z/micropython-ulab/blob/ea588a67df65e4925e6fb58d1a17a51f70b4c9bc/code/linalg.c#L24 
Or have I misunderstood you?

nickovs, 2019-11-12:
Regarding creating properties from C, I believe it is as simple as creating a regular method for the getter, instantiating a static, ROM-resident `mp_obj_property_t` with your getter method in `.proxy[0]` and pointing to that in the property object in the `.locals_dict` array. Since you won't need setters or deleters you can leave `.proxy[1]`  and `.proxy[2]`  as `None`.

Regarding storage structure and strides, I started writing bits of a `numpy` replacement for Micropython myself before I found your code. In my version the `ndarray` class pointed to a regular built-in `array` for the storage and added a base offset for where in that array to start and then length and stride for every dimension. This allowed it to preserve the semantics for both slices and transposition since the resulting slice or transpose would just point to the same underlying `array` (with memory management handled by the normal garbage collection process). Slicing and transposition are then just cases of changing the base, stride and length and done without any copying. This increases the storage needed for a single `ndarray` by one word per dimension (for the strides) but removes all data storage overhead for all the slices and transposes (and transposes of slices!)

The other difference between the code I had started to write and yours is that I handled an arbitrary number of dimensions (i.e. proper `ndarray` rather than `2darray`. I needed that for some of the more complex [Kalman filter](https://en.wikipedia.org/wiki/Kalman_filter) code that I had for reconciling multiple sets of sensor data. If you are planning to leave your code as 2D only then I'll likely finish my code.

v923z, 2019-11-12:
> Regarding creating properties from C, I believe it is as simple as creating a regular method for the getter, instantiating a static, ROM-resident `mp_obj_property_t` with your getter method in [...]

OK, I will check that out. I will also close this issue, and open another one titled `wishlist`. In that way, there is a single point of reference for enhancement-related questions and suggestions. 

> Regarding storage structure and strides, I started writing bits of a `numpy` replacement for Micropython myself before I found your code. In my version the `ndarray` class pointed to a regular [...]

So, in principle, you would like to change the `ndarray` structure https://github.com/v923z/micropython-ulab/blob/97e98c4d3164da5ae2d6c735d118a61d16e0b479/code/ndarray.h#L37 as follows:
```
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t boolean;
    mp_tuple_t dims;
    mp_obj_array_t *array;
} ndarray_obj_t;
```
and then calculate the offsets from the members of the `dims` tuple. 

> This allowed it to preserve the semantics for both slices and transposition since the resulting slice or transpose would just point to the same underlying `array` (with memory management handled by [...]

But doesn't that mean that other operations (e.g., matrix multiplication) will become cumbersome? 

> The other difference between the code I had started to write and yours is that I handled an arbitrary number of dimensions (i.e. proper `ndarray` rather than `2darray`. I needed that for some of the [...]

I am not against moving to higher dimensions, in fact, I have been pondering that lately. But I don't see too much point in writing two more or less identical libraries. So, if you think that you need 3 or 4 dimensions, then we should do that in a single code base. Just think about it: the python community is significantly larger than that of micropython, yet, there is practically only one numerical library, and almost everything builds on that. We should not fragment developer resources. 

I should also mention that writing a specialised library is not so difficult. You run into problems when you want to make it somewhat universal. The fact that `numpy` itself is inconsistent in cases just makes the whole situation more exasperating. I don't know how far you are with your code, but I noticed that I spend more time with properly sorting out the meaning of the arguments than with the numerical algorithms.

closed 2019-11-12 by v923z

nickovs, 2019-11-13:
> So, in principle, you would like to change the ndarray structure as follows:
```C
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t boolean;
    size_t ndim;
    mp_tuple_t shape;
    mp_tuple_t strides;
    mp_obj_array_t *array;
} ndarray_obj_t;
```

Something like that. The `mp_obj_tuple_t` structure keeps its values in consecutive memory after the object header and therefore are of variable size, so you would need those be pointers. Since you don't need the type flexibility of a true tuple I'd probably go for something more like:
```C
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t boolean;
    size_t ndim;
    mp_obj_array_t *array;
    size_t base_offset;
    size_t len_stride_pairs[];
} ndarray_obj_t;
```
so that everything but the underlying array is in the same object. Note that you need to include a `base_offset` in case someone takes a slice that doesn't start at zero.

> But doesn't that mean that other operations (e.g., matrix multiplication) will become cumbersome?

Not really. The only difference is that in the inner-most loop your offset into the array needs to be incremented by the stride of the last dimension instead of being incremented by 1 each time you go around. Note that this even works for negative strides if someone needs to reverse an array (in which case `base_offset` ends up pointing to the last item not the first).

Also note that supporting broadcasting is easy in this model. It goes like this:
 1) Copy the length/stride pairs into a couple of arrays that are max(a.ndim, b.ndim) long. If one of the operands has fewer dimensions than the other then pad the _front_ of a copy with pairs of length=1, stride=0 and put the copied values at the end.
 1) For each dimension, if one of the two dimension lengths is 1 and the other is not then replace the 1 with the length value from the other argument and set the stride to be 0. If both are non-1 and not the same then the arrays are incompatible for broadcast.
 1) Recursively call a function that starts at the given offsets into each array, iterates over the outermost dimension adding the outer-most stride to the offset each time around and either applies the universal function to the values at the offsets or calls itself again with one fewer dimensions.
This even works for broadcasting a single value since; you just treat it as an array with zero dimensions. Note that you can't broadcast in the left-hand side of an in-place binary operator (since the result won't fit!)

This approach will be universal for any number of dimensions and should have reasonable code size and performance.

v923z, 2019-11-13:
> Something like that. The `mp_obj_tuple_t` structure keeps its values in consecutive memory after the object header and therefore are of variable size, so you would need those be pointers. Since you [...]

Tuples are expensive, one can save some RAM with a `size_t`pointer. But I think, you still need the shape somewhere. 

> Note that you need to include a `base_offset` in case someone takes a slice that doesn't start at zero.

That's right. And it had better be a pointer, because you can have a slice for each axis. So, my proposal would be 

```c
typedef struct _ndarray_obj_t {
     mp_obj_base_t base;
     uint8_t boolean;
     size_t ndim;
     size_t *shape;
     size_t *stride;
     size_t *offset;
     mp_obj_array_t *array;
} ndarray_obj_t;
 ```

>  This even works for broadcasting a single value since; you just treat it as an 
> array with zero dimensions. Note that you can't broadcast in the left-hand side of 
> an in-place binary operator (since the result won't fit!)

But this is OK in `numpy`: 
```
a = array([[1, 2, 3], [4, 5, 6], [7, 8, 6]])
b = array([10, 20, 30])
a += b
```
as is 
```
a += 1
```

And ndarrays have to be copied, when the index is another `ndarray`. I don't see, how those cases can be handled with views. (See https://github.com/v923z/micropython-ulab/issues/14)

> This approach will be universal for any number of dimensions and should have reasonable code size and performance.

I have started to whip up an `ndarray` along these lines. The modification of the rest of the functions is almost trivial. I mean, a lot of work, but definitely not rocket science.

nickovs, 2019-11-13:
You don’t need a list of offsets. The slice is always an evenly spaced lattice subset of the original array, so all you need it do know where the origin is and what the strides are on each dimension. 

Yes you need to be able to return the shape as a tuple but it will likely be cheaper to store it as a contiguous array of `size_t` and make the shape tuple on the fly when needed. 

You are correct that you can broadcast the values in the right hand side of an in-place operation. What you can’t do, using you example values, is `b += a`. Thus if you apply the method that I described above then you need to make sure that none of the dimensions of the left value got broadcast if the operation is in-place. 

I don’t think that this model has any real impact when indexing with a Boolean mask array. In those cases you are always going to need to walk the dimensions and copy the data into a new array; you just need to walk the dimensions with the new origin offset and strides. My understanding is that in `numpy` access based on mask arrays always delivers a copy not a view anyway.

v923z, 2019-11-13:
> You don’t need a list of offsets. The slice is always an evenly spaced lattice subset of the original array, so all you need it do know where the origin is and what the strides are on each dimension.

True.

> 
> Yes you need to be able to return the shape as a tuple but it will likely be cheaper to store it as a contiguous array of `size_t` and make the shape tuple on the fly when needed.

Right, that is why I wanted to have `size_t *` for that. 

> You are correct that you can broadcast the values in the right hand side of an in-place operation. What you can’t do, using you example values, is `b += a`. Thus if you apply the method that I [...]

OK, that makes sense now. 

> I don’t think that this model has any real impact when indexing with a Boolean mask array. In those cases you are always going to need to walk the dimensions and copy the data into a new array; you [...]

This statement applies not only to Booleans, but `ndarray`s in general (when they can be interpreted as indices). The example 
```
a = array([1, 2, 3])

k = array([2, 0, 1, 1, 0, 2, 2, 1, 0])
b = a[k] # yields array([3, 1, 2, 2, 1, 3, 3, 2, 1])
```
should result in a new copy I believe. I don't think that this can reasonably be represented as a view.

nickovs, 2019-11-13:
Yes, I think that any access indexed on an array needs to return a copy (and I’m pretty sure that is what numpy does).

nickovs, 2019-11-14:
Note that implementing this model will likely fix the misrepresentation of 1-D arrays and 2-D arrays that have one dimension of length 1, as described in #15.