v923z/micropython-ulab #14: Wishlist

v923z, 2019-11-12:
This is the place, where suggestions related to new function/functionality and the like should be brought up.

v923z, 2019-11-12:
Add properties, like `.T`, `.dtype`, `.shape` and the like to the `ndarray` class.

v923z, 2019-11-12:
Let `ndarray`s be of arbitrary dimension.

v923z, 2019-11-12:
Add support for matrix multiplication, `@`, operator as described in https://github.com/v923z/micropython-ulab/issues/11

v923z, 2019-11-12:
Let `ndarray`s be on the right hand side of binary operators as described in https://github.com/v923z/micropython-ulab/issues/10

mdaeron, 2019-11-13:
Allow indexing an array with another array of integers or booleans, similar to `numpy`:

```
from numpy import array
a = array([1, 2, 3])

k = array([2, 0, 1, 1, 0, 2, 2, 1, 0])
print(a[k]) # yields array([3, 1, 2, 2, 1, 3, 3, 2, 1])

k = array([True, False, True])
print(a[k]) # yields array([1, 3])
```

SummerGGift, 2019-11-19:
I find ur micropython-usermod series articles today, ur works is great, it's very helpful. 

I have used micropython in my work for about two years, during the time, couldn't find many useful articles describe the details in it. i think i will translate some chapters into chinese, and spread them in China. 

Thank you very much.

v923z, 2019-11-19:
> I find ur micropython-usermod series articles today, ur works is great, it's very helpful.
>
> I have used micropython in my work for about two years, during the time, couldn't find many useful [...]

Thank you for your feedback! Since I put the material deliberately into the public domain, I have absolutely nothing against your translating the text.

roger-, 2020-01-23:
How about a circular buffer compatible with ndarray (but fixed size of course)? Would avoid all the memmoves needed with roll().

v923z, 2020-01-23:
This is definitely true, in fact, I also wanted to implement this. This would be extra compared to numpy, right? Actually, this is why I chose the name `ulab` and not `unumpy` or something similar.

Basically, you want a sub-class of ndarray, with a special method, say, `insert`, which would move the pointer. The question is, perhaps, what would you do with the numerical methods? Some do not care about the order of the elements (sum, mean, std, FFT, etc.), while some do (argmin, polyfit, polyval). One other problem is slicing. If you have any suggestions, I am listening.

roger-, 2020-01-23:
I’ve seen discussions about having numpy support this, but currently I don’t think it does. I believe Micropython has it though, but only exposed at the C level. 

Yup I think an insert() would be the main addition. Or maybe insert() and insertleft() for more flexibility. 

Wouldn’t all methods be applicable? Order should be preserved with the only difference being the size will change (up to a maximum) depending on how much data it contains. I think a given circular buffer should be usable the same as a regularly array to the user (so it should support argmin, fft, etc.)

v923z, 2020-01-23:
Oh, snap! I have accidentally edited your comment. Sorry!

roger-, 2020-01-23:
Haha no prob :)

Here’s the MicroPython one: https://github.com/micropython/micropython/blob/master/py/ringbuf.c

Internally the pointer is moving but from the users perspective it’s a linear array, right? The index will only be valid once you don’t change the array.

v923z, 2020-01-23:
> Here’s the MicroPython one: https://github.com/micropython/micropython/blob/master/py/ringbuf.c

Thanks! 

> Internally the pointer is moving but from the users perspective it’s a linear array, right? The index will only be valid once you don’t change the array.

Let us suppose, I insert multiples of 10 into an initially empty (is that zero, by the way, or something else?) buffer:

```
[10, 0, 0, 0]: pointer is 0
[10, 20, 0, 0]: pointer is 1
[10, 20, 30, 0]: pointer is 2
[10, 20, 30, 40]: pointer is 3
[50, 20, 30, 40]; pointer is 0
[50, 60, 30, 40]: pointer is 1
[50, 60, 70, 40]: pointer is 2
[50, 60, 70, 80]: pointer is 3
[90, 60, 70, 80]: pointer is 0
...
```
Is it what you meant by the circular buffer? In other words, when the pointer is at `position`, and you insert a new item, you do it at `position+1 modulo length`.

Now, what is the value of argmin([50, 60, 30, 40])? Do we regard this as a linear array, beginning at position 0, or is it supposed to be interpreted as [30, 40, 50, 60], since we know that the pointer is at the position of 60?

roger-, 2020-01-23:
I envision having a start and end pointer. When the user indexes the array, they mean the offset from the start pointer. 

Not sure if that makes sense.

v923z, 2020-01-23:
> I envision having a start and end pointer. When the user indexes the array, they mean the offset from the start pointer.
> 
> Not sure if that makes sense.

It does, and exactly this is why it is a bit more than an `ndarray`, so certain methods must be equipped with appropriate facilities. It should be doable, though...

roger-, 2020-01-23:
Depends on if you insert at the start or end pointer (one will move at a time). 

It should be possible to wrap all of this so that functions can be used for both linear or circular arrays but it may be a lot of work (probably easier in C++).

v923z, 2020-01-23:
> Depends on if you insert at the start or end pointer (one will move at a time).

OK, so this is not really critical.

> It should be possible to wrap all of this so that functions can be used for both linear or circular arrays but it may be a lot of work (probably easier in C++).

Not necessarily: one simply has to shuffle the array for `argmin`, `polyfit`, `polyval`, and slicing.

jepler, 2020-02-05:
`numpy.convolve`

v923z, 2020-02-05:
> `numpy.convolve`

This is an interesting idea, and it could be quite useful. I will look into that. Alternatively, you can take a stab at the implementation, and then issue a pull request:) I assume you meant 1D, real, right?

jepler, 2020-02-05:
Yes, we want to try using convolve as a way to do high/low pass filters on sensor data in circuitpython, so 1d is fine as an initial case.  We'll create the filters using pure Python code, I think, since they are create once use many times.

v923z, 2020-02-05:
> Yes, we want to try using convolve as a way to do high/low pass filters on sensor data in circuitpython, so 1d is fine as an initial case. We'll create the filters using pure Python code, I think, since they are create once use many times.

Oh, then this is a slightly different issue. For similar reasons, I have been considering a filter module, something along the lines of https://github.com/peterhinch/micropython-filters . I have to admit, I don't know enough about digital filters, though, and that is why I sort of gave up on the idea. 
But I will try to hack a convolution implementation over the weekend, perhaps even earlier.

jepler, 2020-02-05:
@v923z I'll give it a try.  Would convolution go in numerical?  I was thinking that always returning floats was in some sense easiest.  numpy seems to do modulo arithmetic when working with dtypes like int8 fwiw

jepler, 2020-02-05:
also, np.convolve supports 3 modes; I think we'd start with the default of "full" again for simplicitly

v923z, 2020-02-05:
> @v923z I'll give it a try. Would convolution go in numerical? 

Based on my comment above, I would create a new module, filter.c/h, so that later that can be extended with other filter functions. 

> I was thinking that always returning floats was in some sense easiest. 

I believe that, too. upcasting is expensive in terms of flash space. In order to get the float values, you could call `ndarray_get_float_value();` from `ndarray.c.` . 

I have already started to work on an implementation, but I don't want to steal your thunder. Let me know if you want me to take over. 

> also, np.convolve supports 3 modes; I think we'd start with the default of "full" again for simplicity

Fine with me.

mouryarahul, 2020-03-28:
@v923z do you plan to add in near future something equivalent to **_numpy.power_** in the **vector** sub-module as the existing **_arctan2_** ?

v923z, 2020-03-30:
> @v923z do you plan to add in near future something equivalent to **_numpy.power_** in the **vector** sub-module as the existing **_arctan2_** ?

Actually, what I wanted to add is `**` to the binary operators. Or is there a significant difference between `**`, and `power`?

mouryarahul, 2020-03-30:
Thanks for your comment. I tried to see any difference between `**` and `power` in the CPython, but didn't see any difference. So, implementing `**` as the binary operator will serve the purpose. I agree with you that power operation is a bit more involved. When I checked documentation of `numpy.power`, it mentions: "Note that an integer type raised to a negative integer power will raise a ValueError." And for the particular example you mentioned, the results are:
`>> b = np.array([-1,-2,-3],dtype=float)`
`>> np.power(b, 0.5)`
`array([nan, nan, nan])`
`>> b**0.5`
`__main__:1: RuntimeWarning: invalid value encountered in sqrt`
`array([nan, nan, nan])`
`>> c = np.array([-1, -2, -3], dtype=int)`
`>> np.power(c, 0.5)`
`array([nan, nan, nan])`
`>> c**0.5`
`array([nan, nan, nan])`

So, I think we could also raise similar exceptions and restrict to the only frequently used scenarios.

v923z, 2020-03-30:
Many thanks for digging out the details!

> Thanks for your comment. I tried to see any difference between `**` and `power` in the CPython, but didn't see any difference. So, implementing `**` as the binary operator will serve the purpose. 

On second thought, there might be a reason for not doing so: if you implement `**`, then you can no longer exclude it, because it will be a method of the `ndarray`. So, I am leaning towards implementing `power` in `vectorise.c`, and simply provide `**` as an alias for the case, when `vectorise` is included. 

> I agree with you that power operation is a bit more involved. When I checked documentation of `numpy.power`, it mentions: "Note that an integer type raised to a negative integer power will raise a ValueError." 

So, the claim is that integer types should only be raised to positive powers, right? And the results will be integers only, if the positive powers are also integers, otherwise we can return floats. 

> And for the particular example you mentioned, the results are:
> `>> b = np.array([-1,-2,-3],dtype=float)`
> `>> np.power(b, 0.5)`
> `array([nan, nan, nan])`
> `>> b**0.5`
> `__main__:1: RuntimeWarning: invalid value encountered in sqrt`

This would mean that `numpy` recognised 0.5 as 1/2, and called the square root function. I presume, the reason for that is that `sqrt` is probably faster than `pow` with 0.5. This would definitely add a lot of flash overhead, so in the first round I would simply call `pow` with whatever arguments. If it is unbearably slow, we can think about alternatives. 

> `array([nan, nan, nan])`
> `>> c = np.array([-1, -2, -3], dtype=int)`
> `>> np.power(c, 0.5)`
> `array([nan, nan, nan])`
> `>> c**0.5`
> `array([nan, nan, nan])`
> 
> So, I think we could also raise similar exceptions and restrict to the only frequently used scenarios.

OK. Give me a day or two! Or, if you feel adventurous, you can also try to implement it.

mouryarahul, 2020-03-31:
> Many thanks for digging out the details!
>
>
> On second thought, there might be a reason for not doing so: if you implement `**`, then you can no longer exclude it, because it will be a method of the [...]

In fact, NumPy has implemented all these kinds of functions with verbal names rather than using symbolics intentionally to keep it clear,  such as numpy.power, numpy.multiply, numpy.matmul, etc.

> So, the claim is that integer types should only be raised to positive powers, right? And the results will be integers only, if the positive powers are also integers, otherwise we can return floats.

> This would mean that `numpy` recognised 0.5 as 1/2, and called the square root function. I presume, the reason for that is that `sqrt` is probably faster than `pow` with 0.5. This would definitely [...]

> 
> > `array([nan, nan, nan])`
> > `>> c = np.array([-1, -2, -3], dtype=int)`
> > `>> np.power(c, 0.5)`
> > `array([nan, nan, nan])`
> > `>> c**0.5`
> > `array([nan, nan, nan])`
> > So, I think we could also raise similar exceptions and restrict to the only frequently used scenarios.
> 

> OK. Give me a day or two! Or, if you feel adventurous, you can also try to implement it.

In fact, I did the implementation in `vectorise.c`, which is the exact copy of `arctan2` and replaced the final calculation by:

`*array_r++ = MICROPY_FLOAT_C_FUN(pow)(value_x, value_y);`

And this produces exactly the same result as the numpy.power in many cases, except that it upcast the result to float in all cases.

mouryarahul, 2020-03-31:
A minimalistic equivalent of numpy.copy can be useful in many cases.  How do I perform in straight way:
`ulab.linalg.dot(A.transpose(), A)` ? because `transpose` is performed in-place. In Numpy, we can perform:

> a
array([[1., 2., 3.],
       [4., 5., 6.],
       [7., 8., 9.]])
> np.dot(a.transpose(), a)
array([[ 66.,  78.,  90.],
       [ 78.,  93., 108.],
       [ 90., 108., 126.]])

whereas in ulab, it results into:

> np.linalg.dot(a.transpose(), a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: arguments must be ndarrays

@v923z What do you suggest?

v923z, 2020-03-31:
> A minimalistic equivalent of numpy.copy can be useful in many cases. How do I perform in straight way:
> `ulab.linalg.dot(A.transpose(), A)` ? because `transpose` is performed in-place. 

This seems to be an inconsistency. Thanks for spotting it! I will open a new ticket. 

> @v923z What do you suggest?

You could always do something like this:
```
b = a + 0
b = a * 1
```
(The first is a bit cheaper.)

v923z, 2020-03-31:
> 
> `*array_r++ = MICROPY_FLOAT_C_FUN(pow)(value_x, value_y);`
> 
> And this produces exactly the same result as the numpy.power in many cases, except that it upcast the result to float in all cases.

That might be OK in certain cases, and it can be fatal in others. E.g., if you need the remainder of the powers of 3, when they are divided by 256, you can do that as 

```
a = array(range(10), dtype=uint8)
3**a
```
If you do the upcasting, you can no longer use the results. This might be a contrived example, but whenever bit manipulation is involved, you almost certainly need integers. `arctan2` is a no-brainer, because the result is always a float.

mouryarahul, 2020-03-31:
> That might be OK in certain cases, and it can be fatal in others. E.g., if you need the remainder of the powers of 3, when they are divided by 256, you can do that as
>
> ```
> a = array(range(10), [...]

I agree with you. So, I think if when we implement any numpy equivalent function, we should try to be compatible as much as possible but keeping in mind the memory constraint.

v923z, 2020-04-01:
> I agree with you. So, I think if when we implement any numpy equivalent function, we should try to be compatible as much as possible but keeping in mind the memory constraint.

A complete implementation of the `**` operator is already in the https://github.com/v923z/micropython-ulab/tree/power branch. As soon as it is reviewed, i can merge it.

v923z, 2020-04-01:
Closing the issue: feature request should have their own threads.

closed 2020-04-01 by v923z

v923z, 2020-04-02:
> A minimalistic equivalent of numpy.copy can be useful in many cases. How do I perform in straight way: @v923z What do you suggest?

@mouryarahul see https://github.com/v923z/micropython-ulab/pull/80, and a short comment on the performance and subtleties in the docs: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#initialising-by-passing-arrays I hope this is a satisfactory solution.

mouryarahul, 2020-04-02:
> @mouryarahul see #80, and a short comment on the performance and subtleties in the docs: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#initialising-by-passing-arrays I hope this is a [...]

Great! I think this will serve the purpose of `numpy.copy`.