v923z/micropython-ulab #15: Array shapes where one dimension is length 1 not properly represented

nickovs, 2019-11-14:
Currently `ulab` fails to properly displays arrays where the length of one of the dimensions is 1. Furthermore it incorrectly represents the shape of arrays with only one dimension.

To see this consider the following code for regular Python and `numpy`:
```
>>> import numpy as np
>>> aa = np.array(range(3))
>>> aa, aa.shape
(array([0, 1, 2]), (3,))
>>> bb = np.array(range(3)).reshape((1,3))
>>> bb, bb.shape
(array([[0, 1, 2]]), (1, 3))
>>> cc = np.array(range(3)).reshape((3,1))
>>> cc, cc.shape
(array([[0],
       [1],
       [2]]), (3, 1))
```
Compare this with the equivalent code in Micropython with `ulab`:
```
>>> import ulab as np
>>> aa = np.array(range(3))
>>> aa, aa.shape()
(array([0.0, 1.0, 2.0], dtype=float), (1, 3))
>>> bb = np.array(range(3)).reshape((1,3))
>>> bb, bb.shape()
(array([0.0, 1.0, 2.0], dtype=float), (1, 3))
>>> cc = np.array(range(3)).reshape((3,1))
>>> cc, cc.shape()
(array([0.0, 1.0, 2.0], dtype=float), (3, 1))
```
In the `aa` case the shape is misrepresented as being two dimensional even though it is printed as one dimensional. In the `bb` and `cc` cases the shape is correctly displayed but the array is printed as if it were one dimensional.

Two dimensional arrays where the length of both dimensions is greater than one are printed correctly and have the shape correctly represented.

v923z, 2019-11-14:
I note the bug, but I assigned a low-priority label to it.

v923z, 2020-10-30:
This issue is fixed in the latest version.

closed 2020-10-30 by v923z