v923z/micropython-ulab #16: Incorrect results from ulab.dot() in ESP32

nickovs, 2019-11-14:
When running `ulab` on the ESP32 the dot product function `ulab.dot()` yields erroneous results when the inputs are non-square. For instance:
```
>>> import ulab as np
>>> aa = np.array(range(6)).reshape((2,3))
>>> bb = np.array(range(6)).reshape((3,2))
>>> np.dot(aa,bb)
array([[2.0, 3.0],
         [8.0, 15.0]], dtype=float)
>>> np.dot(bb,aa)
array([[3.0, 4.0, 0.0],
         [9.0, 14.0, 0.0],
         [0.0, 0.0, 0.0]], dtype=float)
```
While regular binary operators seem to work fine the dot product function behaves very erratically. The fact that it seems to work correctly for square arrays suggest that the problem is likely some transposition of the dimension lengths.

v923z, 2019-11-14:
Now, this is clearly wrong. I will find out what the problem is.

v923z, 2019-11-28:
Nicko, I think I have found the indexing error, and this is fixed in the latest master. I tested it on the unix port, but I assume that it was not port specific.

closed 2019-12-30 by v923z