v923z/micropython-ulab #17: Subscripting 2-D array returns an array not a single value

nickovs, 2019-11-14:
When subscripting a 1-D array with an integer a single value is returned but when subscripting a 2-D array with two integers the result is a 1x1 array containing the value. For example:
```
>>> a = np.array(range(6))
>>> a[4]
4.0
>>> b = np.array(range(6)).reshape((2,3))
>>> b[1,1]
array([4.0], dtype=float)
>>> _.shape()
(1, 1)
```

Note that in `numpy` the behaviour of subscripting is simple: dimensions that are indexed with a single value are removed from the result while dimensions that are indexed with a slice or which are not index at all (i.e. there are fewer indices than dimensions) remain. Thus when indexing an array `b` with 2 dimensions `b[0,0]` returns a single value, `b[0,0:1]` returns a 1-D result containing that single value and `b[0:1,0:1]` returns a 2-D array containing that value. Ideally `ulab` should do the same.

Note that if the design for N-dimensional arrays discussed in #13 is implemented then this will clearly require the implementation of a more generic subscripting system anyway.

v923z, 2019-11-14:
I know about this, and it is caused by how scalar indices are treated: internally, they are turned into a slice of length one. I could reduce the complexity of the code that way, but I should have caught the output before returning from the function. I am not sure it is worth the trouble now. I hope I can roll out the other implementation sometime during the weekend.

mouryarahul, 2020-04-13:
@v923z Again, I'm not sure if this is intentional:
```
MicroPython v1.12-357-g740946736-dirty on 2020-04-12; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.

>>> a = ulab.zeros((3,3))
>>> a[0,0]
array([0.0], dtype=float)
>>> a[0][0]
0.0
>>> a[0][0] = 1.0
>>> a
array([[0.0, 0.0, 0.0],
         [0.0, 0.0, 0.0],
         [0.0, 0.0, 0.0]], dtype=float)
>>> a[0,0] = 1.0
>>> a
array([[1.0, 0.0, 0.0],
         [0.0, 0.0, 0.0],
         [0.0, 0.0, 0.0]], dtype=float)
```
I see that you've put this issue at "low priority", but I guess this should be considered as a "serious" bug. This has created a bit confusion when I'm trying to use ULAB.

v923z, 2020-04-13:
> @v923z Again, I'm not sure if this is intentional:

No, it is not, and this is why it is registered as an issue. 

> I see that you've put this issue at "low priority", but I guess this should be considered as a "serious" bug. This has created a bit confusion when I'm trying to use ULAB.

I certainly understand your frustration, but you have to keep in mind that `ulab` is an open-source project that we develop in our _free_ time. We fix bugs as our time permits, and unfortunately, we can't overcome the 24-hours-per-day limitation that everyone has. You can't expect that everything will work right away. I find such demands a bit unfair, especially that all your other issues have been solved within days, sometimes within hours.

On the other hand, we never discouraged anyone from contributing, in fact, I pointed out quite a few times that I would like to see more community involvement. If you can implement a fix, please, do that by all means, and I would be more than happy to see the contributor list grow. Till then you have to accept that so many people can achieve only so much.

mouryarahul, 2020-04-13:
> No, it is not, and this is why it is registered as an issue.
>
>
> I certainly understand your frustration, but you have to keep in mind that `ulab` is an open-source project that we develop in our [...]

Please don't take my comments as a "demand". I do understand that `ulab` is an open-source project and you devote quite a time into it. This is highly appreciable work you've been doing.
> On the other hand, we never discouraged anyone from contributing, in fact, I pointed out quite a few times that I would like to see more community involvement. If you can implement a fix, please, do [...]

I'm very much interested to contribute to the project. In fact, I've been going through your other repo `micropython-usermod` user guide. But, I guess I will take some more time to contribute to the codes of `ulab`.
I see that in the the `ulab`  [documentation](https://micropython-ulab.readthedocs.io/en/latest/), you've described in detail the motivation and the purpose of the `ulab`. I guess that you might have already defined/written a loose guidelines/specifications about the scopes and functionalities of `ulab` (I do not recall the specific word for it, but I guess you got me). I think we can be bit more discrete on it and maybe I can initially contribute to writing such documentation. I can start by listing the most frequently used functions with their input/output specifications. What do you say?

v923z, 2020-04-13:
> 
> Please don't take my comments as a "demand". I do understand that `ulab` is an open-source project and you devote quite a time into it. This is highly appreciable work you've been doing.

No offence was taken, I simply wanted to explain, why things don't go as fast as you (or I, for that matter) would like.

> I'm very much interested to contribute to the project. In fact, I've been going through your other repo `micropython-usermod` user guide. But, I guess I will take some more time to contribute to the codes of `ulab`.

Let me know, if you bump into a particular difficulty! I would actually think that contributing to `ulab` is probably easier than going through the user guide, because `ulab` contributions concern implementations of mathematical algorithms, and you don't have to care too much about working across the C-python interface.

> I see that in the the `ulab` [documentation](https://micropython-ulab.readthedocs.io/en/latest/), you've described in detail the motivation and the purpose of the `ulab`. I guess that you might have [...]

I don't quite know what you mean: the documentation is quite detailed, and lists all functions, with their inputs and outputs, benchmarks etc. E.g., here are the numerical functions: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#numerical . It might very well be that the documentation is not complete, even more probable that it is erroneous. Jeff Epler at `circuitpython` has written a rather thorough manual: https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/__init__.html . I don't mean to dissuade you, but I was wondering, whether you have, perhaps, missed these. 

> I think we can be bit more discrete on it and maybe I can initially contribute to writing such documentation. I can start by listing the most frequently used functions with their input/output specifications. What do you say?

I have written the manual as a jupyter notebook: https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-manual.ipynb, because in the notebook, code can easily be executed as part of the documentation. This can be done in shpinx with relatively little effort. All I want to say is that I do not want to detach code execution from the documentation, so you should keep that in mind. You can either work in the notebook, or you would have to write the sphinx script.

mouryarahul, 2020-04-13:
> No offence was taken, I simply wanted to explain, why things don't go as fast as you (or I, for that matter) would like.
>
>
> Let me know, if you bump into a particular difficulty! I would actually [...]

I mean a document something like [NumPy Reference](https://numpy.org/doc/1.18/reference/index.html). At the moment, I don't how it's produced/maintained and I've never use sphinx too. But, such a document will be very handy for the `ulab` contributors such as yourself. I would suggest that if we want `ulab` be easier (useful) for the users, we must implement it with exactly the same input/output behaviours of the `Numpy.` By this, I don't mean that we have to implement all complex functions available in `Numpy`. At the moment, I'm not very familiar with many aspects of `micropython` implementations, so can't guess how difficult and time consuming it could to implement `ulab` behaving very similar to `Numpy`.

> I have written the manual as a jupyter notebook: https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-manual.ipynb, because in the notebook, code can easily be executed as part of the [...]

v923z, 2020-04-14:
Since the topic of this conversation diverted quite considerably from the original subject, I opened a new ticked in https://github.com/v923z/micropython-ulab/issues/95

v923z, 2020-10-30:
This issue is fixed in the latest version.

closed 2020-10-30 by v923z