v923z/micropython-ulab #18: Epsilon double promotion warning/error.

iabdalkader, 2019-11-14:
`MICROPY_FLOAT_IMPL_FLOAT` turns on `-Wdouble-promotion` and `-fsingle-precision-constant` which causes epsilon literal to be float and an implicit conversion from 'float' to 'double' (to match fabs) warning/error. Maybe fabsf should be used instead...

```
QSTR not updated
CC ../../extmod/ulab/code/linalg.c
../../extmod/ulab/code/linalg.c: In function 'linalg_invert_matrix':
../../extmod/ulab/code/linalg.c:127:32: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
         if(fabs(data[m*(N+1)]) < epsilon) {
                                ^
../../extmod/ulab/code/linalg.c: In function 'linalg_det':
../../extmod/ulab/code/linalg.c:310:35: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
         if(fabs(tmp[m*(in->n+1)]) < epsilon) {
                                   ^
../../extmod/ulab/code/linalg.c: In function 'linalg_eig':
../../extmod/ulab/code/linalg.c:349:24: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
             if(epsilon < fabs(array[m*in->n + n] - array[n*in->n + m])) {
                        ^
cc1: all warnings being treated as errors
```

v923z, 2019-11-14:
I think this has been fixed in the testing branch. (All occurrences of fabs have been replaced by MICROPY_FLOAT_C_FUN(fabs).) Can you, please, compile against testing?

iabdalkader, 2019-11-14:
Yes this particular issue is fixed in testing, but in the testing branch numerical.c has other issues:
```
CC ../../extmod/ulab/code/vectorise.c
../../extmod/ulab/code/numerical.c: In function 'numerical_sum_mean_ndarray':
../../extmod/ulab/code/numerical.c:135:5: error: implicit declaration of function 'printf' [-Werror=implicit-function-declaration]
     printf("%ld, %ld\n", m, n);
     ^~~~~~
../../extmod/ulab/code/numerical.c:135:5: error: incompatible implicit declaration of built-in function 'printf' [-Werror]
../../extmod/ulab/code/numerical.c:135:5: note: include '<stdio.h>' or provide a declaration of 'printf'
../../extmod/ulab/code/numerical.c:135:15: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'size_t {aka unsigned int}' [-Werror=format=]
     printf("%ld, %ld\n", m, n);
             ~~^
             %d
../../extmod/ulab/code/numerical.c:135:20: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'size_t {aka unsigned int}' [-Werror=format=]
     printf("%ld, %ld\n", m, n);
                  ~~^
                  %d
```
Including stdio.h and using %zu fixes it.

Can the array fixes be merged in master ? I don't want to have to fork and maintain this repo.

v923z, 2019-11-14:
This printout was there for debugging purposes, and is not needed at all. I have removed it, and pushed the fix to master. Let me know if it still doesn't work. (I could compile it for the unix port.)

iabdalkader, 2019-11-15:
It builds now, thanks for fixing it!

closed 2019-11-15 by iabdalkader