v923z/micropython-ulab #20: multiple definition ulab_ndarray_type

Neutree, 2019-12-05:
Great work!

I want to add your ulab to [MaixPy](https://github.com/sipeed/MaixPy) as a submodule,  and I come across two problems,

* variable multiple defined, maybe should change to ` extern const mp_obj_type_t ulab_ndarray_type; `

https://github.com/v923z/micropython-ulab/blob/107587de0077b96bc538ded177ae63cb655da3c3/code/ndarray.h#L27

https://github.com/v923z/micropython-ulab/blob/107587de0077b96bc538ded177ae63cb655da3c3/code/ulab.c#L110-L120

* maybe should include `py/obj.h` in header files like `fft.h`

v923z, 2019-12-05:
> I want to add your ulab to [MaixPy](https://github.com/sipeed/MaixPy) as a submodule, and I come across two problems,

I am glad to hear that, thanks!

> 
>     * variable multiple defined, maybe should change to `extern const mp_obj_type_t ulab_ndarray_type;`

Does that mean that you can't compile the code for the maixpy boards?

> https://github.com/v923z/micropython-ulab/blob/107587de0077b96bc538ded177ae63cb655da3c3/code/ulab.c#L110-L120
> 
>     * maybe should include `py/obj.h` in header files like `fft.h`

All c files (e.g. fft.c) include py/obj.h. Since my headers contain not much more than the declaration of the functions, I didn't see too much point in including py/obj.h and similar there. I don't have strong feelings about this, so if you say that this makes your life difficult, I could move those includes into the header files. But is that really the case? 

By the way, I am close to significantly improving on ulab. It will be much more numpy-like, and will support tensors of arbitrary rank. If you feel adventurous, you can try it by compiling the `ndim` branch. (Not everything is implemented, so it is not good for production yet.) What I wanted to say with that, amongst other things, is that it might be a good time to point out issues like above, so that it can already be included in the next version.

Neutree, 2019-12-06:
Include "py/obj.h" in headers or not is not important for it is included before `fft.h` in c files.

But with definition `const mp_obj_type_t ulab_ndarray_type;` in header file can cause compile error( multiple definition) when multiple c files include it,  I changed this declaration to `extern const mp_obj_type_t ulab_ndarray_type;` and it compile passed. And I'm wondering why it compile pass in ESP32...maybe the compiler's difference

v923z, 2019-12-06:
> But with definition `const mp_obj_type_t ulab_ndarray_type;` in header file can cause compile error( multiple definition) when multiple c files include it, I changed this declaration to `extern [...]

OK, I can change that, no problem. And thanks for bringing this up!

v923z, 2019-12-31:
Fixed in https://github.com/v923z/micropython-ulab/commit/81acfc7da8cb83dc7db018411c6fb97c82fdb7de.

closed 2019-12-31 by v923z