v923z/micropython-ulab #24: ulab assumes that floats are boxed: MICROPY_OBJ_REPR_A or B

dhalbert, 2020-01-04:
ulab is assuming that MicroPython is built with `MICROPY_OBJ_REPR_A` or `MICROPY_OBJ_REPR_B`, which store floats as boxed objects instead of as immediate values like smallints. In the code below, the float object is assumed to be boxed:

https://github.com/v923z/micropython-ulab/blob/81acfc7da8cb83dc7db018411c6fb97c82fdb7de/code/ulab.c#L29-L34

v923z, 2020-01-04:
Since that is simply a version number, we can definitely change that. What do you suggest? A string, or integers as version major/minor? We could also define `__version__` as an attribute, and hook a function into it, and then we can return whatever we need.

dhalbert, 2020-01-04:
Oops, just answered the same question here: https://github.com/v923z/micropython-ulab/issues/21#issuecomment-570822569

>I would suggest making the version a string, not a float, to have control over the formatting. E.g., is it "1.10" or "1.1"? And if one wanted to use semver-style, x.x.x, then it can't be a float.

v923z, 2020-02-07:
@dhalbert Since this issue has been address somewhere else, I am closing it now.

closed 2020-02-07 by v923z