v923z/micropython-ulab #25: Issue with np.mean

lhargrove, 2020-01-06:
This is a great package, and I was working through the examples provided in the documentation. I had some trouble replicating the np.mean function. (See output below). I am working on the SF6 hardware on Microptyhon v1.12-43-g54a2584de-dirty  using ulab version 0.263 

`import ulab as np`
`a = np.array([[1, 2, 3], [4, 5, 6], [7, 8, 9]])`
`print('a: \n', a)`

a:
 array([[1.0, 2.0, 3.0],
         [4.0, 5.0, 6.0],
         [7.0, 8.0, 9.0]], dtype=float)

`print('sum, flat array: ', np.sum(a))`

sum, flat array:  45.0

`print('mean, horizontal: ', np.mean(a, axis=1))`

mean, horizontal:  array([2048.00048828125, 5.387879937997763e-315, 0.0], dtype=float)

`print('std, vertical: ', np.std(a, axis=0))`

std, vertical:  array([7.191838266139812, 5.314304082212256e-315, 0.0], dtype=float)

v923z, 2020-01-07:
Thanks for reporting the issue, I will look into it.

v923z, 2020-01-07:
I have pushed a fix to master. Could you, please, check if this solves the issue? Thanks!

lhargrove, 2020-01-07:
That fixed the issue for me. Thanks so much for the quick reply.

v923z, 2020-01-09:
OK, glad to hear that. Actually, it has been there in the testing branch for almost two months, but somehow I have forgotten to merge it. Closing now.

closed 2020-01-09 by v923z