v923z/micropython-ulab #26: Can't create 2d np.arrays with only 1 element

Aggropatics, 2020-01-13:
np.array([[1,2]]) should return a two dimensional array with only one element. Accessing said array with array[0] should return [1,2] and it should be passable to arguments such as argsort() which takes in an ndarray. I have not been able to create such an array with ulab. Using reshape() doesn't affect the issue. 

Examples: 

ulab
>>> import ulab as np
>>> arr = [[1,2]]
>>> arr2 = np.array(arr)
>>> arr2[0]
1.0

numpy
import numpy as np
arr = [[1,2]]
arr2 = np.array(arr)
arr2[0]
Out[1]: array([1, 2])

v923z, 2020-01-13:
Thanks for reporting the issue! I think I know what the problem is. I will try to fix it in the next couple of days.

Aggropatics, 2020-01-14:
Thank you for the quick response!

v923z, 2020-01-17:
On second thought, I don't think that this issue can be rectified in the master branch. The `ndim` contains the skeleton of an implementation that can deal with these cases. You can definitely try that out, but I am not sure it can do everything you need. I need a bit of time to finish that.

Aggropatics, 2020-01-20:
I'll try that, thank you. Hope you can get around to this! Having these features for the micro controller world is a game changer.

v923z, 2020-01-20:
> I'll try that, thank you. Hope you can get around to this! 

Actually, I am not that far, I feel. One comment: at the moment, you have to initialise the array from a linear iterable, and then you have to reshape it.

v923z, 2020-10-30:
This issue is fixed in the latest version.

closed 2020-10-30 by v923z