v923z/micropython-ulab #27: Compiling for Nucleo-L476RG

spluttflob, 2020-01-18:
Some changes had to be made to compile ulab into MicroPython v1.12-93 for the Nucleo-L476RG.  After the following changes were made, everything seems to work well.  NOTE: I'm new to building custom C modules for MicroPython, and these problems may have been caused by my doing something (or many somethings) incorrectly. 

* `mp_const_none_obj`, `mp_const_true_obj`, and `mp_const_false_obj` were unrecognized. Replacing macros such as `MP_ROM_PTR(&mp_const_none_obj)` with corresponding macros `MP_ROM_NONE`, `MP_ROM_TRUE`, and `MP_ROM_FALSE` allowed successful compilation.

* As I intend to compile other external C code as well as ulab, it is appropriate to place the ulab directory one level _under_ the main external C code directory, which I've placed inside the root MicroPython directory (is that a _faux pas?_). The ulab source files are then only found if moved from the `ulab/code` subdirectory to the main `ulab` directory.  A workaround which seems to work is to move `micropython.mk` to the main `ulab` directory, leave the source files in `ulab/code`, and edit `micropython.mk` to specify the location of the source and header files as follows: 
```
USERMODULES_DIR := $(USERMOD_DIR)

# Add all C files to SRC_USERMOD.
SRC_USERMOD += $(USERMODULES_DIR)/code/ndarray.c
SRC_USERMOD += $(USERMODULES_DIR)/code/linalg.c
SRC_USERMOD += $(USERMODULES_DIR)/code/vectorise.c
SRC_USERMOD += $(USERMODULES_DIR)/code/poly.c
SRC_USERMOD += $(USERMODULES_DIR)/code/fft.c
SRC_USERMOD += $(USERMODULES_DIR)/code/numerical.c
SRC_USERMOD += $(USERMODULES_DIR)/code/ulab.c

# We can add our module folder to include paths if needed
# This is not actually needed in this example.
CFLAGS_USERMOD += -I$(USERMODULES_DIR)/code
```

Instructions from the ulab README as well as [the MicroPython external C modules page](https://docs.micropython.org/en/latest/develop/cmodules.html) were used to duct tape this solution together. I suspect that the directory problems are my own doing, but I'm not sure about the compiler not recognizing `mp_const_none_obj`, etc.

Happy to supply more details, diffs, patches, or whatever...as soon as I've gone and learned how to generate them correctly.

v923z, 2020-01-20:
First, many thanks for reporting the issue. I really appreciate that you also suggested a fix. 

> Some changes had to be made to compile ulab into MicroPython v1.12-93 for the Nucleo-L476RG. After the following changes were made, everything seems to work well. NOTE: I'm new to building custom C [...]

I have updated the code, and pushed the changes to master. 

> 
>     * As I intend to compile other external C code as well as ulab, it is appropriate to place the ulab directory one level _under_ the main external C code directory, which I've placed inside the root MicroPython directory (is that a _faux pas?_). 

Yes, it is. User modules can now be completely be decoupled from micropython itself, and I think, this is good so: https://micropython-usermod.readthedocs.io/en/latest/usermods_03.html 

The ulab source files are then only found if moved from the `ulab/code` subdirectory to the main `ulab` directory.  A workaround which seems to work is to move `micropython.mk` to the main `ulab` directory, leave the source files in `ulab/code`, and edit `micropython.mk` to specify the location of the source and header files as follows:
> ```
> USERMODULES_DIR := $(USERMOD_DIR)
>
> # Add all C files to SRC_USERMOD.
> SRC_USERMOD += $(USERMODULES_DIR)/code/ndarray.c
> SRC_USERMOD += $(USERMODULES_DIR)/code/linalg.c
> SRC_USERMOD += [...]

If you place your module in the correct place, you don't have to modify the make fragment. 

> Happy to supply more details, diffs, patches, or whatever...as soon as I've gone and learned how to generate them correctly.

Like so: https://github.com/v923z/micropython-ulab/issues/6#issuecomment-549508377
Bonus: if you follow that route, you get credit for your corrections to the code.

Again, thanks for your comments. If you can compile the new version straight away (without having to modify anything), you can close the issue.

spluttflob, 2020-01-20:
I am pleased to report that after downloading a fresh copy of MicroPython and ulab and following your suggestions for correcting my directory structure, the code compiles and runs on the Nucleo-L476RG, Nucleo-F411RE, and Nucleo-F446RE without modification.  Many thanks for the assistance.

closed 2020-01-20 by spluttflob

v923z, 2020-01-21:
> I am pleased to report that after downloading a fresh copy of MicroPython and ulab and following your suggestions for correcting my directory structure, the code compiles and runs on the Nucleo-L476RG, Nucleo-F411RE, and Nucleo-F446RE without modification. Many thanks for the assistance.

Fantastic! Thanks for the feedback!