v923z/micropython-ulab #28: Issue with dot on PYBD_SF6

lhargrove, 2020-01-26:
I am having an issue running np.dot(). I am running on PYBD-SF6 using version 0.26.5. It appears that a row matrix times a square matrix works fine. However, when I use dot with a square matrix times a column matrix, the results aren't as expected. It appears to only provide the first element in the expected result  (14,32,50)

`MicroPython v1.12-92-gd89ed3e62-dirty on 2020-01-26; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import ulab as np
>>> print('you are running ulab version', np.__version__)
you are running ulab version 0.26.5
>>> a = np.array(([1,2,3],[4,5,6],[7,8,9]))
>>> b = np.array(([1,2,3]))
>>> b.transpose()
>>> c = np.dot(a,b)
>>> c
array([14.0, 0.0, 0.0], dtype=float)
>>>
>>>
`

v923z, 2020-01-27:
Thanks for reporting the issue, I will look into it. I am wondering, though, how you printed out the version number: that should be `np.__version__`: https://github.com/v923z/micropython-ulab/blob/b2c47499624b5b58c741c6e83098da06f119d752/code/ulab.c#L117

lhargrove, 2020-01-27:
That was a cut and paste error on my part, sorry about that.

v923z, 2020-01-29:
I have found the error, and corrected the error. I hope it is OK now.

lhargrove, 2020-02-15:
Yes, everything seems to be working fine now. Thanks again for this excellent package!

closed 2020-02-15 by lhargrove

v923z, 2020-02-15:
Thanks for the feedback!