v923z/micropython-ulab #30: Crash (wrong argument types not handled?)

jepler, 2020-02-06:
```MicroPython v1.12-139-gdbed8f576-dirty on 2020-02-06; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab
>>> ulab.dot(ulab.zeros(3), 0)
Segmentation fault
```

I've encountered a number of these; I think it's where both arguments need to be ndarrays, but it's not caught as an exception but leads to segmentation fault / hardfault instead.  Do you want them reported separately as I encounter them?  (this is the only one I wrote down so far)

v923z, 2020-02-07:
> ```
> Use Ctrl-D to exit, Ctrl-E for paste mode
> Segmentation fault
> ```
>
> I've encountered a number of these; I think it's where both arguments need to be ndarrays, but it's not caught as an exception [...]

No, I think the issue is clear, I will look into it. And you are correct, the pointer is converted to an `ndarray_obj_t` without any considerations in the first two lines of https://github.com/v923z/micropython-ulab/blob/0315408e055a991854c53766b36954d603d9ca80/code/linalg.c#L186 . I think this is an easy one.

v923z, 2020-02-07:
@jepler  Jeff, the code of the latest commit bails gracefully out now, if you try to dot improper types. Let me know if this fixes the issue.

closed 2020-03-03 by v923z