v923z/micropython-ulab #32: Slicing difference vs numpy

jepler, 2020-02-08:
It seems that too few elements are in the result.

```>>> import numpy as np
>>> np.array([0,1,2,3,4])[::2]
array([0, 2, 4])
```
vs
```
>>> ulab.array([0,1,2,3,4])[::2]
array([0.0, 2.0], dtype=float)
```
Let me know if you'd like me to look into this.

v923z, 2020-02-09:
> 
> Let me know if you'd like me to look into this.

Thanks for reporting the issue, and the offer. I have located the error, and fixed it in 936bb3b, https://github.com/v923z/micropython-ulab/blob/936bb3bae55dfdc81d1c8d9580905ae0ac760a44/code/ndarray.c#L218

You can close the issue, if you are satisfied with the solution.

jepler, 2020-02-11:
Hi!  thanks for fixing the issue I initially reported.  I think there are still slicing differences, though.  Here's a more extensive test program.  While it tests a lot of cases, it finishes in way under a second on unix  micropython.  When run under python3, it doesn't print any differences between list and array indexing.  Furthermore, I verified that micropython and python3 both slice the lists the same.

```
try:
    import ulab as np
except:
    import numpy as np

for num in range(1,4):
    l = list(range(num))
    a = np.array(l, dtype=np.int8)
    for start in range(-num, num+1):
        for end in range(-num, num+1):
            for stride in (-3, -2, -1, 1, 2, 3):
                sl = l[start:end:stride]
                ll = len(sl)
                try:
                    sa = list(a[start:end:stride])
                    la = len(sa)
                except IndexError as e:
                    sa = la = None
                if not (ll == la or (la is None and ll is 0)):
                    print("%2d [% d:% d:% d]     %-24r %-24r" % (
                        num, start, end, stride, sl, sa))
```

Each line of output shows a case which is different between arrays and lists.  The case where the list is empty and the result of slicing the array is an IndexError is treated as okay, since it looks like it might be a deliberate decision.

The cases my program found all affect negative strides.  There are cases where ndarray slicing returns an element but list slicing doesn't, such as
```
 2 [-1: 1:-3]     []                       [1]                     
```
cases where an element should have been returned, but an exception was raised
```
 2 [ 1:-2:-1]     [1]                      None                    
```
and cases where the wrong number of elements were returned
```
 3 [-1:-3:-1]     [2, 1]                   [2]                     
```

v923z, 2020-02-11:
Uh, then it is a bit more involving than initially thought. I will check this.

commit 380b8b03 (2020-03-01): Add new test of slicing

Closes: #32
closed 2020-03-01 by v923z