v923z/micropython-ulab #43: add cross product

v923z, 2020-02-25:
Add the cross product function https://docs.scipy.org/doc/numpy/reference/generated/numpy.cross.html

rcolistete, 2020-07-18:
Thumbs up to have 'ulab.linalg.cross()'.

v923z, 2020-07-19:
I haven't touched this, because the cross-product is really nasty. It is a lot of case-checking, and I am not sure the speed gain is worth the trouble: https://numpy.org/doc/stable/reference/generated/numpy.cross.html. I am more inclined to yank this now. If someone need it, it should probably be implemented in python.

rcolistete, 2020-07-19:
[Perfect is the enemy of good](https://en.wikipedia.org/wiki/Perfect_is_the_enemy_of_good)...

My suggestion is to implement only some features of 'numpy.cross', the most useful. For example :
- only vector cross-product;
- only vectors with the same size (2 or 3, as numpy.cross);
- so no 'axisa', 'axisb', 'axisc', 'axis' arguments, accepting only 2 arguments : a, b;
- compatible with 1st, 3rd and 4th examples from [numpy.cross reference](https://docs.scipy.org/doc/numpy/reference/generated/numpy.cross.html) would run.

v923z, 2020-07-19:
> [Perfect is the enemy of good](https://en.wikipedia.org/wiki/Perfect_is_the_enemy_of_good)...

That's right, but we have limited resources... (in time, RAM, and flash).

> My suggestion is to implement only some features of 'numpy.cross', the most useful. For example :
>
>     * only vector cross-product;
>
>     * only vectors with the same size (2 or 3, as [...]

This could be a reasonable compromise. Do you want to try your hand at it?

rcolistete, 2020-07-19:
> This could be a reasonable compromise. Do you want to try your hand at it?

Not at the moment. I need to practice writing natice C modules for MicroPython, with simple functions of 1 argument, etc. Then I could contribute with code to ulab.

v923z, 2020-07-19:
OK. Thanks for the feedback! You might find hints under https://micropython-usermod.readthedocs.io .

rcolistete, 2020-07-21:
I've made a simple version of 'linalg.cross(a, b)'. See the [commit 'cross : implemented linalg.cross(a, b)'](https://github.com/rcolistete/micropython-ulab/commit/5e37ec37220eea775e3d6aa96951de51275ea2fd) in my ulab fork. With :
* only vector cross-product;
* only vectors with the same size/dimension, 3;
* no 'axisa', 'axisb', 'axisc', 'axis' arguments, accepting only 2 arguments : a, b;
* output is always float, with no upcasting.

Any suggestion ?

If it is ok, I can make a PR.

v923z, 2020-07-22:
> 
> 
> I've made a simple version of 'linalg.cross(a, b)'. See the [commit 'cross : implemented linalg.cross(a, b)'](https://github.com/rcolistete/micropython-ulab/commit/5e37ec37220eea775e3d6aa96951de51275ea2fd) in the 'cross' branch of my ulab fork. With :
> 
>     * only vector cross-product;
> 
>     * only vectors with the same size/dimension, 3;
> 
>     * no 'axisa', 'axisb', 'axisc', 'axis' arguments, accepting only 2 arguments : a, b;
> 
>     * output is always float, with no upcasting.
> 
> 
> Any suggestion ?

I would have a couple: 

1. I presume, speed is not an issue here (there only 3 elements in an array), so I would call the iterator function. If you do that, the following will be valid:

```
a = ulab.array([1, 2, 3])
b = ulab.array([4, 5, 6])
cross(a, b)
cross([1, 2, 3], [4, 5, 6])
```

2. I would accept 2-component vectors/iterables, so that 
```
a = ulab.array([1, 2])
b = ulab.array([4, 5, 6])
cross(a, b)
```
doesn't throw an error.

3. When adding new function(s), don't forget to increment the version number in `ulab.c`, and insert an entry in https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-change-log.md

4. Don't forget to add your name in the copyright note in `linalg.c`.

v923z, 2020-07-24:
@rcolistete Do you want to work this out, or should I?

rcolistete, 2020-07-24:
I will.
I am also spending time building MicroPython firmwares with ulab on Pyboard's (23 variants) and ESP8266.

v923z, 2020-07-24:
OK, thanks for the feedback!

v923z, 2020-10-30:
@rcolistete `cross` has been added to the latest version. Closing the issue now.

closed 2020-10-30 by v923z