v923z/micropython-ulab #51: No need to define MODULE_ULAB_ENABLED in mpconfigport.h

codemee, 2020-03-01:
Since the micropython.mk file has added the following line

```makefile
CFLAGS_EXTRA = -DMODULE_ULAB_ENABLED=1
```

There's no need to add `#define MODULE_ULAB_ENABLED (1)` in the mpconfigport.h.Or it would make redefined errors while compiling code.

v923z, 2020-03-01:
Could you, please, point to the error? I thought that this was in the manual, but I don't find it.

codemee, 2020-03-02:
In the compiling section in README.MD, it says

> Then you have to include ulab in the compilation process by editing mpconfigport.h of the directory of the port for which you want to compile, so, still on the command line, navigate to micropython/ports/unix, or micropython/ports/stm32, or whichever port is your favourite, and edit the mpconfigport.h file there. All you have to do is add a single line at the end:
> 
> #define MODULE_ULAB_ENABLED (1)
> 
> This line will inform the compiler that you want ulab in the resulting firmware.

v923z, 2020-03-03:
I will close the issue, since you have contributed a fix. Thanks again!

closed 2020-03-03 by v923z

codemee, 2020-03-04:
No problems.Thanks.