v923z/micropython-ulab #58: AttributeErrors when calling np.mean(), np.std(), etc.

lhargrove, 2020-03-10:
I am working with version 0.36.0 and I don't seem to be able to access np.mean(), np.std(), np.min(), np.max() and other functions. There seems to be an "AttributeError" 'module' object has no attribute 'mean' "

v923z, 2020-03-10:
You have to call them from `numerical`, like so

```import ulab
from ulab import numerical

numerical.mean([1, 2, 3])
```
Further examples can be found under https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/__init__.html. 

We have factored out almost all functions into sub-modules, so that the size of the firmware can be reduced: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#customising-ulab. 
I still have to update the manual, though.

closed 2020-03-10 by jepler

v923z, 2020-03-10:
@lhargrove I have updated the documentation, it should be OK now.