v923z/micropython-ulab #64: indexing by boolean list is slow

Marius-450, 2020-03-13:
Trying to optimize an inner loop, I've tried to limit the computations by indexing an array with a boolean list. unfortunately, it take waaaay more time for each iteration.
example code on circuitpython : 
```python
import ulab
import time

a = ulab.ones(1000)
b = [False]*1000
b[5:10] = [True]*5

start = time.monotonic_ns()
a = a * 2
print('time for a*2           : ', time.monotonic_ns() - start )

start = time.monotonic_ns()
a[5:10] = a[5:10] * 2
print('time for a[5:10]*2     : ', time.monotonic_ns() - start )

start = time.monotonic_ns()
a[b] = a[b] * 2
print('time for a[b] = a[b]*2 :', time.monotonic_ns() - start )
```
output (timing in ns) :
```
time for a*2           :  423000
time for a[5:10]*2     :  123000
time for a[b] = a[b]*2 : 4745000
```
I am testing with : *Adafruit CircuitPython 5.0.0-37-geebe76997 on 2020-03-09; Adafruit Monster M4SK with samd51j19*

v923z, 2020-03-13:
If you just want to overwrite the entries `a[5:10]`, then why do you need a Boolean array? The problem is that a slice is just that, the elements between index 5, and 10. On the other hand, the Boolean index runs from 0 to 1000, and you have to inspect all elements, since you don't know, where the Trues are. I don't know how exactly this is done in `numpy`.

Marius-450, 2020-03-14:
This was a simple example to illustrate.
I calculate pixel values for a row at a time. I have a for loop where I record the number of iterations for each pixel before reaching a threshold. Boolean indexing was the solution here, but it's slower (by 3 to 5 times) than the naive non ulab number crushing method... (it is also useful to know when every pixel is calculated. It can save 200 iterations over 15000)
Maybe it would be faster using boolean dtype in a ulab array ?

v923z, 2020-03-14:
>  Boolean indexing was the solution here, but it's slower (by 3 to 5 times) than the naive non ulab number crushing method...

Are you saying that the python-only implementation is faster by a factor of 3 to 5? 

However, I should point out that the Boolean indexing will always be slower than slicing the same array, because you have to traverse the array at least twice: first to find out, whether the indexed array is compatible with the left hand side, and second for the actual computation. Since Boolean indices don't have an obvious structure, it is impossible to store the results of the first pass in an efficient way.

> Maybe it would be faster using boolean dtype in a ulab array ?

Yes, it would be, and it is already in the works. One of the branches has this, but a couple of components are still missing.

Marius-450, 2020-03-14:
yes. a complete mandelbrot set computation, with 256 iterations, takes around 120 secs with a python implementation (going around complex numbers operations was a very effective optimisation)
the better I can get with ulab is 177 sec, computing everything except one boolean slicing (to update only the "not finished" pixels)
I understand it is not a big gain to expect, but I was not expecting a x10 time for those operations...
I tested with numpy directly, and slicing with boolean array is fast, but using a list of boolean is quite slow. maybe I was just too impatient. I'll test further with this when it will be available.
Thanks :)

v923z, 2020-03-15:
> yes. a complete mandelbrot set computation, with 256 iterations, takes around 120 secs with a python implementation (going around complex numbers operations was a very effective optimisation)

Well, the Mandelbrot set doesn't easily lend itself to vectorisation, and a computation of that is slow even in numpy. I also have to admit that fractals are probably the last that would come to my mind, when thinking about ulab.

v923z, 2020-03-31:
@Marius-450 do you want to keep this issue alive?

v923z, 2020-04-06:
@Marius-450 In https://github.com/v923z/micropython-ulab/pull/84, operator testing is moved outside the for loop, hence it should be faster now.

v923z, 2020-04-12:
@Marius-450 I believe, this is all that I could do at this point, so I am closing the issue now.

closed 2020-04-12 by v923z