v923z/micropython-ulab #66: undefined reference to `roundf'

skylin008, 2020-03-20:
I had been download the latest verison ulab,when make it, it showed the error information as follow:

LINK build-PYBV11/firmware.elf
build-PYBV11/code/vectorise.o: In function `vectorise_around':
vectorise.c:(.text.vectorise_around+0x76): undefined reference to `roundf'
Makefile:597: recipe for target 'build-PYBV11/firmware.elf' failed
make: *** [build-PYBV11/firmware.elf] Error 1

How to solve this issue,Thanks!

v923z, 2020-03-20:
I have just found this error myself, and it turns out that it can easily be fixed, however, not in `ulab`, but in `micropython` itself. So, navigate to `/micropython/ports/stm32`, and there find 
```
 SRC_LIBM = $(addprefix lib/libm/,\
        math.c \
        acoshf.c \
        asinfacosf.c \
        asinhf.c \
```
in the `Makefile`. You should add `roundf.c` to the `SRC_LIBM` list like so:

```
 SRC_LIBM = $(addprefix lib/libm/,\
        math.c \
        roundf.c \
        acoshf.c \
        asinfacosf.c \
        asinhf.c \
```

I will try to push a fix upstream to `micropython`. Till then you have to insert `roundf.c` by hand. Sorry for the inconvenience.

skylin008, 2020-03-20:
OK,Thanks!

v923z, 2020-03-24:
@skylin008 the makefiles have been corrected in https://github.com/micropython/micropython/commit/c5cc64175be32cb1e4f3f1a249667bc9f5a12613. I think this should resolve the issue. You can close it, if it is so.

closed 2020-03-26 by skylin008