v923z/micropython-ulab #67: Indexing by Boolean List seems to malfunction

mouryarahul, 2020-03-28:
I'm not sure if I'm doing any mistake, but indexing using boolean list is not working as expected. See below the result:

MicroPython v1.12-167-gf020eac6a-dirty on 2020-03-06; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.

> import ulab as np
> a = np.array(range(9), dtype=np.float)
> print("a:\t", a)
a:       array([0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0], dtype=float)
> print("a < 5:\t", a[a < 5])
a < 5:   array([], dtype=float)
>

Is there any workaround? Thanks for your suggestions.

v923z, 2020-03-30:
> I'm not sure if I'm doing any mistake, but indexing using boolean list is not working as expected. See below the result:
>
> MicroPython v1.12-167-gf020eac6a-dirty on 2020-03-06; PYBD-SF6W with [...]

Thanks for reporting the issue! This indeed seems to be an error. I will try to find out what the problem is, and implement a fix.

v923z, 2020-03-30:
The fix is in master. You can close the issue, if it solved the problem.

closed 2020-03-30 by v923z