v923z/micropython-ulab #72: Ambiguity in Comparison operators

mouryarahul, 2020-04-01:
I was trying to implement numpy.linalg.norm in the pythonic way, e.g., L0-norm, and got this when doing:
`>>> a`
`array([1.2, 0.0, -1.3, 0.25, 1.45, -0.35, 2.0], dtype=float)`
`>>> a != 0.`
`True`
`>>> a[a != 0.]`
`Traceback (most recent call last):`
 ` File "<stdin>", line 1, in <module>`
`IndexError: too many indices `
I see that there isn't any mention of `!=` comparision in the documenation. Is this intensional?

At the moment, I'm achieving a bit inefficient way:
`>>> a = array([1.2, 0.0, -1.3, 0.25, 1.45, -0.35, 2.0], dtype=float)`
`>>> x = a > 0`
`>>> y = a < 0`
`>>> a[[x[i] or y[i] for i in range(len(x))]]`
`array([1.2, -1.3, 0.25, 1.45, -0.35, 2.0], dtype=float) `

@v923z Do you suggest some efficient way?

v923z, 2020-04-01:
> I was trying to implement numpy.linalg.norm in the pythonic way, e.g., L0-norm, and got this when doing:
> `>>> a`
> `array([1.2, 0.0, -1.3, 0.25, 1.45, -0.35, 2.0], dtype=float)`
> `>>> a != 0.`
> `True` [...]

I think this happens, because indices must be either slices, single integers, or a list of Booleans, and `a != 0` returns a single Boolean. But you hit the nail on the head, `!=` is not implemented, so I would suggest that we keep the issue open for now. 
Do you happen to know if in `numpy` the equality is implemented in the strict sense, or the values have to be only close?

mouryarahul, 2020-04-01:
> I think this happens, because indices must be either slices, single integers, or a list of Booleans, and `a != 0` returns a single Boolean. But you hit the nail on the head, `!=` is not implemented, [...]

@v923z Comparing to zero has always been tricky. I'm not sure, how do they implement the `==` or `!=` in Numpy or Python itself. But, can't we call the same function from the micropython and operate it elementwise to have similar behaviour. In fact, both numpy and math have a function like `isclose`, whose precision can be controlled by the user. Maybe we can also implement `isclose` function too using `math.isclose`?

v923z, 2020-04-02:
> @v923z Comparing to zero has always been tricky. I'm not sure, how do they implement the `==` or `!=` in Numpy or Python itself. But, can't we call the same function from the micropython and operate it elementwise to have similar behaviour. 

That will be slow: the whole idea behind `ulab` (and `numpy` for that matter) is that you detach _numerical_ operations from the type-agnostic (micro)python implementation, and that is how you gain speed. Also, (micro)python has no notion of the numeric C types, so you can't just iterate over an array, and call the (micro)python functions with the items. First you would have to convert the elements in the array to an `mp_obj_t` object, and only then can you compare. An at that point, your solution will be on par with the `!=` operator. So, is it worth the flash space?

> In fact, both numpy and math have a function like `isclose`, whose precision can be controlled by the user. Maybe we can also implement `isclose` function too using `math.isclose`?

Yes, that would be possible.

mouryarahul, 2020-04-02:
> > @v923z Comparing to zero has always been tricky. I'm not sure, how do they implement the `==` or `!=` in Numpy or Python itself. But, can't we call the same function from the micropython and operate it elementwise to have similar behaviour.
> 
> That will be slow: the whole idea behind `ulab` (and `numpy` for that matter) is that you detach _numerical_ operations from the type-agnostic (micro)python implementation, and that is how you gain speed. Also, (micro)python has no notion of the numeric C types, so you can't just iterate over an array, and call the (micro)python functions with the items. First you would have to convert the elements in the array to an `mp_obj_t` object, and only then can you compare. An at that point, your solution will be on par with the `!=` operator. So, is it worth the flash space?
> 
> > In fact, both numpy and math have a function like `isclose`, whose precision can be controlled by the user. Maybe we can also implement `isclose` function too using `math.isclose`?
> 
> Yes, that would be possible. I don't know offhand, how much that would cost, though...

I think Python and numpy.equal use the same operator from C, i.e.,  `==` and `!=`. I think we should also use the same to compatible with them. However, as I suggested, we should also provide the `isclose` function where users can choose their precision.

v923z, 2020-04-02:
> I think Python and numpy.equal use the same operator from C, i.e., `==` and `!=`.

But the question, what you operate on: if you compare two ndarrays, then you compare the objects, not the contents. On the other hand, what you will is comparison of contents.

> I think we should also use the same to compatible with them. However, as I suggested, we should also provide the `isclose` function where users can choose their precision.

Your point is taken. Should `isclose` be part of the core, or some module?

mouryarahul, 2020-04-02:
> But the question, what you operate on: if you compare two ndarrays, then you compare the objects, not the contents. On the other hand, what you will is comparison of contents.

I guess `numpy.equal` and `==`  compares the component of the array:
`>>> a = np.array([1.543645274878272])`                                                                                                                    

`>>> b = np.array([1.543645274878271])`                                                                                                                        

`>>> np.equal(a,b) `                                                                                                                                           

`array([False])`                                                                                                                                               

`>>> a == b`                                                                                                                                              

`array([False])`
> 
> > I think we should also use the same to compatible with them. However, as I suggested, we should also provide the `isclose` function where users can choose their precision.
> 
> Your point is taken. Should `isclose` be part of the core, or some module?

I think we can implement `==` and `!=` in the core and `isclose` in the numerical module.

v923z, 2020-04-03:
> I guess `numpy.equal` and `==` compares the component of the array:
> `>>> a = np.array([1.543645274878272])`
>
> `>>> b = np.array([1.543645274878271])`
>
> `>>> np.equal(a,b) `
>
> `array([False])`
>
> `>>> a [...]

I am not trying to re-implement `numpy`: that assumes practically unlimited resources that a microcontroller doesn't have. So, one has to be selective in which functions should be implemented. 

> I think we can implement `==` and `!=` in the core and `isclose` in the numerical module.

Again, all operations that involve two operands are expensive. You can either resolve the C types in both arrays separately, https://github.com/v923z/micropython-ulab/blob/034cdaf7e3ff50b8659fc462e8eb33f1812eaadd/code/ndarray.c#L806 (this takes a lot of flash space) or you do a type conversion with the help of a function https://github.com/v923z/micropython-ulab/blob/034cdaf7e3ff50b8659fc462e8eb33f1812eaadd/code/vectorise.c#L205 (which is slow). 

There is no other way. And this is why I am a bit reluctant to implement everything that comes to mind, without thinking about, whether it is necessary. We can't really beat your original solution 
```
sum([a < 0.]) + sum([a > 0.])
```
by too much. But I could probably add `isclose` to `numerical`.

mouryarahul, 2020-04-03:
Well, I do not agree totally with you. Operators such as `==` and `!=` are very basic in most of the programming languages and numerical libraries because there are many situations where you literally have to compare objects for equality/non-equality. 
I agree that binary operations are expensive and their implementation consume ROM/Flash memory, but if I've to achieve the same in pythonic way:
`[[a[i][j] < 0. for j in range(num_cols)]for i in range(num_rows)]` 
then it is quite slow and consumes a big chunk of RAM, which always in scarcity on uController. See  the comparison below:
```
MicroPython v1.12-326-g8fff0b0ac on 2020-04-03; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.

>>> import micropython
>>> micropython.mem_info()
stack: 540 out of 23552
GC: total: 386816, used: 1648, free: 385168
 No. of 1-blocks: 22, 2-blocks: 7, max blk sz: 40, max free sz: 24061
>>> import ulab as np
>>> micropython.mem_info()
stack: 540 out of 23552
GC: total: 386816, used: 1856, free: 384960
 No. of 1-blocks: 26, 2-blocks: 10, max blk sz: 40, max free sz: 24047
>>> a = np.array([[1,-2,3],[-4,5,6], [7,-8,9]])
>>> micropython.mem_info()
stack: 540 out of 23552
GC: total: 386816, used: 2368, free: 384448
 No. of 1-blocks: 39, 2-blocks: 12, max blk sz: 40, max free sz: 24017
>>> b = a < 0.
>>> micropython.mem_info()
stack: 540 out of 23552
GC: total: 386816, used: 2784, free: 384032
 No. of 1-blocks: 56, 2-blocks: 16, max blk sz: 40, max free sz: 23991
>>> c = [[a[i][j] < 0. for j in range(3)]for i in range(3)]
>>> micropython.mem_info()
stack: 540 out of 23552
GC: total: 386816, used: 4368, free: 382448
 No. of 1-blocks: 104, 2-blocks: 38, max blk sz: 40, max free sz: 23892
```
See the difference between the RAM consumption for the last two operation:
C implementation consumes only: 384448 - 384032  = 416 bytes, where as 
pythonic implementation cosumes: 384032 - 382448 = 1584 bytes.
So, what we do compromise ROM or RAM??

v923z, 2020-04-03:
> Well, I do not agree totally with you. Operators such as `==` and `!=` are very basic in most of the numerical/ndarray libraries because there are many situations where you literally have to compare two arrays for equality/non-equality. I do not understand why are we comparing the whole obj in the case of `==` or `!=` not elementwise?

That is an error in the implementation, and I hadn't yet time to fix it. Sorry.

> I agree that binary operations are expensive and their implementation consume ROM/Flash memory, but if I've to achieve the same in pythonic way:
> `[[a[i][j] < 0. for j in range(num_cols)]for i in [...]

That is definitely true, but there is already a fast workaround, namely, your original solution: 

```
sum([a < 0.]) + sum([a > 0.])
```
All I am claiming is, if there is already a solution, and there is not too much to gain by implementing a different approach, then one has to think twice. I don't see the point in a hackish implementation, and based on my experience with `ulab`, I maintain that the proper implementation requires proper considerations. 

On the other hand, if you already have an implementation, you can point to it, and I would be happy to pull that in. 

> See the difference between the RAM consumption for the last two operation:
> C implementation consumes only: 384448 - 384032 = 416 bytes, where as
> pythonic implementation cosumes: 384032 - 382448 = 1584 bytes.

There is no question about the handicap of the pythonic code. Otherwise, `ulab` wouldn't exist.

> So, what we do compromise ROM or RAM??

I presume you have a PYBD, which has 2MB of flash. But there are other microcontrollers, with much less flash. I could have implemented quite a few things with much less effort, but I tried to squeeze out every last byte without compromising too much on performance. That shouldn't change.

v923z, 2020-04-06:
@mouryarahul I think https://github.com/v923z/micropython-ulab/pull/84 should fix this issue.

mouryarahul, 2020-04-06:
> @mouryarahul I think #84 should fix this issue.

@v923z Great! Thanks!

closed 2020-04-14 by v923z