v923z/micropython-ulab #73: ssize_t is not portable

KaiJan57, 2020-04-01:
https://github.com/v923z/micropython-ulab/blob/ac3f03c3bad9a40417f94be095c8337d1fc9c109/code/ndarray.c#L254

Might give compilation errors on other platforms. I think it would be a good idea to change the type or at least to add `#include<unistd.h>`

v923z, 2020-04-01:
> 
> Might give compilation errors on other platforms. 

Might, or does? Which platforms could potentially be problematic?

> I think it would be a good idea to change the type or at least to add `#include<unistd.h>`

I have absolutely nothing against including unistd.h, if that solves the problem, thanks for the suggestion! But in order to be able to test the hypothesis, we would have to know on which platform we have to compile.

KaiJan57, 2020-04-02:
The compilation error occurs indeed, namely when ulab is integrated into [this](https://github.com/Omega-Numworks/Omega) project, which is a calculator operating system that supports micropython. I could only manage to fix this error by adding said include statement.

v923z, 2020-04-02:
> The compilation error occurs indeed, namely when ulab is integrated into [this](https://github.com/Omega-Numworks/Omega) project, which is a calculator operating system that supports micropython. I could only manage to fix this error by adding said include statement.

Thanks for the comments! I will add the include, then.

KaiJan57, 2020-04-02:
Thank you very much!

v923z, 2020-04-02:
@KaiJan57 Jan, I have merged a fix into master. https://github.com/v923z/micropython-ulab/pull/81
Let me know if this is an acceptable solution.

v923z, 2020-04-02:
> If requested, I may add compiler logs, as integrating ulab into that project is quite a bit of a hassle.

I have overlooked your comment on this. Can we do something upstream that would make your life easier? It would benefit everyone, if `ulab` were a plug-and-play module.

closed 2020-04-02 by v923z

KaiJan57, 2020-04-02:
I think you have done everything you could. The last change made everything compile successfully and the "hassle" I was talking about originates from Omega's build system, so I'm quite sure we're good with what we have right now.