v923z/micropython-ulab #77: Certain malfuction in numerical.sort function

mouryarahul, 2020-04-01:
I'm not sure, is this happening only in my testing! I'm running test on:
> MicroPython v1.12-326-g8fff0b0ac on 2020-04-01; PYBD-SF6W with STM32F767IIK

The numerical.sort function is corrupting a certain element in the matrix:
`>>> a`
`array([[6.058420993051577, 2.018588295338188, 6.307303740008193, 6.743567296659661],
         [3.760399477657426, 1.501626760992716, 1.020398368879967, 1.903483081086812],
         [5.20075447514754, 0.7655233559988428, 0.2730999062713479, 5.448723074844875],
         [3.666587663986214, 0.5528238825462162, 7.759826840359533, 9.887756673795574],
         [3.228270818725618, 1.784248212712498, 5.302361277197747, 6.772730290987155]], dtype=float)`
`>>> ulab.numerical.sort(a, axis=0)`
`array([[3.228270818725618, 0.5528238825462162, 0.2730999062713479, 1.903483081086812],
         [4.550686286587561e-314, 0.7655233559988428, 1.020398368879967, 5.448723074844875],
         [3.666587663986214, 1.501626760992716, 5.302361277197747, 6.743567296659661],
         [3.760399477657426, 1.784248212712498, 6.307303740008193, 6.772730290987155],
         [5.20075447514754, 2.018588295338188, 7.759826840359533, 9.887756673795574]], dtype=float)`

Observe that the two elements of zero column has changed abruptly. However, this does happen when axis=1:
`>>> ulab.numerical.sort(a, axis=1)`
`array([[2.018588295338188, 6.058420993051577, 6.307303740008193, 6.743567296659661],
         [1.020398368879967, 1.501626760992716, 1.903483081086812, 3.760399477657426],
         [0.2730999062713479, 0.7655233559988428, 5.20075447514754, 5.448723074844875],
         [0.5528238825462162, 3.666587663986214, 7.759826840359533, 9.887756673795574],
         [1.784248212712498, 3.228270818725618, 5.302361277197747, 6.772730290987155]], dtype=float)`

Can you also verify at your end?

v923z, 2020-04-01:
> Can you also verify at your end?

Something is definitely wrong in the function. That is also obvious from the fact that you get `4.550686286587561e-314`. I will have to check this out. Sorry for the trouble.

v923z, 2020-04-01:
I have found the glitch. https://github.com/v923z/micropython-ulab/pull/79 should be a fix. If you are satisfied, please, close the issue.

mouryarahul, 2020-04-02:
@v923z is this intentional? Shouldn't be `axis = None` by default when not provided by user?
`>>> ulab.__version__`
`'0.39.1'`
`>>> b`
`array([[1, 2, -5, -4, 0],`
`         [0, -3, 0, 2, -3],`
`         [4, -5, 4, 3, -5],`
`         [0, 1, -1, 0, 0]], dtype=int16)`
`>>> ulab.numerical.sort(b)`
`array([[0, 1, 2, -5, -4],`
`         [0, 0, 2, -3, -3],`
`         [3, 4, 4, -5, -5],`
`         [0, 0, 0, 1, -1]], dtype=int16)`
`>>> ulab.numerical.sort(b, axis=None)`
`array([0, 0, 0, ..., -3, -3, -1], dtype=int16)`

v923z, 2020-04-03:
> @v923z is this intentional? Shouldn't be `axis = None` by default when not provided by user?

Yes, it is intentional: https://docs.scipy.org/doc/numpy/reference/generated/numpy.sort.html 
-1 is just short for last axis.

mouryarahul, 2020-04-03:
> > @v923z is this intentional? Shouldn't be `axis = None` by default when not provided by user?
> 
> Yes, it is intentional: https://docs.scipy.org/doc/numpy/reference/generated/numpy.sort.html
> -1 is just short for last axis.

Great! I missed to see this in NumPy. I guess now we can close this issue.

closed 2020-04-03 by mouryarahul