v923z/micropython-ulab #85: Out of curiosity about using CMSIS library

mouryarahul, 2020-04-06:
@v923z Hi, I was wondering if there exists already optimized Math and DSP library for ARM uController and found that there is one provided by ARM itself called [CMSIS](https://github.com/ARM-software/CMSIS_5), I guess you may be aware of it. I see that Micropython does use the only device driver part from CMSIS library from ST in the port stm32. I was wondering will it be possible to include CMSIS DSP library into ULAB, e.g. using [arm_math.h](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_math.h) and others? I'm not sure about the Licencing thing though!

mouryarahul, 2020-04-06:
> > @v923z Hi, I was wondering if there exists already optimized Math and DSP library for ARM uController and found that there is one provided by ARM itself called [CMSIS](https://github.com/ARM-software/CMSIS_5), I guess you may be aware of it.
> 
> No, I am not, thanks for bringing it to the fore!
> 
> > I see that Micropython does use the only device driver part from CMSIS library from ST in the port stm32. I was wondering will it be possible to include CMSIS DSP library into ULAB, e.g. using [arm_math.h](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_math.h) and others?
> 
> As far as I see, CMSIS is an abstraction library. However, in `micropython`, `micropython` itself is the abstraction layer (you don't have to care about the float implementation, or how the hardware is initialised, `micropython` does that for you). So, in this regard, what would be the advantage of CMSIS?
> 

I think CMSIS is not just abstraction but contains [numerical/mathematical functions](https://github.com/ARM-software/CMSIS_5/tree/c4c3480075295f226774bf134c00e91441aef1aa/CMSIS/DSP/Source) optimized for ARM Cortex M and others using available SIMD and Hard Floating Point instructions, which cannot be always achieved by generic math library and the compiler. I see it as the BLAS & LAPACK library for Numpy.

> Also, `micropython` does not require an ARM processor as such (the unix/windows port runs on a PC, e.g., but one could also mention the microchip MCUs), and linking a module to a specific hardware library would be a major drawback in my opinion.

I'm aware of the fact that `micropython` targets most of the available processor architectures, that's why they have chosen the generic library e.g. [musl](https://musl.libc.org/). However, we can always choose optimized library for at least for `pyboards` initially, which are in fact ARM Cortex M, and later target to support for other board based on ARM Cortex M.

> 
> > I'm not sure about the Licencing thing though!
> 
> CMSIS runs under the Apache 2.0 licence, which could be OK, though, mixing different types of licences is probably not a smart idea.

Well, micropython itself is using different libraries under different licences. 

> 
> When I set out to develop `ulab`, I decided to be independent of any licensing and linking constraints, because I didn't think that a purely numerical module for microcontrollers would require special libraries. I still stand by my original decision, and I think, there should be very good arguments in favour of a different approach.
> 
> To end on a more constructive note, which DSP functions do you miss? If it is within the `numpy`/`scipy` domain, we could think about adding an implementation.

All of the functions implemented in the Universal functions, Numerical, Linalg, Polynomials, Fourier transform and Filtering in the ULAB can get the advantage of CMSIS library, so we do not need to invent the wheel again for all those. However, including this library in ULAB can be exhausting work initially.

v923z, 2020-04-07:
> However, we can always choose optimized library for at least for pyboards initially, which are in fact ARM Cortex M, and later target to support for other board based on ARM Cortex M.

That would be the maintenance hell. At the moment, the single assumption of `ulab` is that `micropython` works. I do not want to work at the hardware level, however tempting that is.

> > > I'm not sure about the Licencing thing though!
> > 
> > 
> > CMSIS runs under the Apache 2.0 licence, which could be OK, though, mixing different types of licences is probably not a smart idea.
> 
> Well, micropython itself is using different libraries under different licences.

https://github.com/micropython/micropython : 
Note that MicroPython is licenced under the MIT license, and all contributions should follow this license.

And what do you do, if ARM decides to change their licence conditions later? I have released everything under the MIT licence,  and that will stay so forever, so that people can re-use the code with basically no strings attached. The code is also guaranteed to compile, because there are no dependencies. Try to appreciate the freedom of it!

> All of the functions implemented in the Universal functions, Numerical, Linalg, Polynomials, Fourier transform and Filtering in the ULAB can get the advantage of CMSIS library, so we do not need to [...]

But I am yet to be convinced of the advantages. What are they? If you say that we would gain a factor of 10 in speed, I would definitely consider it. If it is 2, then it's probably not worth it. If it is only 20%, then forget it! In the `ulab` manual, I published benchmarks, and I think such an argument should be based on numbers. 

Just an example: `ulab` gains a factor of 50 over `python` for the FFT, and a factor of 4 over an optimised assembly implementation. On top of that, everything happens in place, with basically no RAM overhead. I don't think you can squeeze out much more. 

I would also like to see the impact on flash. 

Also note that CMSIS is a C library, so you would still have to write the `numpy` front-end. If you look at the code in `ulab`, most of it is parsing, and not the algorithms themselves. As for universal functions, they would definitely not work out of the box, because CMSIS has no notion of iterators etc. 

And a very personal comment: re-inventing the wheel (i.e. learning the ins and outs of the algorithms) was the the enjoyable part. 

> However, including this library in ULAB can be exhausting work initially.

Not only that, but you would probably need a lot of extra flash. Basically, at that point it would be better to re-write `micropython` from scratch, and base everything on CMSIS. 

If you need a specific function, you can add that based on the guidelines in https://micropython-ulab.readthedocs.io/en/latest/ulab.html#extending-ulab. If you can't, let us know, and I am more than happy to see to implementing it. But I am not going to scrap everything, just so that we can include an external library/dependency.

mouryarahul, 2020-04-07:
> That would be the maintenance hell. At the moment, the single assumption of `ulab` is that `micropython` works. I do not want to work at the hardware level, however tempting that is. [...]

Thanks! Well, at the moment, I'm following your guidelines on how to extend ULAB. If I've or I need any suggestion, I'll convey to you.

closed 2020-04-07 by mouryarahul

v923z, 2020-04-07:
> Thanks! Well, at the moment, I'm following your guidelines on how to extend ULAB. If I've or I need any suggestion, I'll convey to you.

If you think that the community would profit from your implementation, you can issue a PR.