v923z/micropython-ulab #87: argmin/argmax of iterable: results differ from ndarray and numpy

jepler, 2020-04-06:
```
>>> ulab.numerical.argmax([[1,2],[4,3]])
1
>>> ulab.numerical.argmin([[1,2],[4,3]])
0
```
but
```
>>> ulab.numerical.argmin(ulab.array([[1,2],[4,3]]))
0
>>> ulab.numerical.argmax(ulab.array([[1,2],[4,3]]))
2
```
and
```
>>> numpy.argmin([[1,2],[4,3]])
0
>>> numpy.argmax([[1,2],[4,3]])
2
```

v923z, 2020-04-06:
This is sort of OK, and intentional: argmax and similar functions will work with one-dimensional iterables only: https://github.com/v923z/micropython-ulab/blob/280f04660e4ee7b56117c21bfe87a0274429e965/code/numerical.c#L151 
We should probably throw an exception, if this is not the case. The same applies to the other numerical functions: https://github.com/v923z/micropython-ulab/blob/280f04660e4ee7b56117c21bfe87a0274429e965/code/numerical.c#L60

If one really needs the maximum of a list of lists, then it should be converted to an `ndarray` first, thereby delegating the inspection to the initialisation function. Do you think that lists of lists should be supported here?

jepler, 2020-04-06:
No, I don't have a strong opinion. I looked at this code because there was an unused argument diagnostic nearby. The behavior isn't actually anyone to the best of my knowledge.

v923z, 2020-04-07:
> No, I don't have a strong opinion. I looked at this code because there was an unused argument diagnostic nearby. The behavior isn't actually anyone to the best of my knowledge.

I think at least 1D iterables should be supported, and the question really is how we prevent accidents like your example from happening. An easy way out would be to call `mp_obj_get_float`, and then compare numbers, instead of using `mp_binary_op(...)`.

And I have to go back on my earlier comment: `std`, `mean` and the like will not produce the error you found, because they do call `mp_obj_get_float`, so an exception will be raised automatically, if you try to pass lists of lists to the function.

v923z, 2020-04-07:
@jepler https://github.com/v923z/micropython-ulab/pull/91 is an attempt at a solution. If you agree, you can merge it, otherwise let me know if you have other suggestions.

closed 2020-04-13 by jepler