v923z/micropython-ulab #93: CNN extend for Numpy-like interface

yxdragon, 2020-04-11:
Hi, Glad to find this repo. 
Numpy is the base of science analysis in python. So micropython-ulab is a meaningful repo.

And now CNN is very prevent everywhere. So we wrote [Planer](https://github.com/Image-Py/planer), Powerful Light Artificial NEuRon. It is a CNN framework based on pure Numpy-like interface (with out any kernel code, just use reshape, transpose, concadinate, dot, and + - * /). I think it can run on with ulab as backend (with no modify or little modify).

I will buy a micropython board to have a try, if I encounter any problem, could you give a support?

v923z, 2020-04-11:
> Hi, Glad to find this repo.
> Numpy is the base of science analysis in python. So micropython-ulab is a meaningful repo.

Thanks!

> And now CNN is very prevent everywhere. So we wrote [Planer](https://github.com/Image-Py/planer), Powerful Light Artificial NEuRon. It is a CNN framework based on pure Numpy-like interface (with out [...]

I don't think that you are going to run into problems with compilation etc., so I guess, the question really is, whether, if necessary,  `ulab` could be modified/extended. I would be more than happy to do that. I see that you want to have concatenate. At the moment, it is not in `ulab`, but I have been planning to include that. It's is relatively little work, so would be no problem.

v923z, 2020-04-12:
@yxdragon Since this is actually a question, rather than an issue, I am closing it now. Open a new ticket, if you run into an implementation glitch, or if you need modifications of any kind! Thanks!

closed 2020-04-12 by v923z