v923z/micropython-ulab #95: reference document

v923z, 2020-04-14:
The continuation of the conversation in https://github.com/v923z/micropython-ulab/issues/17

v923z, 2020-04-14:
> I mean a document something like [NumPy Reference](https://numpy.org/doc/1.18/reference/index.html). At the moment, I don't how it's produced/maintained and I've never use sphinx too. But, such a document will be very handy for the `ulab` contributors such as yourself. 

OK, it's clear now. I am working on it, but I am not quite done yet. It didn't make sense to release a complete document before the internals are sorted out.

> I would suggest that if we want `ulab` be easier (useful) for the users, we must implement it with exactly the same input/output behaviours of the `Numpy.` By this, I don't mean that we have to implement all complex functions available in `Numpy`. At the moment, I'm not very familiar with many aspects of `micropython` implementations, so can't guess how difficult and time consuming it could to implement `ulab` behaving very similar to `Numpy`.

We haven't implemented anything un-numpythonic, and we made a point of it. From the users' standpoint, `ulab`'s behaviour is identical to `numpy`'s, with missing implementations. (E.g., the `convolve` function won't take all possible keyword arguments.) But the differences are pointed out everywhere in the manual. Or, where do you think the differences are confusing?

v923z, 2020-10-30:
I believe, this should be resolved in https://micropython-ulab.readthedocs.io/en/latest/.

v923z, 2020-10-31:
@mouryarahul, I think I would like to close the issue. Since the docs are published now, specific problems could be addressed with concrete references.

mouryarahul, 2020-11-01:
Great Work!

closed 2020-11-10 by v923z