v923z/micropython-ulab #100: Does not compile when using most current MicroPython

teuler, 2020-04-18:
After the fix https://github.com/v923z/micropython-ulab/issues/98 concerning the warnings (see there), I thought it might be also a good idea to use the newest MicroPython code. Unfortunately, compiling ulab now throws a lot of errors. The list below is just the beginning. Looks like a systematic issue. 

CC ../../../modules/ulab/code/ulab.c
In file included from ../../py/mpstate.h:35:0,
                 from ../../py/runtime.h:29,
                 from ../../../modules/ulab/code/fft.c:16:
../../../modules/ulab/code/fft.c:170:58: error: 'MP_QSTR_fft' undeclared here (not in a function)
     { MP_OBJ_NEW_QSTR(MP_QSTR___name__), MP_OBJ_NEW_QSTR(MP_QSTR_fft) },
                                                          ^
../../py/obj.h:94:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^
../../../modules/ulab/code/fft.c:172:23: error: 'MP_QSTR_ifft' undeclared here (not in a function)
     { MP_OBJ_NEW_QSTR(MP_QSTR_ifft), (mp_obj_t)&fft_ifft_obj },
                       ^
../../py/obj.h:94:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^
../../../modules/ulab/code/ndarray.c: In function 'ndarray_init_helper':
../../../modules/ulab/code/ndarray.c:158:11: error: 'MP_QSTR_dtype' undeclared (first use in this function)
         { MP_QSTR_dtype, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = NDARRAY_FLOAT } },
           ^
../../../modules/ulab/code/ndarray.c:158:11: note: each undeclared identifier is reported only once for each function it appears in
In file included from ../../../modules/ulab/code/extras.c:15:0:
../../../modules/ulab/code/extras.c:207:56: error: 'MP_QSTR_extras' undeclared here (not in a function)
   { MP_OBJ_NEW_QSTR(MP_QSTR___name__), MP_OBJ_NEW_QSTR(MP_QSTR_extras) },
                                                        ^
../../py/obj.h:94:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))

teuler, 2020-04-18:
P.S.: I am referring to the ESP32 port
```
ESPIDF = $(HOME)/esp/esp-idf
BOARD = GENERIC
PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-
include Makefile
```

v923z, 2020-04-18:
@teuler Thomas, could you, please, roll back your `micropython` to a previous version that still compiles? I am not set up for compiling for the ESP32 at the moment. I have tested https://github.com/micropython/micropython/commit/28833690bb9701cc9c381d0fd71f3a94d67f9c2f  on the unix and stm32 ports without issues. It would help, if we could find out, which `micropython` commit triggers the errors.

teuler, 2020-04-18:
Sure, I can do that for my stuff. I just wanted to report it because at first, it did not look like a port specific issue but rather that it systematically cannot find a specific set of macros.
Hmm, if other ports work fine then I could be ESP32 specific ...

teuler, 2020-04-18:
Before updating, I used a commit from last week. I just saw that 4 days ago, the ESP32 port was updated to using IDF v3.3.2. I'll post if I find out more

v923z, 2020-04-18:
> Before updating, I used a commit from last week. I just saw that 4 days ago, the ESP32 port was updated to using IDF v3.3.2. I'll post if I find out more

I am pretty sure this is a port-specific issue, I just don't see what it could be. E.g., here is the my unix prompt

```
MicroPython v1.12-375-g28833690b-dirty on 2020-04-18; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab
>>> 
```

The ESP32-related commits appear to be quite innocent. I will also set up my ESP32 environment, and try to find out, what's going on here.

teuler, 2020-04-18:
Wait with that for a minute. I think I found the problem. Hang on ...

teuler, 2020-04-18:
Now it compiles. The issue was the recent switch from IDF v3.3.1 to IDF v3.3.2. Somehow just updating the environment was not sufficient. I setup the whole chain again and now it works. Sorry for the inconvenience.

closed 2020-04-18 by teuler

v923z, 2020-04-18:
> Now it compiles. The issue was the recent switch from IDF v3.3.1 to IDF v3.3.2. Somehow just updating the environment was not sufficient. I setup the whole chain again and now it works. Sorry for the inconvenience.

Absolutely no problem, and thanks for the feedback! I also forgot to point out that 
```
git submodule update --init 
```
is probably required.