v923z/micropython-ulab #98: Failure to compile

rprr, 2020-04-18:
Cannot seem to get it to compile for an ESP32. On both Linux (Ubuntu) and Mac OS X. Here is the failure. The same error seems to occur when building for the PYBV11 as well. 

```
$ make -f GNUMakefile  USER_C_MODULES=../../../ulab all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../ulab/code
Building with ESP IDF v3
mkdir -p build-GENERIC_SPIRAM/code/
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c: In function 'ndarray_init_helper':
../../../ulab/code/ndarray.c:158:11: error: 'MP_QSTR_dtype' undeclared (first use in this function)
         { MP_QSTR_dtype, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = NDARRAY_FLOAT }
           ^
../../../ulab/code/ndarray.c:158:11: note: each undeclared identifier is reported only once for each function it appears in
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c: In function 'ndarray_make_new_core':
../../../ulab/code/ndarray.c:196:39: error: passing argument 1 of 'mp_raise_ValueError' from incompatible pointer type [-Werror=incompatible-pointer-types]
         mp_raise_ValueError(translate("first argument must be an iterable"));
                                       ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:167:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
               ^
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c:212:51: error: passing argument 1 of 'mp_raise_ValueError' from incompatible pointer type [-Werror=incompatible-pointer-types]
                     mp_raise_ValueError(translate("iterables are not of the sam
                                                   ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:167:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
               ^
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c: In function 'true_length':
../../../ulab/code/ndarray.c:276:46: error: passing argument 1 of 'mp_raise_TypeError' from incompatible pointer type [-Werror=incompatible-pointer-types]
                 mp_raise_TypeError(translate("wrong index type"));
                                              ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:168:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_TypeError(mp_rom_error_text_t msg);
               ^
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c: In function 'generate_slice':
../../../ulab/code/ndarray.c:296:57: error: passing argument 2 of 'mp_raise_msg' from incompatible pointer type [-Werror=incompatible-pointer-types]
             mp_raise_msg(&mp_type_IndexError, translate("index is out of bounds
                                                         ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:165:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_msg(const mp_obj_type_t *exc_type, mp_rom_error_text_t m
               ^
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c:302:53: error: passing argument 2 of 'mp_raise_msg' from incompatible pointer type [-Werror=incompatible-pointer-types]
         mp_raise_msg(&mp_type_IndexError, translate("indices must be integers,
                                                     ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:165:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
[... 134 lines cut ...]
../../../ulab/code/ndarray.c:991:39: error: passing argument 1 of 'mp_raise_ValueError' from incompatible pointer type [-Werror=incompatible-pointer-types]
         mp_raise_ValueError(translate("shape must be a 2-tuple"));
                                       ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:167:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
               ^
In file included from ../../../ulab/code/ndarray.c:21:0:
../../../ulab/code/ndarray.c:1003:39: error: passing argument 1 of 'mp_raise_ValueError' from incompatible pointer type [-Werror=incompatible-pointer-types]
         mp_raise_ValueError(translate("cannot reshape array (incompatible input
                                       ^
../../../ulab/code/ndarray.h:31:22: note: in definition of macro 'translate'
 #define translate(x) x
                      ^
In file included from ../../../ulab/code/ndarray.c:17:0:
../../py/runtime.h:167:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
               ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-GENERIC_SPIRAM/code/ndarray.o' failed
make: *** [build-GENERIC_SPIRAM/code/ndarray.o] Error 1
```

teuler, 2020-04-18:
I ran into the same problem. Compiling ulab produces a lot of warnings that (with default setting) are treated as errors. Try removing the flag `-Werror` in `~/micropython/ports/esp32/Makefile` (see `CFLAGS_COMMON`). This should not be a permanent solution, of course. It would be better to fix the ulab code such that these warnings are treated.

v923z, 2020-04-18:
> I ran into the same problem. Compiling ulab produces a lot of warnings that (with default setting) are treated as errors. Try removing the flag `-Werror` in `~/micropython/ports/esp32/Makefile` (see [...]

No, it should not be. I am working on a solution. Sorry for the trouble.

v923z, 2020-04-18:
@rprr Thanks for reporting the issue! It crept in unnoticed via an upstream change in `micropython`: https://github.com/micropython/micropython/commit/def76fe4d9bbc2c342594dc05861b24d7165d274

https://github.com/v923z/micropython-ulab/pull/99 should fix it. I tested it on the `stm32` port. If you still have difficulties, let me know, otherwise, you can close the issue.

@teuler Thomas, thanks for stepping in with a temporary solution!

rprr, 2020-04-18:
It builds and loads. On my esp32 with PSRAM. Thank you for your quick response. I pulled the latest.  

```
MPY: soft reboot
MicroPython v1.12-375-g28833690b-dirty on 2020-04-17; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import ulab as np
>>> print('you are running ulab version', np.__version__)
you are running ulab version 0.41.4
>>>
```

closed 2020-04-18 by rprr

teuler, 2020-04-18:
Thanks for taking action that quickly