v923z/micropython-ulab #102: Does not compile on PBD_SF6

lhargrove, 2020-04-20:
I noticed a few recent issues on ulab not compiling for the newest version of Micropython. I am having a similar issue. There seem to be lots of errors/warnings. 

lhargrove@lhargrove-VirtualBox:~/micropython/ports/stm32$ sudo make BOARD=PYBD_SF6 USER_C_MODULES=../../../ulab all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../ulab/code
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c: In function 'insert_binary_value':
../../../ulab/code/ndarray.c:320:37: error: conversion to 'float' from 'double' may alter its value [-Werror=float-conversion]
         int32_t x = (int32_t)floorf(mp_obj_get_float(tmp)+0.5);
                                     ^~~~~~~~~~~~~~~~
In file included from ../../py/mpstate.h:31:0,
                 from ../../py/runtime.h:29,
                 from ../../../ulab/code/ndarray.c:17:
../../../ulab/code/ndarray.c: In function 'ndarray_binary_op':
../../../ulab/code/ndarray.h:112:135: error: conversion to 'uint8_t {aka unsigned char}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:814:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_UINT8, uint8_t, uint8_t, uint8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:816:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, uint8_t, int8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'uint16_t {aka short unsigned int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:818:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_UINT16, uint16_t, uint8_t, uint16_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:820:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, uint8_t, int16_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:826:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, int8_t, uint8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int8_t {aka signed char}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:828:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT8, int8_t, int8_t, int8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
[... 48 lines cut ...]
                                  ^~~
../../../ulab/code/ndarray.c:850:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, int16_t, uint8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:852:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, int16_t, int8_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
../../../ulab/code/ndarray.h:112:135: error: conversion to 'int16_t {aka short int}' from 'double' may alter its value [-Werror=float-conversion]
 len); i++, left+=linc, right+=rinc) *odata++ = MICROPY_FLOAT_C_FUN(pow)(*left, *right); }\
                                                                    ^
../../py/mpconfig.h:681:34: note: in definition of macro 'MICROPY_FLOAT_C_FUN'
 #define MICROPY_FLOAT_C_FUN(fun) fun
                                  ^~~
../../../ulab/code/ndarray.c:856:6: note: in expansion of macro 'RUN_BINARY_LOOP'
      RUN_BINARY_LOOP(NDARRAY_INT16, int16_t, int16_t, int16_t, ol, or, op, m, n, len, linc, rinc);
      ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-PYBD_SF6/code/ndarray.o' failed
make: *** [build-PYBD_SF6/code/ndarray.o] Error 1

v923z, 2020-04-20:
Thanks for reporting the issue! I will look into it.

v923z, 2020-04-20:
@lhargrove Again, thanks for bringing this up! These errors are caused by https://github.com/micropython/micropython/commit/30840ebc9925bb8ef025dbc2d5982b1bfeb75f1b. I will try to fix it as soon as possible, but till then I would recommend roll back `micropython` to the state prior, https://github.com/micropython/micropython/commit/f31f9a8b70db03cbcbcf39b493f959d0e284962a. I have tried that, that compiles.

v923z, 2020-04-20:
@lhargrove https://github.com/v923z/micropython-ulab/pull/104 fixes the issue, I could compile `ulab` for the PYBD-SF6 board.

closed 2020-04-20 by v923z