v923z/micropython-ulab #111: better separation of core and user functions

v923z, 2020-04-27:
Currently the `extras` module is meant to be a user module, yet, contains the implementation of `spectrogram`. Such a mingling of user and core code make updating from git difficult.

Two alternatives could be to either 
1. move `spectrogram` to a core module (`fft` would make sense here), thereby emptying `extras`

or 

2. leave everything as it is, and add an empty `user` module, which would then remain always empty.

@jepler  @tannewt any comments?

tannewt, 2020-04-27:
Why do you need a user module at all? Couldn't they just make their own outside of ulab?

v923z, 2020-04-27:
> Why do you need a user module at all? Couldn't they just make their own outside of ulab?

Because the module still has to be bound at the top level: https://github.com/v923z/micropython-ulab/blob/cf61d728e70b9ec57e5711b40540793a89296f5d/code/ulab.c#L97

If there is a user module, they don't have to touch anything in the core, they can implement their functions separately.

teuler, 2020-04-28:
In my view, the extras module is quite useful to add a bit of own functionality while using all the infrastructure of ulab. Still, I agree that extras should be empty except maybe a small example function as a starting point for users. This would also be better than having the example code only in the documentation, where it needs to be kept up to date in addition to the version in extras

v923z, 2020-04-28:
> In my view, the extras module is quite useful to add a bit of own functionality while using all the infrastructure of ulab. Still, I agree that extras should be empty except maybe a small example [...]

Actually, I realised that something was off, when I saw your code. If we add anything to `extras`, you'll have difficulties.

teuler, 2020-04-28:
I agree that I "misuse" extras currently. I am testing by how much I can speed up my MicroPython code by strategically replacing bits with C code; these bit currently may not be of general use - either because they feature no thorough type checking or because they consider special cases. I am still experimenting. An your library is a great playground.

In general, my thinking was that extras could be the place to customize ulab. Assuming the ulab conventions stay the same, one could take the latest version of ulab and "plug-in" an own extras module for customization. In this sense, I see extras rather as a user module - which was probably not the way you intended it to be.

This touches also upon another point that I realised when looking more into the ulab source code (e.g. transpose). Wouldn't it be more memory efficient to separate the actual function from the ulab interface? Say, I'd want to re-use transpose within extras; I would want to call the function w/o the overhead of type checking and converting python objects, would'nt I?

v923z, 2020-04-28:
> I agree that I "misuse" extras currently. 

No, actually, this was the original idea behind `extras`, and then I accidentally placed `spectrogram` into that module. That was the mistake. 

> In general, my thinking was that extras could be the place to customize ulab. Assuming the ulab conventions stay the same, one could take the latest version of ulab and "plug-in" an own extras [...]

At the moment, there is no easy way of sub-classing at the C level, so the customisation that you envision is probably not doable. It would be great, in fact, if that was the case, then you would only need what is in `ndarray.c`, nothing more. But I don't expect that this is going to happen anytime soon. That would have to be done in `micropython` proper, and not in `ulab`.

> This touches also upon another point that I realised when looking more into the ulab source code (e.g. transpose). Wouldn't it be more memory efficient to separate the actual function from the ulab [...]

If you want to use `transpose`, or any other function, for that matter, then you could remove the static keyword from the function implementation, and add the declaration to `linalg.h`, and include `linalg.h` in `extras.c`. But you are right, the algorithm itself would have to be decoupled from the interface. However, you still have to safeguard against feeding the wrong stuff to the function, otherwise, the segmentation fault is going to rear its ugly head.

@jepler Jeff, any comments on that? You insisted on making things static. I see the reason behind that, but I also see some limitations.

dhalbert, 2020-04-28:
I won't try to speak for @jepler, but CircuitPython doesn't try to be pedantic about `static`. If a function is useful, it could be called from elsewhere.

In CircuitPython, the Python bindings and the argument validation is done in `shared-bindings/<modulename>/<submodule_or_Class>.c`. Then the actual implementation is done in either `shared-module/<modulename>/<submodule_or_Class>.c` (for port-independent code), or `port/<portname>/common-hal/<modulename>/<submodule_or_Class>.c` (for port-specific code such as peripheral access). The routines in `shared-module/` or `common-hal/` are usually named `common_hal_<modulename>_...` and of course they are not static.

Because the `common_hal` routines are not static, they are readily available to call from anywhere. The `shared-bindings/` routines do not usually reach directly into the `struct`s defined by `common_hal`; instead they call a `common_hal` routine to manipulate the `struct` members.

This split into different directory hierarchies doesn't exist in MicroPython, and we haven't tried to impose it on `ulab`. But the Python side and the implementation side could still be split up in `ulab` to achieve the same effect and allow easy substitution or reuse.

v923z, 2020-04-28:
> This split into different directory hierarchies doesn't exist in MicroPython, and we haven't tried to impose it on `ulab`. But the Python side and the implementation side could still be split up in `ulab` to achieve the same effect and allow easy substitution or reuse.

@dhalber, @teuler thanks for the comments! Since this developed into a separate issue, I have opened a new ticket: https://github.com/v923z/micropython-ulab/issues/112

dhalber, 2020-04-28:
> @dhalber, @teuler thanks for the comments! Since this developed into a separate issue, I have opened a new ticket: #112

@dhalbert not @dhalber

v923z, 2020-04-29:
> >thanks for the comments! Since this developed into a separate issue, I have opened a new ticket: #112
> 
> @dhalbert not @dhalber

Indeed. Sorry for the noise!

v923z, 2020-07-24:
> In my view, the extras module is quite useful to add a bit of own functionality while using all the infrastructure of ulab. Still, I agree that extras should be empty except maybe a small example [...]

Since we have replaced `extras` by the almost-empty `user` module (https://github.com/v923z/micropython-ulab/tree/master/code/user), I am going to close this issue. The one about the out-factoring of math functions (https://github.com/v923z/micropython-ulab/issues/112) stays alive for the time being.

closed 2020-07-24 by v923z