v923z/micropython-ulab #112: factor out algorithms in a re-useable form

v923z, 2020-04-28:
The issue came up in a different context here: https://github.com/v923z/micropython-ulab/issues/111, and in particular in the comment https://github.com/v923z/micropython-ulab/issues/111#issuecomment-620483197, and asks for some re-factoring of the code, so that functions could be re-used in the user modules. 

It should still be discussed, how exactly the decoupling should happen, and which functions should be considered. Some functions, in particular in the `numerical` sub-module are particularly hard to separate from the python interface. On the other hand, there are a number of functions that are legitimate candidates.

@teuler @danh

teuler, 2020-04-29:
A possibility here would be to do it the way I did it with `transpose` (unfortunately, I haven't yet pushed the respective code to the repository for here to show): To use ulab elements (i.e. `ndarray`) to pass data but keep all micropython related code and the type checking in the "interface" - assuming that new user functions that offer a micropython interface do their own respective checking.

v923z, 2020-04-29:
Could you compile a list of functions that would be useful for you? Before turning the whole library upside down, we could try to do the decoupling for the functions that you actually need. That could also serve as a test bed. We might have to iterate a couple of rounds before we find a working solution, and I would rather do that on a small sample.

v923z, 2020-04-30:
> A possibility here would be to do it the way I did it with `transpose` (unfortunately, I haven't yet pushed the respective code to the repository for here to show): To use ulab elements (i.e. `ndarray`) to pass data but keep all micropython related code and the type checking in the "interface" - 

I think, `transpose` is not a good example: it is in `ndarray.c` with the assumption that it is a method of the array itself, so absolutely no type checking takes place: https://github.com/v923z/micropython-ulab/blob/cf61d728e70b9ec57e5711b40540793a89296f5d/code/ndarray.c#L960
It is also exposed in the header file: https://github.com/v923z/micropython-ulab/blob/cf61d728e70b9ec57e5711b40540793a89296f5d/code/ndarray.h#L83

I just don't see at the moment, what you had to hack in relation to `transpose`.

teuler, 2020-05-01:
Sorry for the delay. My daytime job ...
You are right about transpose.

v923z, 2020-11-12:
@teuler Thomas, https://github.com/v923z/micropython-ulab/pull/217 implements some of the changes that you requested. If you see other functions that could be good candidates for out-factoring, let me know.

You also wanted to have `numpy` compatibility, math constants exposed in `ulab` itself, and `radians` and `degrees`. These can all be switched on in versions past 1.0.0.

closed 2020-11-12 by v923z