v923z/micropython-ulab #113: Feature requests: ability to adjust "PRINT_MAX"

DeqingSun, 2020-05-01:
ulab has a fixed PRINT_MAX, and this makes print array inconvenient in the debugging process. 

numpy.set_printoptions can do it. And the default value is larger in Numpy

v923z, 2020-05-01:
I think this is relatively easy to fix, I can do that. I would, however, only support the `threshold`, and perhaps the `edgeitems` keyword arguments in the first round. I hope that is OK.

v923z, 2020-05-01:
@DeqingSun Could you, please, look at https://github.com/v923z/micropython-ulab/pull/114? Just check out the print branch, and see if it does what it is supposed to. Thanks!

DeqingSun, 2020-05-01:
Awesome!
Here is my test:
~~~
np.get_printoptions()
result = np.array(range(15))
print(result)
np.set_printoptions(edgeitems=5)
print(result)
np.set_printoptions(threshold=20)
print(result)
~~~
output is
~~~
{'threshold': 10, 'edgeitems': 3}
array([0.0, 1.0, 2.0, ..., 12.0, 13.0, 14.0], dtype=float)
array([0.0, 1.0, 2.0, 3.0, 4.0, ..., 10.0, 11.0, 12.0, 13.0, 14.0], dtype=float)
array([0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0, 11.0, 12.0, 13.0, 14.0], dtype=float)
~~~

closed 2020-05-19 by v923z