v923z/micropython-ulab #117: Weirdness with "scalar * array"

jepler, 2020-05-18:
This may be "*" only, "/", "-", and "+" seemed to work.

```
$ ./micropython/ports/unix/micropython 
MicroPython v1.12-368-gbd63c26dd-dirty on 2020-05-18; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab
>>> 3200 * ulab.array([0.1])
array([12.8], dtype=float)
>>> 32000 * ulab.array([0.1])
array([0.0], dtype=float)
>>> ulab.array([0.1]) * 3200
array([320.0], dtype=float)
>>> ulab.array([0.1]) * 32000
array([3200.0], dtype=float)
```

v923z, 2020-05-18:
Indeed, this is not correct. I will look into this. Thanks for reporting!

v923z, 2020-05-19:
@jepler https://github.com/v923z/micropython-ulab/pull/119 fixes the issue! It was a trivial (and embarrassing) mistake in the binary code.

closed 2020-05-19 by v923z