v923z/micropython-ulab #124: fft error

skylin008, 2020-06-10:


v923z, 2020-06-10:
@skylin008 Thanks for reporting the issue! This is caused by recent changes in `micropython` itself, and the culprit is the `translate` macro in https://github.com/v923z/micropython-ulab/blob/4a3c12e427d3a3fb64443b521a5fd15c3df792bd/code/ndarray.h#L33 which expands into the original `micropython` definition in https://github.com/micropython/micropython/blob/29e258611ace8e4de51d810d5ff14364a5fbb950/py/misc.h#L308

It was brought in in the https://github.com/micropython/micropython/commit/154b4eb354d97c7c28253bdc5212b2e58ea6462e#diff-26e26b322018ede8ae704b19c20b13f7 commit. 

I presume that `openmv` uses an earlier version of `micropython`. I would suggest that we bring in someone from `openmv`. My solution would be something like this

```
#if CIRCUITPY
#define mp_obj_is_bool(o) (MP_OBJ_IS_TYPE((o), &mp_type_bool))
#define mp_obj_is_int(x) (MP_OBJ_IS_INT((x)))
#elif OPENMV
#define translate(x) x
#else
#define translate(x) MP_ERROR_TEXT(x)
#endif
```
but I would have to know, whether the contanst `OPENMV` or something similar is defined.  @iabdalkader could you, please, comment on this? I would be happy to add the required changes to `ulab`. 

Otherwise, as a temporary measure, you can simply replace the `#define translate(x) MP_ERROR_TEXT(x)` definition in `ndarray.h` with `#define translate(x) x`, and I think that should fix it.

skylin008, 2020-06-11:
@v923z Thanks for your kindly advice.I will be try.

iabdalkader, 2020-06-11:
Hello, yes we're using MicroPython 1.9.4 and older ulab. There's no plan to update any time soon as things are stable. 
>> but I would have to know, whether the contanst OPENMV or something similar is defined
No I set the board name, so you'd have check for all boards. Maybe it's better to check for MicroPython version if possible, or I could just add that macro to misc.h until we update.

v923z, 2020-06-11:
> Hello, yes we're using MicroPython 1.9.4 and older ulab. There's no plan to update any time soon as things are stable.

I think that won't work: this change was committed on the 14th of April, but `mpconfig.h`, where the `micropython` version is defined,  was not modified around that time

https://github.com/micropython/micropython/commits/246f3f640dd635f67269bfdbebe23408f77cd2b0/py/mpconfig.h

It seems to me that the `micropython` version is updated only, if user-facing modifications (the python API) are implemented, and this one concerns only how you compile code, so users wouldn't even notice it. 

> or I could just add that macro to misc.h until we update.

Either way is fine with me, though, I have the feeling that it should be fixed on this side. Wouldn't it be better, if you defined a constant in `openmv` that we could use in `ulab`?

iabdalkader, 2020-06-12:
> I have the feeling that it should be fixed on this side

Why don't you just check if `MP_ERROR_TEXT` is defined ?
```
#ifdef MP_ERROR_TEXT
#define translate(x) MP_ERROR_TEXT(x)
#else
#define translate(x) x
#endif
```

v923z, 2020-06-12:
@iabdalkader @skylin008 Could you, please, check, whether you can compile with https://github.com/v923z/micropython-ulab/pull/125 (https://github.com/v923z/micropython-ulab/tree/openmv-fix)? Thanks!

v923z, 2020-06-12:
> Hello, yes we're using MicroPython 1.9.4 and older ulab. There's no plan to update any time soon as things are stable.

@iabdalkader We have fixed a number of smaller bugs recently, so you might want to update `ulab`.

iabdalkader, 2020-06-12:
This fixes the `MP_ERROR_TEXT` issue, but there's a lot of other errors, for example we're missing  is_bool and mp_obj_slice_indices, and the ulab_user_cmodule has been changed from const to non const. I think 1.9.4 is just not compatible with the latest ulab. I'll make sure to update the submodule when updating micropython. @skylin008 can probably get this building with some modifications in the mean time.

v923z, 2020-06-12:
@iabdalkader I can try to look into this. Could you please, tell, where to bind `ulab` to `openmv`?

iabdalkader, 2020-06-12:
Just clone the repo recursively, it will pull all the dependencies:

https://github.com/openmv/openmv.git

You'll find ulab in src/micropython/extmod/ulab

If you want to pull the latest, in ulab `git fetch origin && git pull` or try `git pull origin master`

v923z, 2020-06-12:
OK, thanks!

skylin008, 2020-06-13:
@v923z  @iabdalkader  Yes,I'm here 。I had update the lasted ulab. Rebuild the firmware, Any Error as show:
#define translate(x)   x
#endif
"micropython/extmod/ulab/code/ndarray.h" 149L, 6461C written
phonelink@phonelink-Ubuntu:~/Desktop/openmv/openmv/src$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Warning: FROZEN_MPY_DIR is deprecated in favour of FROZEN_MANIFEST
CC ../../extmod/ulab/code/fft.c
CC ../../extmod/ulab/code/linalg.c
CC ../../extmod/ulab/code/ndarray.c
../../extmod/ulab/code/ndarray.c: In function 'true_length':
../../extmod/ulab/code/ndarray.c:314:13: error: implicit declaration of function 'mp_obj_is_bool'; did you mean 'mp_obj_new_bool'? [-Werror=implicit-function-declaration]
         if(!mp_obj_is_bool(item)) {
             ^~~~~~~~~~~~~~
             mp_obj_new_bool
../../extmod/ulab/code/ndarray.c: In function 'generate_slice':
../../extmod/ulab/code/ndarray.c:330:9: error: implicit declaration of function 'mp_obj_slice_indices'; did you mean 'mp_obj_slice_get'? [-Werror=implicit-function-declaration]
         mp_obj_slice_indices(index, n, &slice);
         ^~~~~~~~~~~~~~~~~~~~
         mp_obj_slice_get
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target '/home/phonelink/Desktop/openmv/openmv/src/build/micropython/extmod/ulab/code/ndarray.o' failed
make[1]: *** [/home/phonelink/Desktop/openmv/openmv/src/build/micropython/extmod/ulab/code/ndarray.o] Error 1
Makefile:584: recipe for target 'FIRMWARE_OBJS' failed
make: *** [FIRMWARE_OBJS] Error 2

v923z, 2020-06-13:
I have tried to dig a bit deeper, and here is what I've found:

1. objslice.c has gone through some major re-work since 1.9.4. I think, this could be solved by defining the required functions ourselves (I can, at least, compile the code, if I do that) 

2. in ulab.c, I had to change the declaration of `ulab_user_cmodule` like so

```
const struct _mp_obj_module_t ulab_user_cmodule = {
    .base = { &mp_type_module },
    .globals = (mp_obj_dict_t*)&mp_module_ulab_globals,
};
```
3. in ndarray.h, I had to add

```
#else
#define mp_obj_is_bool(o) (MP_OBJ_IS_TYPE((o), &mp_type_bool))
#define translate(x) x

typedef struct _mp_obj_slice_t {
    mp_obj_base_t base;
    mp_obj_t start;
    mp_obj_t stop;
    mp_obj_t step;
} mp_obj_slice_t;

void mp_obj_slice_get(mp_obj_t self_in, mp_obj_t *, mp_obj_t *, mp_obj_t *);
#endif
```

With these modifications, I can compile the code, but get linking errors later: 

```
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x2c): undefined reference to `create_zeros_obj'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x34): undefined reference to `create_ones_obj'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x3c): undefined reference to `create_eye_obj'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x44): undefined reference to `create_linspace_obj'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x74): undefined reference to `ulab_filter_module'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x7c): undefined reference to `ulab_compare_module'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: /home/v923z/sandbox/openmv/openmv/src/build/micropython/extmod/ulab/code/ulab.o:(.rodata.ulab_globals_table+0x84): undefined reference to `ulab_approx_module'
collect2: error: ld returned 1 exit status
make: *** [Makefile:609: firmware] Error 1
```
These are caused by sub-modules that were added after you bound `ulab` to `openmv`, so I presume that `openmv` does not pick up the make fragment in `openmv/openmv/src/micropython/extmod/ulab/code`, even if it is up to date. @iabdalkader do you have a suggestion as to how to fix this?

v923z, 2020-06-13:
Oh, I see, the make fragment is not used at all. I had to modify `Makefile` in `openmv/src' as

```
FIRM_OBJ += $(addprefix $(BUILD)/$(MICROPY_DIR)/extmod/ulab/,\
        code/fft.o          \
        code/linalg.o       \
        code/ndarray.o      \
        code/numerical.o    \
        code/poly.o         \
        code/ulab.o         \
        code/vectorise.o    \
        code/create.o           \
        code/approx.o           \
        code/filter.o           \
        code/compare.o          \
        )
endif
```
and then `extmod.mk` in `openmv/src/micropython/extmod` as

```
ULAB_DIR = extmod/ulab
SRC_MOD += $(addprefix $(ULAB_DIR)/,\
        code/fft.c          \
        code/linalg.c       \
        code/ndarray.c      \
        code/numerical.c    \
        code/poly.c         \
        code/ulab.c         \
        code/vectorise.c    \
        code/approx.c   \
        code/filter.c   \
        code/compare.c  \
        code/create.c   \
        )
endif
```
With that I get a piece of firmware that I can't test, since I haven't a board:
```
   text	   data	    bss	    dec	    hex	filename
  19956	    224	  10692	  30872	   7898	/home/v923z/sandbox/openmv/openmv/src/build/bin/bootloader.elf
   text	   data	    bss	    dec	    hex	filename
1821012	   2256	 783276	2606544	 27c5d0	/home/v923z/sandbox/openmv/openmv/src/build/bin/firmware.elf
```

@skylin008 In addition to what I had written in the last post and this, you also have to add a function to `ndarray.c`: 

```
void mp_obj_slice_indices(mp_obj_t self_in, mp_int_t length, mp_bound_slice_t *result) {
    mp_obj_slice_t *self = MP_OBJ_TO_PTR(self_in);
    mp_int_t start, stop, step;

    if (self->step == mp_const_none) {
        step = 1;
    } else {
        step = mp_obj_get_int(self->step);
        if (step == 0) {
            mp_raise_ValueError(translate("slice step can't be zero"));
        }
    }

    if (step > 0) {
        // Positive step
        if (self->start == mp_const_none) {
            start = 0;
        } else {
            start = mp_obj_get_int(self->start);
            if (start < 0) {
                start += length;
            }
            start = MIN(length, MAX(start, 0));
        }

        if (self->stop == mp_const_none) {
            stop = length;
        } else {
            stop = mp_obj_get_int(self->stop);
            if (stop < 0) {
                stop += length;
            }
            stop = MIN(length, MAX(stop, 0));
        }
    } else {
        // Negative step
        if (self->start == mp_const_none) {
            start = length - 1;
        } else {
            start = mp_obj_get_int(self->start);
            if (start < 0) {
                start += length;
            }
            start = MIN(length - 1, MAX(start, -1));
        }

        if (self->stop == mp_const_none) {
            stop = -1;
        } else {
            stop = mp_obj_get_int(self->stop);
            if (stop < 0) {
                stop += length;
            }
            stop = MIN(length - 1, MAX(stop, -1));
        }
    }

    result->start = start;
    result->stop = stop;
    result->step = step;
}
```
just before `static mp_bound_slice_t generate_slice(mp_int_t n, mp_obj_t index) {...`

Let me know if you still have difficulties.

@iabdalkader I could add the required changes to `ulab`, but the makefiles you would have to change on your end. Also, it would be great, if we could stick with 
```
mp_obj_module_t ulab_user_cmodule = {
    .base = { &mp_type_module },
    .globals = (mp_obj_dict_t*)&mp_module_ulab_globals,
};
```
Furthermore, I would really suggest that you define a constant that identifies `openmv`, so that we can sensibly separate code on our side with pre-processor macros.

skylin008, 2020-06-13:
@v923z ,Thank you for your support .The issue be solved.

v923z, 2020-06-13:
> @v923z ,Thank you for your support .The issue be solved.

@skylin008  Thanks for the feedback, I am glad it worked out! I would re-open the issue, so that @iabdalkader has a chance to comment on the `openmv`-specific remarks. I believe, with a bit of work, we could make the code `openmv`-compatible.

iabdalkader, 2020-06-14:
Hi, yes I can update the Makefiles and add `mp_obj_slice_indices` to py/objslice.c, also can export OPENMV in CFLAGS. Any reason why ulab_user_cmodule should not be const ? Also all the other modules in mpconfigport.h are declared like this:
```
extern const struct _mp_obj_module_t xxx_module;
```
I don't know why `struct _mp_obj_module_t` is used instead of the typedef `mp_obj_module_t`, probably old code that never got updated.. But it should be const.

Also any plans to do a release soon ? I like to update submodules to release tags.

v923z, 2020-06-15:
> Hi, yes I can update the Makefiles and add `mp_obj_slice_indices` to py/objslice.c, 

I was thinking that I could add `mp_obj_slice_indices` to a local file that we include only for `openmv`, so that we wouldn't mess up your side. 

> also can export OPENMV in CFLAGS. 

That would be great, thanks!

> Any reason why ulab_user_cmodule should not be const ? 

I don't know. I don't care, as long as it compiles. @jepler Jeff, is it OK, if we change the function signature here?

> Also any plans to do a release soon ? I like to update submodules to release tags.

We can definitely add a tag. People usually take the latest code, and compile it into their firmware without second thoughts, but I understand that you want to do this in a bit more formal way. We'll do.

iabdalkader, 2020-06-15:
> I was thinking that I could add mp_obj_slice_indices to a local file that we include only for openmv, so that we wouldn't mess up your side.

It's okay, I'll add it and when I update MicroPython will fix conflicts..

> I don't know. I don't care, as long as it compiles. @jepler Jeff, is it OK, if we change the function signature here?

const keeps it in flash, like all the other micropython modules, otherwise it goes into initialized ram and we're tight on ram.

v923z, 2020-06-15:
I think https://github.com/v923z/micropython-ulab/pull/125/files should fix it now. In addition, you'll need https://github.com/v923z/openmv/commit/f8261abd66f23653005c5718d433f21232c5e1ce, and a similar one in https://github.com/openmv/micropython/blob/19c2bc76e0e65371db21cc690652f139d82d97ba/extmod/extmod.mk#L233

I couldn't commit to that, because that is a sub-module, so here is the content:

```
ifeq ($(MICROPY_PY_ULAB),1)
CFLAGS_MOD += -DMICROPY_PY_ULAB=1 -I$(TOP)/extmod/ulab/code
ULAB_DIR = extmod/ulab
SRC_MOD += $(addprefix $(ULAB_DIR)/,\
	code/fft.c          \
	code/linalg.c       \
	code/ndarray.c      \
	code/numerical.c    \
	code/poly.c         \
	code/ulab.c         \
	code/vectorise.c    \
	code/create.c    	\
	code/approx.c		\
	code/filter.c		\
	code/compare.c		\
	code/extras.c		\
	)
endif
```
If you can compile with these changes, I will merge `openmv-fix` branch into master.

iabdalkader, 2020-06-16:
Hi I tried adding `mp_obj_slice_indices` to objslice.c it depends on a missing function which in turn depend on other missing functions. It's just becoming messy to update this way, I'll wait until we update MicroPython then all of this won't be necessary.

v923z, 2020-06-16:
> Hi I tried adding `mp_obj_slice_indices` to objslice.c it depends on a missing function which in turn depend on other missing functions. It's just becoming messy to update this way, I'll wait until we update MicroPython then all of this won't be necessary.

That's right, but _your_ users might be of a different opinion;)

I could add the function to `ulab`, if you want...

iabdalkader, 2020-06-16:
> I could add the function to ulab, if you want...

Yes maybe it's best this way, when you do please add a tag, I'll then update the submodule and fix the makefiles.

v923z, 2020-06-16:
> Yes maybe it's best this way, when you do please add a tag, I'll then update the submodule and fix the makefiles.

If you define OPENMV somethere, you should be able to compile with the 
https://github.com/v923z/micropython-ulab/tree/openmv-fix branch. If you succeed, let me know, so that I can merge the stuff into master, and add a release tag.

Also, please, ping me, when you update `micropython`, so that we can remove redundant code here. (Redundant code would also prevent you from compiling `ulab` on your side.)

iabdalkader, 2020-06-16:
No doesn't build, I'm getting a lot of errors about missing qstrs:

```
../../extmod/ulab/code/create.c: In function 'create_eye':
../../extmod/ulab/code/create.c:66:11: error: 'MP_QSTR_M' undeclared (first use in this function); did you mean 'MP_QSTR_n'?
         { MP_QSTR_M, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = mp_const_none } },
           ^~~~~~~~~
           MP_QSTR_n
../../extmod/ulab/code/create.c:66:11: note: each undeclared identifier is reported only once for each function it appears in
../../extmod/ulab/code/filter.c: In function 'filter_convolve':
../../extmod/ulab/code/filter.c:23:11: error: 'MP_QSTR_a' undeclared (first use in this function); did you mean 'MP_QSTR_n'?
         { MP_QSTR_a, MP_ARG_REQUIRED | MP_ARG_OBJ, {.u_rom_obj = mp_const_none } },
           ^~~~~~~~~
           MP_QSTR_n
../../extmod/ulab/code/filter.c:23:11: note: each undeclared identifier is reported only once for each function it appears in
../../extmod/ulab/code/create.c:67:11: error: 'MP_QSTR_k' undeclared (first use in this function); did you mean 'MP_QSTR_M'?
         { MP_QSTR_k, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 0} },
           ^~~~~~~~~
           MP_QSTR_M
../../extmod/ulab/code/approx.c: In function 'approx_bisect':
../../extmod/ulab/code/approx.c:39:11: error: 'MP_QSTR_xtol' undeclared (first use in this function); did you mean 'MP_QSTR_xnor'?
         { MP_QSTR_xtol, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_PTR(&xtolerance)} },
           ^~~~~~~~~~~~
           MP_QSTR_xnor
../../extmod/ulab/code/approx.c:39:11: note: each undeclared identifier is reported only once for each function it appears in
../../extmod/ulab/code/filter.c:24:11: error: 'MP_QSTR_v' undeclared (first use in this function); did you mean 'MP_QSTR_a'?
         { MP_QSTR_v, MP_ARG_REQUIRED | MP_ARG_OBJ, {.u_rom_obj = mp_const_none } },
           ^~~~~~~~~
           MP_QSTR_a
../../extmod/ulab/code/create.c: In function 'create_linspace':
../../extmod/ulab/code/create.c:110:11: error: 'MP_QSTR_num' undeclared (first use in this function); did you mean 'MP_QSTR_sum'?
         { MP_QSTR_num, MP_ARG_INT, {.u_int = 50} },
           ^~~~~~~~~~~
           MP_QSTR_sum
../../extmod/ulab/code/create.c:110:11: error: initialization makes integer from pointer without a cast [-Werror=int-conversion]
../../extmod/ulab/code/create.c:110:11: note: (near initialization for 'allowed_args[2].qst')
../../extmod/ulab/code/create.c:110:11: error: initializer element is not constant
../../extmod/ulab/code/create.c:110:11: note: (near initialization for 'allowed_args[2].qst')
In file included from ../../extmod/ulab/code/compare.c:15:0:
../../extmod/ulab/code/compare.c:180:23: error: 'MP_QSTR_equal' undeclared here (not in a function); did you mean 'MP_QSTR_eval'?
     { MP_OBJ_NEW_QSTR(MP_QSTR_equal), (mp_obj_t)&compare_equal_obj },
                       ^
/home/mux/openmv/src/micropython/py/obj.h:92:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^~~
../../extmod/ulab/code/approx.c:40:11: error: 'MP_QSTR_maxiter' undeclared (first use in this function); did you mean 'MP_QSTR_maxsize'?
         { MP_QSTR_maxiter, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 100} },
```
And this goes on.

v923z, 2020-06-16:
The only thing that is not in the PR above is the definition of 

```
#define OPENMV
```
I have that at the top of `ndarray.h`, because I was lazy to find out how I could pass it to the makefile. Otherwise, I pulled your full repository yesterday night, and that is what I have been working. Here is my compile log:
```
...
CC ../../lib/berkeley-db-1.xx/btree/bt_split.c
CC ../../lib/berkeley-db-1.xx/btree/bt_utils.c
CC ../../lib/berkeley-db-1.xx/mpool/mpool.c
CC ../../extmod/ulab/code/fft.c
CC ../../extmod/ulab/code/linalg.c
CC ../../extmod/ulab/code/ndarray.c
CC ../../extmod/ulab/code/numerical.c
CC ../../extmod/ulab/code/poly.c
CC ../../extmod/ulab/code/ulab.c
CC ../../extmod/ulab/code/vectorise.c
CC ../../extmod/ulab/code/create.c
CC ../../extmod/ulab/code/approx.c
CC ../../extmod/ulab/code/filter.c
CC ../../extmod/ulab/code/compare.c
CC ../../extmod/ulab/code/extras.c
CC mbedtls/mbedtls_port.c
CC /home/v923z/sandbox/openmv/openmv2/openmv/src/build/micropython/pins_OPENMV4.c
...
CC src/usbd_ioreq.c
CC src/usbd_desc.c
CC src/systick.c
   text	   data	    bss	    dec	    hex	filename
  19956	    224	  10692	  30872	   7898	/home/v923z/sandbox/openmv/openmv2/openmv/src/build/bin/bootloader.elf
   text	   data	    bss	    dec	    hex	filename
1821056	   2264	 783268	2606588	 27c5fc	/home/v923z/sandbox/openmv/openmv2/openmv/src/build/bin/firmware.elf
```
skylin008 also seemed to be able to compile.

v923z, 2020-06-16:
Since I have no boards, I just ran `make` in `openmv/src/micropython`. Is that the difference?

iabdalkader, 2020-06-16:
Never mind it's building fine now, that's weird yeah probably didn't make clean. Okay so you can now merge and add the tag, I'll do the rest.

v923z, 2020-06-16:
I have merged the code with the PR https://github.com/v923z/micropython-ulab/pull/125
Here is the release tag: 

https://github.com/v923z/micropython-ulab/releases/tag/0.50.2

closed 2020-06-16 by v923z