v923z/micropython-ulab #126: Question: IIR filters?

khashdroid, 2020-06-14:
Hi there!
Great library! 
I'm working with a nrf52840 board and I need to do some basic filtering. say I have a precomputed IIR filter and I want to apply it to a signal. I was wondering if ulab offers a function to apply an IIR filter to a signal (since you have more advanced functions such as FFT).
Thanks!

v923z, 2020-06-14:
At the moment, there is only convolution in the `filter` sub-module. You can find a great example under https://learn.adafruit.com/ulab-crunch-numbers-fast-with-circuitpython/fir-filter-example-pulse-rate-estimation, and https://learn.adafruit.com/ulab-crunch-numbers-fast-with-circuitpython/filter-example-measuring-barometric-pressure

If you need something more advanced, we could definitely add that. Can you point to a `scipy` (I think `numpy` does not contain any filter functions) function that you would like to see in `ulab`?

khashdroid, 2020-06-14:
Thanks a lot for the swift response!
Would be awesome to see the ulab version of the "**sosfilt**" function from scipy. It would match exactly what I'm looking for.
i.e. []((https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.sosfilt.html#scipy.signal.sosfilt))

v923z, 2020-06-15:
I don't know this particular filter, so I would have to look into how it actually works. But I see that it could be useful. 
@jepler is it OK, if we stick this into the filter module?

v923z, 2020-06-17:
@khashdroid Could you, please, check out the `sosfilter` branch (https://github.com/v923z/micropython-ulab/tree/sosfilter), and see if it does what you want? I have tried a couple of combinations, and tested it against `scipy`. It appears to be OK. 

Actually, it would be fantastic, if you could provide a test script along the lines https://github.com/v923z/micropython-ulab/tree/sosfilter/tests

If you find that the function is OK, I will write up the documentation. For now, you have to put up with this:

```
import ulab
from ulab import filter as filter

x = ulab.array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
sos = [[1, 2, 3, 1, 5, 6], [1, 2, 3, 1, 5, 6]]

print(filter.sosfilt(sos, x))
```

khashdroid, 2020-06-18:
oh many thanks! 
Actually, one minor change is needed. The input argument 'Zi' is also required (to update the state of the filter). 
So, basically, to give you an idea of what kind of functionality is needed, it can be described as the following (in pure circuitpython code):
`def sosfilt_2d_py(sos,X,zi,axis=-2):
    ''' pure python fallback for second-order-sections filter in case scipy isn't available '''
    X = ulab.array(X,dtype=ulab.float)
    sos = ulab.array(sos)
    #zi = ulab.array(zi) # cannot do this as zi is 3d (for now zi is declared as a list)
​
    Xshape = X.shape
    if not len(X.shape) == 2:
        print("Warning: X>2d.... treating as 2d...")
        X = X.reshape((-1,Xshape[-1]))
​
    if axis < 0:
        axis = len(X.shape) + axis
​
    if not axis == len(X.shape)-2:
        raise ValueError("Only for time in dim 0/-2")
​
    if len(sos.shape) != 2 or sos.shape[1] != 6:
        raise ValueError('sos must be shape (n_sections, 6)')
    if len(zi[0][0]) == 0 or len(zi[0]) != 2 or len(zi[0][0]) != X.shape[1]:
        raise ValueError('sos must be shape (n_sections, 2, dim)')
    for j in range(sos.shape[0]):
        if sos[j,3] != 1.0:
            sos[j,:] = sos[j,:]/sos[j,3]
    # loop over outputs
    v0 = 0
    for d in range(X.shape[1]):
        # loop over second order sections
        for j in range(sos.shape[0]):
            #print(sos.shape[0])
            # extract the old filter state
            v1, v2 = zi[j][0][d],zi[j][1][d] if not zi is None else (0,0)
            # extract the second order section coefficients & normalize
            #b0,b1,b2,a0,a1,a2 = sos[j,:]
            # format = [b0,b1,b2,a0,a1,a2]
            a0 = sos[j,3]
            a1 = sos[j,4]#/a0
            a2 = sos[j,5]#/a0
            b0 = sos[j,0]#/a0
            b1 = sos[j,1]#/a0
            b2 = sos[j,2]#/a0
​
            # loop over samples
            Xd=X[:,d]
            for i in range(len(Xd)):
                v0 = Xd[i] - a1*v1 - a2*v2
                Xd[i] = b0*v0 + b1*v1 + b2*v2
                v2 = v1
                v1 = v0
            # save the filter state
            #zi[j,:,d] = [v0,v1]
            zi[j][0][d] = v0
            zi[j][1][d] = v1
​
    return X, zi
​
` 
And an example of input data could something as following:

`def gen_dummy_data():
    SOS =  [[ 7.37970909e-01 ,-1.47594182e+00,  7.37970909e-01  ,1.00000000e+00
  ,-1.71607129e+00 , 7.37462332e-01],
 [ 1.00000000e+00 ,-2.00000000e+00 , 1.00000000e+00 , 1.00000000e+00,
  -1.77863177e+00 , 8.00802644e-01],
 [ 1.00000000e+00 ,-2.00000000e+00  ,1.00000000e+00 , 1.00000000e+00,
  -1.89850941e+00 , 9.22174574e-01],
 [ 1.05164680e-03 , 2.10329359e-03 , 1.05164680e-03 , 1.00000000e+00,
  -8.40286922e-01 , 1.88345161e-01],
 [ 1.00000000e+00 , 2.00000000e+00  ,1.00000000e+00  ,1.00000000e+00,
  -9.42809042e-01 , 3.33333333e-01],
 [ 1.00000000e+00 , 2.00000000e+00 , 1.00000000e+00 , 1.00000000e+00,
  -1.19543396e+00 , 6.90598923e-01]]
  
    Zi = [[[-0.72777233, -0.54736934, -0.55781371, -0.,        ],
  [ 0.72777233,  0.54736934,  0.55781371,  0.,        ]],
​
 [[-0.,         -0. ,        -0.  ,       -0.  ,      ],
  [ 0.,          0.,          0. ,         0. ,       ]],
​
 [[-0.,         -0. ,        -0.  ,       -0.  ,      ],
  [ 0.,          0.,          0.  ,        0. ,       ]],
​
 [[ 0.,          0. ,         0. ,         0. ,       ],
  [-0.,         -0. ,        -0.  ,       -0. ,       ]],
​
 [[ 0.,          0. ,         0. ,         0. ,       ],
  [-0.,         -0. ,        -0. ,        -0. ,       ]],
​
 [[ 0.,          0. ,         0. ,         0. ,       ],
  [-0.,         -0.,         -0.  ,       -0. ,       ]]]
    X =  [[0.7141362 , 0.64519614, 0.33859795, 0.        ],
 [0.63943064 ,0.7651917  ,0.1236788  ,0.        ],
 [0.44646132 ,0.9212642  ,0.6797532  ,0.        ],
 [0.3522703  ,0.287701   ,0.7862383  ,0.        ],
 [0.7481707  ,0.7931746  ,0.09341699 ,0.        ],
 [0.51579565 ,0.15268207 ,0.62807536 ,0.        ]]
    return SOS, Zi, X
`
And an example of usage could be:
`SOS,Zi,X=gen_dummy_data()
X_filtered=sosfilt_2d_py(SOS,X,Zi)
`

v923z, 2020-06-18:
@khashdroid Could you, please, check out the latest code? This works now
```
import ulab
from ulab import filter as filter

x = ulab.array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
sos = [[1, 2, 3, 1, 5, 6], [1, 2, 3, 1, 5, 6]]
zi = ulab.array([[1, 2], [3, 4]])

print(filter.sosfilt(sos, x, zi=zi))
```
Let me know, if you find glitches.

khashdroid, 2020-06-22:
so, for quick testing, I built on a unix system (Raspberry pi 3A+) and tested it against the scipy one. it works great. Seems to match the functionality of the scipy version. If I understand correctly, it works under a few assumption(s) such as: 
- the input value of axis is always -1
- X must be 1d (which actually implicitly implies that axis=-1).
so for an MxN matrix, the filter must be called M times to each row to fully filter the data. Or if the user wants to filter over axis=-2, the matrix can be transposed first and then the filter is applied. To summerize, I would say the ulab sosfilt provides the right functionality, thumbs up!

v923z, 2020-06-22:
Thanks for the feedback! You are right, this works for 1D arrays only. If needed, we can extend this to higher dimensions later. Also, since the array is one-dimensional, the axis keyword makes no difference, so I haven't implemented that. 
I will just merge this into master for now.

closed 2020-06-22 by v923z