v923z/micropython-ulab #128: Slicing inconsistency with ndarrays

mz3-andrey, 2020-06-23:
Was doing slicing on ulab version 1.19.0 and noticed something
I create a list with:
`test = [x for x in range(10)]`
and do a slice to give every third value:
`test[0::3]`
This returns as expected:
`[0, 3, 6, 9]`
However when doing the same slice to the equivalent ndarray i get a different result:
`np.array(test, dtype=np.int8)[0::3]`
returns
`[0, 3, 6]`
Doing both in python 3.6 with lists and np.ndarray returns the expected result [0, 3, 6, 9]
Am I missing something?
Thanks!

mz3-andrey, 2020-06-23:
Sorry just noticed #50, any hints on updating an openmv's firmware to include the most recent version?
Thanks.

closed 2020-06-23 by mz3-andrey

v923z, 2020-06-24:
You can take a look at https://github.com/v923z/micropython-ulab/issues/124, where we discussed at length, how you can compile `openmv` with `ulab`. I also modified my side of the code, so that Ibrahim could easily include it on the `openmv` side. You should raise an issue over there.

mz3-andrey, 2020-06-24:
Great! Thank you so much for your support.