v923z/micropython-ulab #129: 50.2 argmax, argmin, min, max weird behaviour int8 and uint8 datatypes

mz3-andrey, 2020-06-29:
Am running 50.2 on my openmv's now, but my code started doing weird things.
I looked further into it and noticed min, max, argmin and argmax were the culprit.
Issue #2984 mentions a fix as they weren't working for int16's before, but now only int16s seem to work.

My test:

`print(min, max, argmin, argmax)` of a given array.
dtypes of float, and uint16 and int16 work exactly as expected.
But for:
`[-2, 4, 2, -1, 6, 1, 0, -1], dtype=np.int8` the print result is `-2, 116, 0, 5`
and for
`[5, 7, 1, 4, 8], dtype=np.uint8` the print result is `0, 5, 2, 0`

v923z, 2020-06-29:
Thanks for reporting the issue! Something is clearly wrong here, I will look into it.

v923z, 2020-06-29:
@mz3-andrey https://github.com/v923z/micropython-ulab/pull/130 fixes the issue.

@jepler Jeff, could you, please, pull this into `circuitpython`? You can use the https://github.com/v923z/micropython-ulab/tree/0.51.1 tag.

closed 2020-06-29 by v923z

mz3-andrey, 2020-06-29:
Great thanks! I appreciate it.