v923z/micropython-ulab #132: Error building/linking ulab with double precision float (FP64/DP)

rcolistete, 2020-07-13:
Using source code of ulab v0.51.1 & MicroPython v1.12-623-gf743bd3d2 on 2020-07-13, compiling on Manjaro Linux v20.0.3 + arm-none-eabi-cpp 8-2019-q3-update 8.3.1 20190703.
Error building/linking ulab with double precision float (FP64/DP) :
```
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: build-PYBV11/code/vectorise.o: in function `vectorise_around':
vectorise.c:(.text.vectorise_around+0x80): undefined reference to `round'
make: *** [Makefile:614: build-PYBV11/firmware.elf] Erro 1
```
While building ulab with single precision float (FP32/SP) works :

v923z, 2020-07-13:
Thanks for bringing this to the fore! I think I know, where to dig for a solution. I will try to patch this up tonight, when I have a computer on which to compile the code. 

If you feel adventurous, or pressed for time, here is a hint: https://github.com/micropython/micropython/pull/5789 . Most probably, we will have to push the issue upstream.

v923z, 2020-07-13:
@rcolistete I have issued a PR in https://github.com/micropython/micropython/pull/6242.

v923z, 2020-07-21:
This has been solved upstream https://github.com/micropython/micropython/pull/6242, so I am closing now.

closed 2020-07-21 by v923z