v923z/micropython-ulab #133: ulab not compatible with enabling _thread from the make command

rcolistete, 2020-07-14:
Using source code of ulab v0.51.1 & MicroPython v1.12-623-gf743bd3d2 2020-07-14, compiling on Manjaro Linux v20.0.3 + arm-none-eabi-cpp 8-2019-q3-update 8.3.1 20190703.

Building ulab with thread enabled in MicroPython :
$ make -j8 CFLAGS_EXTRA='-DMICROPY_PY_THREAD=1' BOARD=PYBV11 USER_C_MODULES=../../../ulab all
builds a firmware wit'h '_thread' module, but without 'ulab' module, because 'CFLAGS_EXTRA' is used in the make command and in 'ulab/code/micropython.mk', so the value of the last is ignored.

The solution is to use 'override' and '+=' :
`override CFLAGS_EXTRA += -DMODULE_ULAB_ENABLED=1`
in last line of 'ulab/code/micropython.mk'.

v923z, 2020-07-14:
> The solution is to use 'override' and '+=' :
> `override CFLAGS_EXTRA += -DMODULE_ULAB_ENABLED=1`
> in last line of 'ulab/code/micropython.mk'.

Thanks for reporting the issue! Since you have already figured out the solution, do you want to fix it yourself? By doing so, you would get credit: https://github.com/v923z/micropython-ulab/graphs/contributors ;-)

rcolistete, 2020-07-14:
Done, I've made a PR :
[fixed incompatibility when enabling '_thread' module from the make command #134](https://github.com/v923z/micropython-ulab/pull/134)

v923z, 2020-07-14:
@rcolistete Many thanks! I have merged your PR.

closed 2020-07-14 by v923z