v923z/micropython-ulab #135: Incompatibility between gcc-arm 10.1 and ulab+MicroPython

rcolistete, 2020-07-15:
Due to a bug/incompatibility between gcc-arm 10.1.0 and ulab+MicroPython, it gives a link error :
```
[stm32]$ make -j8 BOARD=PYBD_SF6 USER_C_MODULES=../../../ulab all
    ...
    Including User C Module from ../../../ulab/code
    ...
    LINK build-PYBD_SF6/firmware.elf
    arm-none-eabi-ld: build-PYBD_SF6/code/ulab.o:(.bss.ulab_extras_module+0x0): multiple definition of `ulab_extras_module'; build-PYBD_SF6/code/extras.o:(.data.ulab_extras_module+0x0): first defined here
    arm-none-eabi-ld: build-PYBD_SF6/code/ulab.o:(.bss.ulab_vectorise_module+0x0): multiple definition of `ulab_vectorise_module'; build-PYBD_SF6/code/vectorise.o:(.data.ulab_vectorise_module+0x0): first defined here
    make: *** [Makefile:605: build-PYBD_SF6/firmware.elf] Error 1
```
With ulab, the firmware building works ok.

For example, using :  
- Manjaro/Arch Linux package 'arm-none-eabi-gcc v10.1.0-1' from official repository;
-  'gcc-arm-none-eabi-10-2020-q2-preview-x86_64-linux.tar.bz2' from [official ARM download site](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads), v10.1.1 20200529.

The solution is using older versions of gcc-arm from GNU Arm Embedded Toolchain Downloads, e. g.,
'gcc-arm-none-eabi-8-2019-q3-update', 'gcc-arm-none-eabi-9-2020-q2-update', etc.

The problem is :
- gcc-arm-none-eabi no preview v10.x being released in the following months may retain this incompatibility;
- gcc-arm-none-eabi  v10.x is a lot faster than previous versions (even half the compiling time).

v923z, 2020-07-16:
It seems to me that only two sub-modules, namely, `extras`, and `vectorise` cause the problem, so my feeling is that it could be solved. I am running ubuntu, and I think I installed the arm compiler separately, so I can't really comment on the issue.

v923z, 2020-07-16:
@rcolistete I believe, I found the error: `ulab_extras_module` and `ulab_vectorise_module` are not declared `extern` in `extras.h`, and `vectorise.h`. Could you, please, see whether you still have the issue, if you add `extern` to the declaration, as e.g., in `linalg.h`? If that is the fix, let me know, if you still want to collect brownie points, or I should change the code ;-)

rcolistete, 2020-07-16:
Thanks !
Yeah, it worked, now 'gcc-arm-none-eabi-10-2020-q2-preview' can compile ulab + MicroPython.
I'll made a PR.

rcolistete, 2020-07-16:
[PR#136](https://github.com/v923z/micropython-ulab/pull/136) made.

v923z, 2020-07-16:
> 
> 
> [PR#136](https://github.com/v923z/micropython-ulab/pull/136) made.

Thanks! Merged.

closed 2020-07-16 by v923z