v923z/micropython-ulab #137: Add numpy.trapz

rcolistete, 2020-07-16:
Please, add :
[numpy.trapz](https://numpy.org/doc/stable/reference/generated/numpy.trapz.html).

v923z, 2020-07-16:
Could be done, the only question is, which sub-module should hold the function. I would offer two candidates: `numerical` (it is already a bit heavy), and `approx`. I am open to other suggestions. Keep in mind that you had difficulties with the firmware size, so if you want to exclude sub-modules, then this function should be where it really belongs.

v923z, 2020-07-16:
@rcolistete Roberto, could you, please, check out the https://github.com/v923z/micropython-ulab/pull/138/ PR? This implements numerical integration for 1D ndarrays.

rcolistete, 2020-07-16:
Sub-module approx was a good choice, IMHO.

v923z, 2020-07-16:
@rcolistete I have run a couple of tests, and compared the results against `numpy`, but it would be great, if you could also verify the behaviour of the code.

rcolistete, 2020-07-16:
Ok, I'll do in the following hours.

v923z, 2020-07-16:
It's not so urgent. I am going to sleep now, anyway:)

v923z, 2020-07-17:
Actually, you can always do this:

```
y = array([1, 2, 3, 4, 5])
x = array([1, 2, 3, 4, 5])

0.5*sum((y[:-1]+y[1:])*diff(x))
```
This delivers  the same results as `trapz`. I haven't measured the execution times. The approach with the slices is probably faster (the array type is resolved at the beginning of the calculation, and the `for` loop just adds the numbers, while `trapz` calls a function repeatedly to retrieve the values of the array), but requires more RAM (because you have to create the `ndarrays`).

rcolistete, 2020-07-18:
Took more time to better test. Using the branch 'trapz' I've compiled ulab v0.52.0 + MicroPython v1.12 with single precision, without threads.

About the above question, answer is no, as trapz is a lot faster and uses a lot less RAM than the solution with slice+sum+diff :+1:
`
import time
import gc
import os
import ulab as np
from ulab import numerical
gc.collect()
print('MicroPython {}, {}, with initial free RAM {} bytes'.format(os.uname().version, os.uname().machine, gc.mem_free()))

numz = 8800
x = np.linspace(0.0, numz, num=numz + 1)/numz
y = np.vector.sin(x)
print('Free RAM after x and y vectors : {} bytes'.format(gc.mem_free()))

gc.collect()
ram1 = gc.mem_free()
t1 = time.ticks_us()
v1 = np.approx.trapz(y, x=x)
t2 = time.ticks_us()
dt21 = time.ticks_diff(t2, t1)
ram2 = gc.mem_free()
print('Value from trapz = {}, in {} us, using {} bytes'.format(v1, dt21, ram1 - ram2))

gc.collect()
ram3 = gc.mem_free()
t3 = time.ticks_us()
v2 = 0.5*numerical.sum((y[:-1]+y[1:])*numerical.diff(x))
t4 = time.ticks_us()
dt43 = time.ticks_diff(t4, t3)
ram4 = gc.mem_free()
print('Value from slice+sum+diff = {}, in {} us, using {} bytes'.format(v2, dt43, ram3 - ram4))
`
Output :
```
MicroPython v1.12-624-g57333f6ea on 2020-07-18, PYBD-SF2W with STM32F722IEK, with initial free RAM 180736 bytes
Free RAM after x and y vectors : 74576 bytes
Value from trapz = 0.4596981, in 3543 us, using 64 bytes
Value from slice+sum+diff = 0.4596981, in 15068 us, using 105888 bytes
```
With slice+sum+diff, numz = 8900 yields a "MemoryError: memory allocation failed, allocating 35600 bytes".
With trapz, numz = 14900 works.

rcolistete, 2020-07-18:
Anyway, 'trapz' importance is to have an useful feature (almost) compatible with NumPy.

rcolistete, 2020-07-18:
Sugestion fror 'ulab.approx.trapz()' :'
- accept 'scalar-valued iterable (ranges, lists, tuples containing numbers)', not only ndarrays.

Some tests from NumPy documentation for 'trapz' :
`import ulab as np
np.approx.trapz([1,2,3])
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    TypeError: wrong operand type
y = np.array([1,2,3]); y
    array([1.0, 2.0, 3.0], dtype=float)
np.approx.trapz(y)
    4.0
x = np.array([4,6,8]); x
np.approx.trapz(y, x=x)
    8.0
np.approx.trapz(y, dx=2)
    8.0
a = np.array(range(6), dtype=np.uint8); a
    array([0, 1, 2, 3, 4, 5], dtype=uint8)
np.approx.trapz(a)
    12.5
a.reshape((2, 3))
    array([[0, 1, 2],
           [3, 4, 5]], dtype=uint8)
np.approx.trapz(a, axis=0)
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    TypeError: extra keyword arguments given
np.approx.trapz(a, axis=1)
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    TypeError: extra keyword arguments given`

v923z, 2020-07-19:
> Suggestion fror 'ulab.approx.trapz()' :
> 
>     * accept 'scalar-valued iterable (ranges, lists, tuples containing numbers)', not only ndarrays.

I can add that. 

> 
>     * axis=0 and axis=1 is sometimes useful.
> 

I understand this. I am working on overhauling `ulab`, so that it could treat tensors of arbitrary dimensions. I wouldn't like to invest too much time into implementing something that will automatically become obsolete in two months' time. But I promise to do this properly in the upgraded version. For the time being, you can take a slice, and pass the 1D array to `trapz`.

v923z, 2020-07-19:
> Output :
>
> ```
> MicroPython v1.12-624-g57333f6ea on 2020-07-18, PYBD-SF2W with STM32F722IEK, with initial free RAM 180736 bytes
> Free RAM after x and y vectors : 74576 bytes
> Value from trapz = [...]

Thanks, this is a convincing argument.

rcolistete, 2020-07-19:
> I can add that.
>
>
> I understand this. I am working on overhauling `ulab`, so that it could treat tensors of arbitrary dimensions. I wouldn't like to invest too much time into implementing something [...]

The 1st one is more important. I don't use the 2nd feature, but some few others can.

v923z, 2020-07-19:
> The 1st one is more important. I don't use the 2nd feature, but some few others can.

I remember now, why I didn't implement this: if both `x`, and `y` can be iterables, then you'll have four cases (both are iterables, one of them is an iterable, or both are `ndarray`s. So, you would have to implement everything 4 times. The alternative, that you use the iterator object, thereby factoring out the four cases in a single pass, will be slow, because the iterator returns an `mp_obj_t`, and then you'll have to extract the value from that. The upshot is that if `trapz` will run at the same speed for iterables, and `ndarray`s, and I am pretty sure that is not what you want.

What is wrong with casting the iterable to an `ndarray` first?

rcolistete, 2020-07-19:
Ok, so it is harder than universal functions with only 1 argument.

Yes, casting with 'ulab.array(<iterable>)' is ok. I've commented just for the sake to be more compatible with 'numpy.trapz'.

But MicroPython and its modules can't be like Python 3.x and its modules due to RAM constraints, firmware size, etc.

v923z, 2020-07-19:
> Ok, so it is harder than universal functions with only 1 argument.

The complexity scales quadratically with the number of arguments. 

> 
> Yes, casting with 'ulab.array()' is ok. I've commented just for the sake to be more compatible with 'numpy.trapz'.

It will eat into your RAM;)

v923z, 2020-07-19:
@rcolistete Apart from the fact that `trapz` won't work with generic iterables, are you satisfied with the code? If so, I would like to merge it.

rcolistete, 2020-07-19:
Yes, tested it a lot, it works very wel.
Thanks.

v923z, 2020-07-19:
OK, done.

closed 2020-07-19 by v923z