v923z/micropython-ulab #139: Add numpy.arange()

rcolistete, 2020-07-18:
Suggestion to add ulab.arange() because [numpy.arange()](https://numpy.org/doc/stable/reference/generated/numpy.arange.html)is a lot popular, so it is easier for new ulab users to migrate from NumPy.

I prefer 'numpy.linspace()' in many situations, but 'numpy.arange()' is everywhere and the translation to 'numpy.linspace()' syntax takes a little bit of time.

v923z, 2020-07-19:
This function would have to be added at the top level (https://github.com/v923z/micropython-ulab/blob/master/code/create.c, https://github.com/v923z/micropython-ulab/blob/fbe8151f50797952a17a235ffa442dac83255d67/code/ulab.c#L75-L78), meaning that this would always be included. (I guess, it would add 300-500 bytes to the firmware.) On the other hand, the function itself is not quite safe: 

_For floating point arguments, the length of the result is ceil((stop - start)/step). Because of floating point overflow, this rule may result in the last element of out being greater than stop._

https://numpy.org/doc/stable/reference/generated/numpy.arange.html

Despite the popularity, `linspace` is the recommended way. 

The implementation is quite trivial, but before adding it, I would like to hear what other people have to say.

rcolistete, 2020-07-19:
Yes, it would be in top.level, 'ulab.arange()'.

The 'not safe' is a feature of float point numbers. Behaving like NumPy is not a problem, even when it is a limitation.

I also prefer 'numpy.linspace()', but 'arange' is very popular.

Another option, not optimum, is to describe in 'ulab.linspace()' docs how to convert 'numpy.arange()' syntax to 'ulab.linspace()' .

v923z, 2020-07-19:
OK, I will look into it. 

@jepler Jeff, could we add this generally, or you think it's not worth it? If the latter case, we could switch off the function for `circuitpython`. I think you've mentioned that on some platforms you are very tight with flash now, so I am a bit cautious with throwing in random functions.

jepler, 2020-07-19:
I think ulab is disabled on all of the "nearly full" ports, I wouldn't worry about it

v923z, 2020-07-19:
Thanks for the feedback! I will then try to add this in the next couple of days.

v923z, 2020-07-19:
@rcolistete https://github.com/v923z/micropython-ulab/tree/arange contains a draft implementation. I still have to figure out, how the potentially missing `start` argument should be treated. For now, you can do 
```
import ulab
print(ulab.arange(1, 13, 3, dtype=ulab.uint8))
print(ulab.arange(1, 13, dtype=ulab.uint8))
```
i.e., the number of arguments must be at least 2.

v923z, 2020-07-20:
@rcolistete Roberto, I have just pushed my most recent changes to the `arange` branch. Could you, please, check, whether it does what it is supposed to? Thanks!

rcolistete, 2020-07-20:
Ok. I've tested 'arange()' from arange branch of ulab in ports/unix. Output is Indented :
```
import ulab as np
np.arange(3)
    array([0.0, 1.0, 2.0], dtype=float)    # Different than NumPy : array([0, 1, 2])
np.arange(3.0)
    array([0.0, 1.0, 2.0], dtype=float)    # Ok, NumPy : array([0., 1., 2.])
np.arange(3, dtype=np.uint8)
    array([0, 1, 2], dtype=uint8)            # Ok, equal to NumPy
np.arange(3, 7)
    array([3.0, 4.0, 5.0, 6.0], dtype=float)    # Different than NumPy : array([3, 4, 5, 6])
np.arange(3, 7, 2)
    array([3, 5], dtype=int16)                 # Ok, NumPy : array([3, 5])
np.arange(3, 7, 2.0)
    array([3.0, 5.0], dtype=float)            # Ok, NumPy : array([3., 5.])
```
Nice to have 'arange()' with all features (1 to 3 arguments, etc).

We see that when there is a step argument as integer (as well start and stop as integers), the output is an integer array.
But without the 3rd argument, with start as integer, or start and stop as integers, the output is a float array.
I know that in ulab the default type (of ndarray, etc) is float, but **the above behaviour is discrepancy**.

v923z, 2020-07-20:
Thanks for the feedback!

v923z, 2020-07-20:
@rcolistete I think this is fixed now. Let me know, whether we can merge this.

rcolistete, 2020-07-20:
Now the default is integer, even when start, stop and set are floats, but dtype=np.float outputs a float array :
``
import ulab as np
np.arange(3)
    array([0, 1, 2], dtype=int16)
np.arange(3.0)
    array([0, 1, 2], dtype=int16)
np.arange(3, dtype=np.float)
    array([0.0, 1.0, 2.0], dtype=float)
np.arange(3.0, dtype=np.float)
    array([0.0, 1.0, 2.0], dtype=float)
np.arange(3, 7)
    array([3, 4, 5, 6], dtype=int16)
np.arange(3, 7.0)
    array([3, 4, 5, 6], dtype=int16)
np.arange(3.0, 7.0)
    array([3, 4, 5, 6], dtype=int16)
np.arange(3, 7, dtype=np.float)
    array([3.0, 4.0, 5.0, 6.0], dtype=float)
np.arange(3, 7, 2)
    array([3, 5], dtype=int16)
np.arange(3, 7, 2.0)
    array([3, 5], dtype=int16)
np.arange(3, 7, 2.0, dtype=np.float)
    array([3.0, 5.0], dtype=float)
```
So there is no discrepancy when using 1, 2 or 3 arguments.
Only different behaviour with respect to NumPy, with the need to use dtype=np.float to have float array output.

v923z, 2020-07-20:
Did you pull the very latest code?

rcolistete, 2020-07-20:
After the commit 'found and fixed small bug in dtype assignment' :
```
>>> import ulab as np
>>> np.arange(3)
array([0, 1, 2], dtype=int16)
>>> np.arange(3.0)
array([0.0, 1.0, 2.0], dtype=float)
>>> np.arange(3, dtype=np.float)
array([0.0, 1.0, 2.0], dtype=float)
>>> np.arange(3.0, dtype=np.float)
array([0.0, 1.0, 2.0], dtype=float)
>>> np.arange(3, 7)
array([3, 4, 5, 6], dtype=int16)
>>> np.arange(3, 7.0)
array([3.0, 4.0, 5.0, 6.0], dtype=float)
>>> np.arange(3.0, 7)
array([3.0, 4.0, 5.0, 6.0], dtype=float)
>>> np.arange(3.0, 7.0)
array([3.0, 4.0, 5.0, 6.0], dtype=float)
>>> np.arange(3, 7, dtype=np.float)
array([3.0, 4.0, 5.0, 6.0], dtype=float)
>>> np.arange(3, 7, 2)
array([3, 5], dtype=int16)
>>> np.arange(3, 7, 2.0)
array([3.0, 5.0], dtype=float)
>>> np.arange(3.0, 7, 2)
array([3.0, 5.0], dtype=float)
>>> np.arange(3, 7.0, 2)
array([3.0, 5.0], dtype=float)
>>> np.arange(3, 7, 2, dtype=np.float)
array([3.0, 5.0], dtype=float)
```
Seems it is 100% ok.

v923z, 2020-07-20:
Many thanks for sticking with it till the very end! I will merge this now.

rcolistete, 2020-07-20:
It is me to say thanks !

v923z, 2020-07-20:
https://github.com/v923z/micropython-ulab/pull/140 closes the issue.

rcolistete, 2020-07-21:
There is a bug compiling with SP (single precision, FP32) :
```
[stm32]$ make -j8 BOARD=PYBLITEV10 USER_C_MODULES=../../../ulab all
...
CC ../../../ulab/code/ndarray.c
CC ../../../ulab/code/create.c
CC ../../../ulab/code/linalg.c
CC ../../../ulab/code/vectorise.c
CC ../../../ulab/code/poly.c
CC ../../../ulab/code/fft.c
CC ../../../ulab/code/numerical.c
CC ../../../ulab/code/filter.c
CC ../../../ulab/code/compare.c
CC ../../../ulab/code/approx.c
CC ../../../ulab/code/extras.c
CC ../../../ulab/code/ulab.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC build-PYBLITEV10/pins_PYBLITEV10.c
CC build-PYBLITEV10/frozen_content.c
LINK build-PYBLITEV10/firmware.elf
arm-none-eabi-ld: build-PYBLITEV10/code/create.o: in function `create_arange':
create.c:(.text.create_arange+0x90): undefined reference to `ceil'
make: *** [Makefile:618: build-PYBLITEV10/firmware.elf] Erro 1
```
While the DP (Double Precison, FP64) build is ok :
```
[stm32]$ make -j8 MICROPY_FLOAT_IMPL=double BOARD=PYBLITEV10 USER_C_MODULES=../../../ulab all
...
CC ../../../ulab/code/ndarray.c
CC ../../../ulab/code/create.c
CC ../../../ulab/code/linalg.c
CC ../../../ulab/code/vectorise.c
CC ../../../ulab/code/poly.c
CC ../../../ulab/code/fft.c
CC ../../../ulab/code/numerical.c
CC ../../../ulab/code/filter.c
CC ../../../ulab/code/compare.c
CC ../../../ulab/code/approx.c
CC ../../../ulab/code/extras.c
CC ../../../ulab/code/ulab.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC build-PYBLITEV10/pins_PYBLITEV10.c
CC build-PYBLITEV10/frozen_content.c
LINK build-PYBLITEV10/firmware.elf
   text    data     bss     dec     hex filename
 393036     124   43604  436764   6aa1c build-PYBLITEV10/firmware.elf
GEN build-PYBLITEV10/firmware.dfu
GEN build-PYBLITEV10/firmware.hex
```

rcolistete, 2020-07-21:
@v923z, see the above bug, we forgot to test with ARM building.

rcolistete, 2020-07-21:
> @v923z, see the above bug, we forgot to test with ARM building.

Sent [PR#141 "fix arange : missing MICROPY_FLOAT_C_FUN(ceil) in create.c"](https://github.com/v923z/micropython-ulab/pull/141).

rcolistete, 2020-07-22:
Another difference from `numpy.arange`, error happens only on ulab :
ulab :
```
import ulab as np
np.arange(10, 2)
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    MemoryError: memory allocation failed, allocating 4294967280 bytes
```
NumPy :
```
import numpy as np
np.arange(10, 2)                                                                                                                           
    array([], dtype=int64)
```

v923z, 2020-07-22:
> @v923z, another difference from `numpy.arange`, errors happens only on ulab :
> ulab :
> 
> ```
> import ulab as np
> np.arange(10, 2)
>     Traceback (most recent call last):
>       File "<stdin>", line 1, in <module>
>     MemoryError: memory allocation failed, allocating 4294967280 bytes
> np.arange(10, 20, -2)
>     Traceback (most recent call last):
>       File "<stdin>", line 1, in <module>
>     MemoryError: memory allocation failed, allocating 4294967286 bytes
> ```

This comes from the line that you fixed. I don't mean to say that your fixed screwed anything up, I am just saying that the problem is in https://github.com/v923z/micropython-ulab/blob/b3e76e1f0a7041e5f41e7e4df146e066cdcc1635/code/create.c#L197 

I will fixed it later today.

v923z, 2020-07-22:
Has been fixed in https://github.com/v923z/micropython-ulab/pull/146, closing now.

closed 2020-07-22 by v923z