v923z/micropython-ulab #142: Extend 'ulab.linalg.dot' to accept 2 vectors

rcolistete, 2020-07-21:
Suggestion to extend 'ulab.linalg.dot' to accept 2 vectors, so 'ulab.linalg.dot' becomes more similar to [`numpy.dot`](https://numpy.org/doc/stable/reference/generated/numpy.dot.html?highlight=numpy%20dot).
Output is indented :
```
import ulab as np
a = np.array([1, 2, 3]); a
    array([1.0, 2.0, 3.0], dtype=float)
b = np.array([4, 5, 6]); b
    array([4.0, 5.0, 6.0], dtype=float)
b2 = np.array([4, 5]); b2
    array([4.0, 5.0], dtype=float)
np.linalg.dot(a, b)
    32.0
np.linalg.dot(a, a)
    14.0
np.linalg.dot(b, b)
    77.0
# So it is not so difficult to calculate the norm of a vector using ulab :
import math
math.sqrt(np.linalg.dot(a, a))
    3.741657386773941
```
I've made a [commit 'linalg.dot : also accepts 2 vectors of same dimension'](https://github.com/rcolistete/micropython-ulab/commit/0d80175933c08366ef2ebe9415e698361e7eed9b) to the branch 'dot_vectors' of my ulab fork.

Any suggestions ?

If ok, I can make a PR. As it is my first day adding feature to ulab in C, I'm asking before sending any PR.

rcolistete, 2020-07-21:
I missed that there is also another form with 'ulab' to have the inner product of vectors :
```
np.numerical.sum(a*a)
    14.0
np.numerical.sum(a*b)
    32.0
np.numerical.sum(b*b)
    77.0
import math
math.sqrt(np.numerical.sum(a*a))   # norm of a
    3.741657386773941
```

rcolistete, 2020-07-21:
But a 'np.linalg.norm(vector)' would be a lot more practical then the last 2 input lines above. 8-)

v923z, 2020-07-22:
> But a 'np.linalg.norm(vector)' would be a lot more practical then the last 2 input lines above. 8-)

That could be right, but `norm` has at least 8 cases. https://numpy.org/doc/stable/reference/generated/numpy.linalg.norm.html
Most of these are either not very relevant, or aliases for functions that are already implemented.

v923z, 2020-07-22:
> Suggestion to extend 'ulab.linalg.dot' to accept 2 vectors, so 'ulab.linalg.dot' becomes more similar to [`numpy.dot`](https://numpy.org/doc/stable/reference/generated/numpy.dot.html?highlight=numpy%20dot). So the 1st case will be accepted :
> `- If both a and b are 1-D arrays, it is inner product of vectors (without complex conjugation).`
> besides the 2nd one :
> `If both a and b are 2-D arrays, it is matrix multiplication, ...`

> I've made a [commit 'linalg.dot : also accepts 2 vectors of same dimension'](https://github.com/rcolistete/micropython-ulab/commit/0d80175933c08366ef2ebe9415e698361e7eed9b) to the branch 'dot_vectors' of my ulab fork.
> 
> Any suggestions ?
> 
> If ok, I can make a PR. As it is my first day adding features to ulab in C, I'm asking before sending any PR.

This is OK, could be merged. Thanks!

rcolistete, 2020-07-22:
> That could be right, but `norm` has at least 8 cases. https://numpy.org/doc/stable/reference/generated/numpy.linalg.norm.html
> Most of these are either not very relevant, or aliases for functions [...]

I was thinking about norm with only 1 argument, vector x, doing the calculation `math.sqrt(np.numerical.sum(a*a))` but more optimized. I will implement one version and make a benchmark.

v923z, 2020-07-22:
> I was thinking about norm with only 1 argument, vector x, doing the calculation `math.sqrt(np.numerical.sum(a*a))` but more optimized. I will implement one version and make a benchmark.

But you have only a single un-optimised function call, `sqrt`. You can't by-pass `np.numerical.sum(a*a)`.

v923z, 2020-07-22:
https://github.com/v923z/micropython-ulab/pull/143 solves the issue.

closed 2020-07-22 by v923z