v923z/micropython-ulab #155: What's the difference between ulab and numpy?

yang8621, 2020-07-28:
I know it's not fully compatible with numpy and not all functions are implemented. But Is there any place that details the difference between ulab and numpy? 

2nd question: Is it possible to use minimal port with ulab?

v923z, 2020-07-28:
> I know it's not fully compatible with numpy and not all functions are implemented. But Is there any place that details the difference between ulab and numpy?

1. The number of functions implemented. Obviously, on a microcontroller, flash space is limited, as is RAM.

2. `ulab` is organised into sub-modules, so that you can easily remove functions that are not required, thereby saving flash space. https://github.com/v923z/micropython-ulab/#modules This also means that apart from the five top-level array-creation functions, you have to append the function name with the module's name, like 

```
import ulab
from ulab import poly

p = ulab.array([1, 2, 3])
poly.polyval(p, [4, 5, 6])
```
Alternatively, you can do something like this

```
import ulab
from ulab.poly import *

p = ulab.array([1, 2, 3])
polyval(p, [4, 5, 6])
```

3. You can find a lot on this in https://micropython-ulab.readthedocs.io/en/latest. For each function, I pointed out the differences to `numpy`, e.g., if a function doesn't take a particular keyword argument, how the up-casting rules work etc.

> 2nd question: Is it possible to use minimal port with ulab?

What is a minimal port? `ulab` is a simple user-module that you can compile for any `micropython` platform. You simply have to follow the instructions in https://github.com/v923z/micropython-ulab/#compiling.

yang8621, 2020-07-28:
Thanks for your explanation.

Yes. I compared dir(numpy) and dir(ulab), and there were many unimplemented functions. And I believe some are viable to implement, e.g. reshape.

I refer to "micropython/ports/minimal". It works under "micropython/ports/unix". But seems it's not built under "ports/minimal" with _make USER_C_MODULES=../../../ulab all_.

v923z, 2020-07-28:
> Thanks for your explanation.
> 
> Yes. I compared dir(numpy) and dir(ulab), and there were many unimplemented functions. And I believe some are viable to implement, e.g. reshape.

`reshape` is implemented: https://micropython-ulab.readthedocs.io/en/latest/ulab.html?highlight=reshape#reshape

> I refer to "micropython/ports/minimal". It works under "micropython/ports/unix". But seems it's not built under "ports/minimal" with _make USER_C_MODULES=../../../ulab all_.

Oh, I can't really comment on this. I have never used the minimal port. Actually, I don't even know how and what for I could use it.

yang8621, 2020-07-29:
And take reshape as an example, seems it's not full compatible with numpy:

np.array(range(15)).reshape((3,5)) -- OK
np.arange(15)   -- OK
np.arange(15).reshape(3,5)  -- not supported

v923z, 2020-07-29:
> And take reshape as an example, seems it's not full compatible with numpy:
> 
> np.array(range(15)).reshape((3,5)) -- OK
> np.arange(15) -- OK
> np.arange(15).reshape(3,5) -- not supported

Your last example is syntactically incorrect: you have to pass a `tuple` to `reshape`.

closed 2020-07-29 by v923z